arendst/Tasmota #10092: MGC3130 integration not workng

dedepene, 2020-12-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The sensor is recognized as connected but the readings reported are switched or missing. For instance, a tap/doubletap on the South side of the sensor board reports correctly South, a tap on the west reports east, a tap on the actual East side reports North, while the North side does not register anything at all in this mode.

The other thing is that modes cannot be changed at all. [COMMAND : SENSOR36 0 ](https://tasmota.github.io/docs/MGC3130/)always results in error:

`12:29:55 CMD: SENSOR36 0
12:29:55 RSL: stat/tasmota_C1D57C/RESULT = {"Command":"Error"}`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Nodemcu v3
- [ x] Tasmota binary firmware version number used: 
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [x ] IDE / Compiler used: Platformio
- [x ] Flashing tools used: Tasmotizer
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
12:18:02 CMD: Backlog Template
12:18:02 RSL: stat/tasmota_C1D57C/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}

12:18:45 CMD: module
12:18:45 RSL: stat/tasmota_C1D57C/RESULT = {"Module":{"18":"Generic"}}

12:19:10 CMD: GPIO 255
12:19:10 RSL: stat/tasmota_C1D57C/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"640":"I2C SDA"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"2336":"MGC3130 Xfr"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"2368":"MGC3130 Rst"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:20:34 CMD: Status 0
12:20:34 RSL: stat/tasmota_C1D57C/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_C1D57C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:20:34 RSL: stat/tasmota_C1D57C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:09:49","StartupUTC":"2020-12-08T11:10:45","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2020-12-08T11:59:03","SaveCount":14,"SaveAddress":"F6000"}}
12:20:34 RSL: stat/tasmota_C1D57C/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020-12-08T10:54:50","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"342/699"}}
12:20:34 RSL: stat/tasmota_C1D57C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":[".|.",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
12:20:34 RSL: stat/tasmota_C1D57C/STATUS4 = {"StatusMEM":{"ProgramSize":610,"Free":392,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010053C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,36"}}
12:20:34 RSL: stat/tasmota_C1D57C/STATUS5 = {"StatusNET":{"Hostname":"tasmota_C1D57C-5500","IPAddress":"192.168.1.148","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"54.93.173.153","Mac":"EC:FA:BC:C1:D5:7C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:20:34 RSL: stat/tasmota_C1D57C/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C1D57C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:20:34 RSL: stat/tasmota_C1D57C/STATUS7 = {"StatusTIM":{"UTC":"2020-12-08T11:20:34","Local":"2020-12-08T12:20:34","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:31","Sunset":"16:53"}}
12:20:34 RSL: stat/tasmota_C1D57C/STATUS10 = {"StatusSNS":{"Time":"2020-12-08T12:20:34"}}
12:20:34 RSL: stat/tasmota_C1D57C/STATUS11 = {"StatusSTS":{"Time":"2020-12-08T12:20:34","Uptime":"0T00:09:49","UptimeSec":589,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":".|.","BSSId":"F0:9F:C2:DD:75:99","Channel":11,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:04"}}}
12:20:45 APP: Serial logging disabled

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
12:25:38 RSL: tele/tasmota_C1D57C/SENSOR = {"Time":"2020-12-08T12:25:38","MGC3130":{"TH_S":1}}
12:25:39 RSL: tele/tasmota_C1D57C/SENSOR = {"Time":"2020-12-08T12:25:39","MGC3130":{"TP_S":1}}
12:25:39 RSL: tele/tasmota_C1D57C/SENSOR = {"Time":"2020-12-08T12:25:39","MGC3130":{"DT_S":1}}
12:25:40 RSL: tele/tasmota_C1D57C/SENSOR = {"Time":"2020-12-08T12:25:40","MGC3130":{"TH_E":1}}
12:25:40 RSL: tele/tasmota_C1D57C/SENSOR = {"Time":"2020-12-08T12:25:40","MGC3130":{"TP_E":1}}
12:25:41 RSL: tele/tasmota_C1D57C/SENSOR = {"Time":"2020-12-08T12:25:41","MGC3130":{"DT_E":1}}
12:25:43 RSL: tele/tasmota_C1D57C/SENSOR = {"Time":"2020-12-08T12:25:43","MGC3130":{"TH_N":1}}
12:25:43 RSL: tele/tasmota_C1D57C/SENSOR = {"Time":"2020-12-08T12:25:43","MGC3130":{"TP_N":1}}
12:25:45 RSL: tele/tasmota_C1D57C/SENSOR = {"Time":"2020-12-08T12:25:45","MGC3130":{"TH_N":1}}
12:25:45 RSL: tele/tasmota_C1D57C/SENSOR = {"Time":"2020-12-08T12:25:45","MGC3130":{"TP_N":1}}
12:25:45 RSL: tele/tasmota_C1D57C/SENSOR = {"Time":"2020-12-08T12:25:45","MGC3130":{"DT_N":1}}
12:25:45 RSL: tele/tasmota_C1D57C/SENSOR = {"Time":"2020-12-08T12:25:45","MGC3130":{"NONE":1}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-12-08:
You have to use tasmota-sensors version

dedepene, 2020-12-08:
> You have to use tasmota-sensors version

thanks for pointing out. So adding #define USE_MGC3130 to user_config_override.h is not enough?

dedepene, 2020-12-08:
nope. that didn't do anything at all - exact same problems after switching to the sensors build. the sensor is detected but it won't work!

does anyone have any other idea????

`00:00:00 Project tasmota Tasmota Version 9.1.0.2(sensors)-2_7_4_9(2020-12-08T12:00:52)
00:00:00 I2C: MGC3130 found at 0x42
`

Jason2866, 2020-12-08:
@Staars can you help?

Staars, 2020-12-08:
I will take a look.

Staars, 2020-12-08:
How unfunny, it does not work here too.
I get inconsistent readings from the sensor, so I suspect problems with I2C.
What works is, that the firmware info can be read. But then I get readings with zeros or simply nothing.

Stay tuned ...

Staars, 2020-12-09:
After fiddling with the current driver without access, I began with more base work and tried out a basic sketch (no Tasmota involved) and this did not work either.
Next step will be to test with on Arduino.
It could be, that my sensor is simply broken.

@dedepene If you find a demo sketch, that works for you, please post a link.

Staars, 2020-12-12:
So ... one skywriter board later (my old one was really defective from lying in the drawer 😠).

Driver still works as expected for me, at least for the 5 minutes that I had the chance to play with.
The errors you see regarding `sensor36 0` are only cosmetical, because no code for the console was implemented back in time.
I will try to re-work some bits of the code in order to get and present more information from the device (read failures, environmental noise, ...). Maybe we can find something useful.

@dedepene 
Which exact MGC3130-hardware do you use? Does it work in another setup (any other demo sketch, maybe another non-ESP8266-MCU)?

dedepene, 2020-12-13:
@Staars, I have the Pimoroni Skywriter as well. I still can't get it to work - what are the chances of mine being defective as well???

Let's see if I get this right. So I would reboot the mcu, just to be on the safe side, and then issue SENSOR36 1 to get into gesture mode, right? I get the error message and I keep getting no reading for events on the North side, south is fine, east reports a north event, and west a north. :( Issuing SENSOR36 2 does nothing really - i don't get any airwheel readings at all.

Maybe point me to a specific sketch to try?

Thanks!

Staars, 2020-12-13:
Okay, I have this board too.

In normal operation there is nothing to configure, because you always will get `mode 1` after boot. It should start automatically.

Such errors (regarding bitfields) are always a bit suspicious of incorrect memory alignments. But this should not happen here and it does not for me.

I can not point to another sketch, that will work out of the box. The next example are no minimal working examples and you need WS2812-boards for these:
https://github.com/pimoroni/skywriter-hat/tree/master/arduino/Skywriter/examples
It would have been nice, if Pimoroni had provided a simple sketch just for the sake of testing the integrity of the device.

I pushed a PR, maybe it helps to narrow down the problem for you. This is not really a bugfix, because with my new board, there was no perceptible bug to fix.

dedepene, 2020-12-14:
so you're saying i got a defective board? also, if you could please give me a link to the PR so i can follow up and test when merged. thanks!

Staars, 2020-12-14:
It is already merged. You can test, but this will only show a bit more status messages to you. Maybe these help to better understand what’s going on.
The known facts suggest, that your device is defective, but I encourage you to try it out with other example sketches too. I can only tell, that developing the driver back in the days was a pretty stable experience and did not show weird behaviour.

closed 2021-01-01 by github-actions[bot]