arendst/Tasmota #10099: SwitchMode 1 and 2 swapped (inverted)

vic42, 2020-12-09:
### PROBLEM DESCRIPTION
SwitchModes 1 and 2 do not behave like stated in the docs. They behave exactly opposite.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 DevBoard V4
- [ ] Tasmota binary firmware version number used: 9.1.0.2
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: platformio
- [x] Flashing tools used: esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
12:21:22 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
12:21:22 SRC: Backlog
12:21:22 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
12:21:22 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
12:21:23 SRC: Backlog
12:21:23 CMD: Group 0, Index 1, Command "MODULE", Data ""
12:21:23 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
12:21:23 SRC: Backlog
12:21:23 CMD: Group 0, Index 1, Command "GPIO", Data "255"
12:21:23 RSL: RESULT = {"GPIO0":{"33":"Button2"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"160":"Switch1"},"GPIO5":{"1376":"WS2812"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"608":"I2C SCL"},"GPIO26":{"640":"I2C SDA"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GP ...

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
N/A

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:23:35 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_EA4C14","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":3600,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,1,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:23:35 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software reset CPU","Uptime":"0T00:52:51","StartupUTC":"2020-12-09T10:30:44","Sleep":50,"CfgHolder":4617,"BootCount":74,"BCResetTime":"2020-11-21T02:46:39","SaveCount":18952}}
12:23:35 RSL: STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020-12-07T22:16:46","Core":"1_0_4_2","SDK":"v3.3.4-232-g44ec7972b","CpuFrequency":80,"Hardware":"ESP32","CR":"427/699"}}
12:23:35 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["2ghz",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008003","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
12:23:35 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1321,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","00140001","000000CF","010013C0","40000981","00004000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29","Sensors":"1,2,3,5,6,62"}}
12:23:35 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_EA4C14-3092","IPAddress":"172.23.0.96","Gateway":"172.23.0.254","Subnetmask":"255.255.255.0","DNSServer":"31.12.7.254","Mac":"24:0A:C4:EA:4C:14","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:23:35 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-12-09T11:23:35","Local":"2020-12-09T12:23:35","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:32","Sunset":"16:52"}}
12:23:35 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-12-09T12:23:35","Switch1":"OFF"}}
12:23:35 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-12-09T12:23:35","Uptime":"0T00:52:51","UptimeSec":3171,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":56,"Color":"8F008F","HSBColor":"300,100,56","Channel":[56,0,56],"Scheme":7,"Width":1,"Fade":"ON","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"attingo_2ghz","BSSId":"B4:FB:E4:11:20:B7","Channel":1,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

12:27:40 SRC: Switch
12:27:40 RSL: RESULT = {"POWER":"ON"}
12:27:40 RSL: POWER = ON

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Switch on / off a "switch" component GPIO electrically

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Switch CLOSED (shorted to GND) should report "ON"
Switch OPEN (left floating to 3.3V) should report "OFF"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-12-09:
Hi, 

Sorry but the OPEN or CLOSED of your switch will depends on how it is electrically connected your switch. For electrical stability (to not have a floating voltage) your switch must have a pull up resistor or a pull down. Any of those circuits will produce the inverse logic than the other and you should choose switchmode1 1 or switchmode1 2 depending on that.

For a standard sonoff basic (the first device Tasmota supports) the explanations of the docs are correct.

So, the option are there so you can choose the one that fits your own case.

Thanks.

vic42, 2020-12-09:
OK. Understood. You didn't even mind to read.
BTW: the sonoff basic does not even have a switch or switch input.

I have to re-think contributing code fixes to the project.

vic42, 2020-12-09:
A standard sonoff basic doesn't even have a relevant switch or switch input.

ascillato, 2020-12-09:
A sonoff basic is used for a lot of users with PIR sensors, reed switches, etc etc due to it has some gpios available in its header to be used.

As explained in the docs, you need to use switchmode1 1 for a PIR sensor.

blakadder, 2020-12-09:
switchmodes  1 and 2; 9 and 10; 13 and 14 have changed/inverted their default behaviour somewhere during this year and are not behaving as written in the documentation

vic42, 2020-12-09:
Barbudor reversed SwitchMode 3 & 4 in the docs some three weeks ago BTW:
https://github.com/tasmota/docs/commit/1f54d084e2662ea2e764b8b72c77f9b68213ee0f

blakadder, 2020-12-10:
> Barbudor reversed SwitchMode 3 & 4 in the docs some three weeks ago BTW:
> [tasmota/docs@1f54d08](https://github.com/tasmota/docs/commit/1f54d084e2662ea2e764b8b72c77f9b68213ee0f)

since what's written in docs is actually true shouldn't that change be reverted then @barbudor

ascillato, 2020-12-10:
For switchmode 1 and 2, it is correct as it is in the docs.

The confusion comes that depending on the circuit that is connected to a GPIO, you will see a 1 or a 0 when closing the switch. If you have a pull up resistor, when closing the switch you will see a 0. If you have a circuit with a pull down resistor you will see a 1. All switches must have a pull up or pull down resistor in order to avoid ghost switching due to floating voltage. The ESP8266 and ESP32 have internal configurable pull up resistors, but as explained in their datasheets, those resistor are weak, so with long wires to a switch, noise can trigger the logic.

So it is, and always will be, circuit dependant. So to match the real state of the switch with the 0 and 1 logic (ON OFF), you have the option of SWITCHMODE 1 or 2

vic42, 2020-12-10:
no need to explain electricity to an R&D engineer

Jason2866, 2020-12-10:
@vic42  i know you know ;-)
Since github issues can be found via search engines it is good thing to have a correct detailed answer for everyone.

ascillato, 2020-12-10:
There is something more or we can close this? There is no need to modify neither the code or the docs.

The short answer is that commercial devices have pull up resistors so, it is correct that `switchmodeN 1` is used for switch CLOSED = switch ON and switch OPEN = switch OFF

Yes, for pull up resistor circuit, is "inverted" the logical voltage of 0 volts measured on the GPIO for ON state (switch closed) and 3.3 volts for OFF state (switch open), but this is the NORMAL behavior on commercial devices.

Anyway, you are able to select the SWITCHMODE that fits better your circuit+logic desired.

blakadder, 2020-12-10:
dunno, i'm completely lost now in engineer lingo

closed 2020-12-10 by blakadder

vic42, 2020-12-10:
In fact it is the other way around.

3.3V on GPIO == "ON" == switch "OPEN" in that circuit
0V on GPIO == "OFF" == switch "CLOSED" in that circuit

NB: for a "button" it is vice versa

You can easily get confused here and that is why i was after clarification.

Sorry @ascillato for my previous comment, and thanks for @Jason2866 pointing it out.

ascillato, 2020-12-10:
Yes, I totally agree that this can be a source of confusion. 

Anyway, the voltage on the gpio is just dependant on the circuit. Different hardware, different meaning for the logic of ON or OFF.

To match the real voltage on the pin, to the real state of a switch, on every case, you have the configuration of switchmode 1 or 2. For some hardware you will see that it match exactly the docs. For other hardwares (pull down resistor switches) it is going to be inverted as the docs. So, it is just a config that the user has to do. Must be Tested and put the correct switchmode to match the real state of the switch with the logic. Nothing more than that.

Switch state -> Voltage ->  logic

Switch can be open or closed but depending on the hardware can be open=3.3v or open=0v

So, to match open with the ON logic, you have the switchmode command.

For standard commercial devices, it is already taken care on the templates and on the configuration explained in the docs for each device.