arendst/Tasmota #10103: PZEM-004 - wrong Energy totals

lustyffh, 2020-12-10:
### PROBLEM DESCRIPTION
I have wierd calculation of Energy Totals: Today, Yesterday, Total. They seem to be getting randon even during the day.
I tried both external counter and firmware counter(with reset of totals). Below output is using PZEM counter.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- X[ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): ESP-01
- [X ] Tasmota binary firmware version number used: 9.1.0.1
  - [ ] Pre-compiled
  - [X ] Self-compiled
    - [X ] IDE / Compiler used: VSCode
- [X ] Flashing tools used: NodeMCU PyFlasher
- [X ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

18:47:06 MQT: stat/boxroom_powermeter/RESULT = {"NAME":"ESP01-PZEM","GPIO":[1,1536,1,1472,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
18:47:06 MQT: stat/boxroom_powermeter/RESULT = {"Module":{"0":"ESP01-PZEM"}}
18:47:06 MQT: stat/boxroom_powermeter/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"1536":"PZEM016 Rx"},"GPIO2":{"0":"None"},"GPIO3":{"1472":"PZEM0XX Tx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [X ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

18:47:38 MQT: stat/boxroom_powermeter/STATUS = {"Status":{"Module":0,"DeviceName":"boxroom_powermeter","FriendlyName":["boxroom_powermeter"],"Topic":"boxroom_powermeter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:47:38 MQT: stat/boxroom_powermeter/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Hardware Watchdog","Uptime":"0T01:36:51","StartupUTC":"2020-12-10T14:10:47","Sleep":50,"CfgHolder":4617,"BootCount":249,"BCResetTime":"2020-11-15T13:32:02","SaveCount":1245,"SaveAddress":"3F8000"}}
18:47:38 MQT: stat/boxroom_powermeter/STATUS2 = {"StatusFWR":{"Version":"9.1.0.1(tasmota)","BuildDateTime":"2020-11-15T13:24:14","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"429/699"}}
18:47:38 MQT: stat/boxroom_powermeter/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["default-2",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00400200","00006000","00000000"]}}
18:47:38 MQT: stat/boxroom_powermeter/STATUS4 = {"StatusMEM":{"ProgramSize":645,"Free":3428,"Heap":21,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FC0C787","04368001","00171FCF","012012C0","C400F991","0000C084","00000000"],"Drivers":"1,2,3,4,6,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,38,40","Sensors":"1,3,4,5,6,7,8,9,10,13,14,15,17,41,54,62"}}
18:47:38 MQT: stat/boxroom_powermeter/STATUS5 = {"StatusNET":{"Hostname":"boxroom_powermeter-0867","IPAddress":"x.x.x.x","Gateway":"x.x.x.x","Subnetmask":"x.x.x.x","DNSServer":"x.x.x.x","Mac":"18:FE:34:x:x:x","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:47:38 MQT: stat/boxroom_powermeter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.99.1","MqttPort":1883,"MqttClientMask":"boxroom_powermeter","MqttClient":"boxroom_powermeter","MqttUser":"xxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:47:38 MQT: stat/boxroom_powermeter/STATUS7 = {"StatusTIM":{"UTC":"2020-12-10T15:47:38","Local":"2020-12-10T18:47:38","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+03:00","Sunrise":"10:33","Sunset":"18:52"}}
18:47:38 MQT: stat/boxroom_powermeter/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:47:38 MQT: stat/boxroom_powermeter/STATUS10 = {"StatusSNS":{"Time":"2020-12-10T18:47:38","ENERGY":{"TotalStartTime":"2020-12-10T00:00:00","Total":16646.145,"Yesterday":6692.529,"Today":36910.949,"Power":143,"ApparentPower":152,"ReactivePower":52,"Factor":0.94,"Frequency":50,"Voltage":227,"Current":0.669}}}
18:47:38 MQT: stat/boxroom_powermeter/STATUS11 = {"StatusSTS":{"Time":"2020-12-10T18:47:38","Uptime":"0T01:36:51","UptimeSec":5811,"Vcc":3.545,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"default-2","BSSId":"4C:5E:0C:5C:40:07","Channel":11,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just turn on device and let it run. Screenshots were taken within the same day.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The totals should be consistent and correllate with the average power utilisation

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-12-10:
Wow, too much consumption!!! ;) hehhehe

If you have a PZEM004, try to use in gpio1 the `pzem004 Rx` instead of your "GPIO1":{"1536":"PZEM016 Rx"}

lustyffh, 2020-12-10:
> 
> 
> Wow, too much consumption!!! ;) hehhehe
> 
> If you have a PZEM004, try to use in gpio1 the `pzem004 Rx` instead of your `"GPIO1":{"1536":"PZEM016 Rx"}`

I have PZEM-004T V3 to be precise. It has different protocol. Therefore If try to use `pzem004 Rx` on GPIO1 then don't get any readings at all.

Jason2866, 2020-12-10:
Did you modify the PZEM device to be 3.3V level conform?

lustyffh, 2020-12-10:
> 
> 
> Did you modify the PZEM device to be 3.3V level conform?

Yes, I used 6CH logic level converter to connect it to ESP. Otherwise, I guess, I didn't even get voltage/current readings. Which seem to be correct.

TrDA-hab, 2020-12-11:
> > Did you modify the PZEM device to be 3.3V level conform?
> 
> Yes, I used 6CH logic level converter to connect it to ESP. Otherwise, I guess, I didn't even get voltage/current readings. Which seem to be correct.

This is my wiring diagram, you may see this will help you.
[image]

lustyffh, 2020-12-11:
> 
> 
> > > Did you modify the PZEM device to be 3.3V level conform?
> > 
> > 
> > Yes, I used 6CH logic level converter to connect it to ESP. Otherwise, I guess, I didn't even get voltage/current readings. Which seem to be correct.
> 
> This is my wiring diagram, you may see this will help you.
> [image]

Your diagram doesn't indicate that there should be voltage level converter on the Tx/Rx pins or a midifed PZEM module. In my case voltage level converter was used.

TrDA-hab, 2020-12-12:
> На вашей диаграмме не указано, что на выводах Tx / Rx должен быть преобразователь уровня напряжения или модифицированный модуль PZEM. В моем случае использовался преобразователь уровня напряжения.

Everything is correct. No voltage divider is needed for my circuit. There is no error.

closed 2020-12-27 by ascillato2