arendst/Tasmota #10105: Tasmota 9.1.0 (wemo and hue emulation) and alexa woes

hgschmie, 2020-12-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Gosund WP3
- [X] Tasmota binary firmware version number used: 9.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
20:28:55 CMD: Backlog Template; Module; GPIO 255
20:28:55 RSL: RESULT = {"NAME":"Gosund WP3","GPIO":[0,0,0,0,32,0,0,0,320,321,224,0,0,0],"FLAG":0,"BASE":18}
20:28:55 RSL: RESULT = {"Module":{"0":"Gosund WP3"}}
20:28:55 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"32":"Button1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"320":"Led_i1"},"GPIO13":{"321":"Led_i2"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```

- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
20:29:49 CMD: STATUS 0
20:29:49 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"tree","FriendlyName":["tree"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:29:49 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release","RestartReason":"Software/System restart","Uptime":"0T00:07:24","StartupUTC":"2020-12-10T19:22:25","Sleep":50,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2020-12-10T19:05:36","SaveCount":54,"SaveAddress":"FB000"}}
20:29:49 RSL: STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"367/699"}}
20:29:49 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WTam",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
20:29:49 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
20:29:49 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-0407","IPAddress":"192.168.2.221","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.18","Mac":"F4:CF:A2:00:61:97","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
20:29:49 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-12-10T19:29:49","Local":"2020-12-10T20:29:49","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:33","Sunset":"16:52"}}
20:29:49 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-12-10T20:29:49"}}
20:29:49 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-12-10T20:29:49","Uptime":"0T00:07:24","UptimeSec":444,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WTam","BSSId":"F0:9F:C2:24:20:1A","Channel":1,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
WeMo:

20:30:27 UPP: WeMo Type 2, Response sent to 192.168.2.65:50000
20:30:27 HTP: WeMo setup from 192.168.2.65
20:30:27 HTP: WeMo basic event from 192.168.2.65

Hue:
20:34:06 UPP: Hue 3 response packets sent to 192.168.2.77:50000
20:34:06 HTP: Hue setup
20:34:06 HTP: Hue API (/6197/lights) from 192.168.2.77
20:34:06 HTP: Hue Result ({"399729":{"state":{"on":false,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"tree","modelid":"Gosund WP3","manufacturername":"Tasmota","uniqueid":"f4:cf:a2:00:61:97:00:11-1"}})
20:34:07 HTP: Hue API (/6197/lights/399729) from 192.168.2.77
20:34:07 /lights path=/lights/399729
20:34:07 HTP: Hue Result ({"state":{"on":false,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"tree","modelid":"Gosund WP3","manufacturername":"Tasmota","uniqueid":"f4:cf:a2:00:61:97:00:11-1"})
20:34:07 HTP: Hue API (/6197/lights) from 192.168.2.77
20:34:07 HTP: Hue Result ({"399729":{"state":{"on":false,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"tree","modelid":"Gosund WP3","manufacturername":"Tasmota","uniqueid":"f4:cf:a2:00:61:97:00:11-1"}})
```

### TO REPRODUCE
* Update to Tasmota 9.1.0
* Enable WeMo emulation
* Search for new devices in the Alexa app

### EXPECTED BEHAVIOUR

* Alexa detects the new device

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

I have a few other Tasmota Devices (with older firmware) successfully paired with Alexa and use them in the house (a few of them are Gosund WP3 as well). I tried to set up another with the latest firmware. It does not work. I can see the WeMo or Hue setup packets arrive (192.168.2.65 and 192.168.2.77 are Alexa Echo devices) but they never show in the Alexa app.

s-hadinger, 2020-12-10:
I'm surprised, your logs are normal and 9.1 is known to work with Alexa. Are you sure you don't have a conflicting name with another device,

hgschmie, 2020-12-10:
no obvious conflicts (all the tasmotas are hostnamed tasmota-%04d) and the devices have different friendly names. This new device is named "tree" (it is that time of the year) and I have no other "tree" device in the house.

s-hadinger, 2020-12-10:
You can try removing an existing one and trying to discover it again. This would tmmaybe give us more information

hgschmie, 2020-12-10:
I did that and it does not discover anymore either.

hgschmie, 2020-12-11:
any additional information that I can provide? I tried this with Echo devices on the same Wifi access point and on different ones.

hgschmie, 2020-12-11:
I ran a capture on my network; this is the result. It shows that the device receives two requests, one for setup.xml and a SOAP request to report the status of the device. If you can load this, 192.168.2.216 is my laptop, there is a console session open, so there are a lot of http requests from that; 192.168.2.77 is the alexa device, 192.168.2.251 is the plug in question. 

(this may be missing the broadcast/multicast solicitation for the device as I limited the capture to the IP of the plug; the UDP answer (packet no. 32) is very visible)

[capture3.zip](https://github.com/arendst/Tasmota/files/5676867/capture3.zip)

s-hadinger, 2020-12-11:
One possibility is that an Alexa skill for Smart Home is disturbing the discovery process. I had this once when I was developing my own skill. 

Please check your current skills and try disabling some smart home skills (not the regular ones)

Jason2866, 2020-12-11:
Every application doing interaction with hue and wemo devices can disturb the discovery.
Delete all discovered Tasmota devices. Switch off all apps and devices which are not tasmota or a Alexa hardware. Restart the Alexa hardware (unplug power supply). Let Alexa (hardware) do the
device discovery. If not all devices found at first try. Let Alexa discover again.

tpxtron, 2020-12-11:
I‘m experiencing tha same issue. Some Tasmota enabled Smart Plugs (Gosund SP1) worked fine since yesterday, today they are shown in the alexa app that they‘re not responding. When asking alexa to toggle one of the switches, the voice response says the device is not found. I‘ve tried removing and re-adding the affected plugs, but they were not discovered. Restarted both the plugs and the echo dot (1st gen), but to no avail. Setting „setoption19“ to „1“ doesn‘t help either (found that in another issue). Strangely, some of the plugs are still working fine...

hgschmie, 2020-12-11:
So I created a fully isolated network (only connected to the internet and I only put the echo device and the plug into it) and did the discovery again. Did not work. I see the packets arrive at the plug but then they do not show up in the Alexa app. 

Not sure what to do next.

Jason2866, 2020-12-12:
@tpxtron  echo 1st gen does not work since it does not support this function.

tpxtron, 2020-12-12:
> A echo dot 2nd generation is needed. Better is a echo dot 3nd generation.

My bad. It's a 2nd Gen Dot. It previously worked with all (freshly flashed) Tasmota sockets and now only finds a few of them, the others "dont't answer" and are not discovered when they've been removed and I search for new devices.

Jason2866, 2020-12-12:
For 2nd Generation `setoption109 1` is needed

tpxtron, 2020-12-12:
As I already wrote: tried that to no avail.

hgschmie, 2020-12-14:
Guess I am a bit smarter now: The WeMo devices are discovered using SSDP (which is a standard protocol, using UDP on a multicast group to discover devices). This seems to work well. I got a scanner and the tasmotas chirp happily their information every time any device (including the Echo devices) send out the "anyone there?" packets.

However, even with me seeing the alexa device fetching the metainformation (setup.xml etc.), the devices do not show. I don't want to sound paranoid or anything, but this starts to feel like Belkin/Amazon quietly changed their side of the deal to maybe fingerprint some information (the 3.1415 model number looks very fingerprintable) and ignore devices that report information that they do not know about. 

Don't know what else to say. This *should* work but short of recompiling a custom tasmota where those values are changes, I got nothing.

hgschmie, 2020-12-14:
(yes, I tried both setoption109 0 and setoption109 1)

s-hadinger, 2020-12-14:
I confirm both Wemo and Hue emulation DO work well with Alexa. I just tested the full discovery cycle on both protocols yesterday.

Why did you have 2 different source IPs in your logs? 192.168.2.65 and 192.168.2.77. Are they both Echo devices?

Would you be kind enough to do a Hue discovery with `Weblog 4` and paste the entire log sequence, including packet received by other devices?

hgschmie, 2020-12-14:
Well, I have three. :-) (Two Echo Dot, one Echo Show 5). Discovery tends to happen from the dots, but the three are chatting happily all the time. Yes, .65 and .77 are both 2nd gen Echo Dots. 

Hue setup:

```
18:54:33 WIF: Checking connection...
18:54:34 UDP: Packet (119)
18:54:34 UDP: Packet (119)
18:54:34 UDP: Packet (119)
18:54:35 UDP: Packet (119)
18:54:35 UDP: Packet (119)
18:54:35 UDP: Packet (119)
18:54:35 UDP: Packet (119)
18:54:35 UDP: Packet (119)
18:54:36 UPP: Hue 3 response packets sent to 192.168.2.66:50000
18:54:37 HTP: Hue setup
18:54:37 HTP: Hue API () from 192.168.2.66
18:54:37 HTP: Hue POST args ({"devicetype": "Echo"})
18:54:37 HTP: Hue Authentication Result ([{"success":{"username":"f2926f"}}])
18:54:37 HTP: Hue API (/f2926f/lights) from 192.168.2.66
18:54:37 HTP: Hue Result ({"254355185":{"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"wall-lamp","modelid":"Gosund WP3","manufacturername":"Tasmota","uniqueid":"60:01:94:f2:92:6f:00:11-1"}})
18:54:37 HTP: Hue API () from 192.168.2.66
18:54:37 HTP: Hue POST args ({"devicetype": "Echo"})
18:54:37 HTP: Hue Authentication Result ([{"success":{"username":"f2926f"}}])
18:54:37 HTP: Hue API (/f2926f/lights) from 192.168.2.66
18:54:37 HTP: Hue Result ({"254355185":{"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"wall-lamp","modelid":"Gosund WP3","manufacturername":"Tasmota","uniqueid":"60:01:94:f2:92:6f:00:11-1"}})
18:54:37 HTP: Hue API (/f2926f/lights/254355185) from 192.168.2.66
18:54:37 /lights path=/lights/254355185
18:54:37 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"wall-lamp","modelid":"Gosund WP3","manufacturername":"Tasmota","uniqueid":"60:01:94:f2:92:6f:00:11-1"})
18:54:43 UDP: Packet (119)
18:54:43 UDP: Packet (119)
18:54:43 UDP: Packet (119)
18:54:43 UDP: Packet (119)
18:54:43 UDP: Packet (119)
18:54:43 UDP: Packet (119)
18:54:44 UDP: Packet (101)
18:54:46 UPP: Hue 3 response packets sent to 192.168.2.16:41302
18:54:50 UDP: Packet (119)
18:54:50 UDP: Packet (119)
18:54:50 UDP: Packet (119)
18:54:50 UDP: Packet (119)
18:54:50 UDP: Packet (119)
18:54:50 UDP: Packet (119)
18:54:50 UDP: Packet (119)
18:54:50 UDP: Packet (119)
18:54:50 UDP: Packet (119)
18:54:53 WIF: Checking connection...
18:54:54 UDP: Packet (101)
18:54:56 UPP: Hue 3 response packets sent to 192.168.2.16:41302
18:55:04 UDP: Packet (101)
18:55:06 UPP: Hue 3 response packets sent to 192.168.2.16:41302
18:55:13 WIF: Checking connection...
18:55:14 UDP: Packet (101)
18:55:16 UPP: Hue 3 response packets sent to 192.168.2.16:41302
18:55:20 UDP: Packet (119)
18:55:20 UDP: Packet (119)
18:55:20 UDP: Packet (119)
18:55:20 UDP: Packet (119)
18:55:20 UDP: Packet (119)
18:55:20 UDP: Packet (119)
18:55:21 UDP: Packet (119)
18:55:21 UDP: Packet (119)
18:55:21 UDP: Packet (119)
18:55:23 UDP: Packet (119)
18:55:24 UDP: Packet (119)
18:55:24 UDP: Packet (119)
18:55:24 UDP: Packet (101)
18:55:25 UDP: Packet (119)
```

In a surprising twist of events, this time it discovered the switch successfully with the Hue emulation. So with the Hue emu, it now works.

hgschmie, 2020-12-14:
(the 192.168.2.16 is a plex server; the packets that the switch reports are UPnP/SSDP discovery packets)

hgschmie, 2020-12-14:
this is from another plug (which sits literally in the outlet next to the one that discovered ok) and does not show:

```
19:06:24 UDP: Packet (101)
19:06:26 HTP: Hue setup
19:06:30 UDP: Packet (119)
19:06:30 UDP: Packet (119)
19:06:30 UDP: Packet (119)
19:06:33 UPP: Hue 3 response packets sent to 192.168.2.66:50000
19:06:33 WIF: Checking connection...
19:06:34 UDP: Packet (101)
19:06:34 HTP: Hue setup
19:06:34 HTP: Hue API (/6197/lights) from 192.168.2.66
19:06:34 HTP: Hue Result ({"399729":{"state":{"on":true,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"tree-plug","modelid":"Gosund WP3","manufacturername":"Tasmota","uniqueid":"f4:cf:a2:00:61:97:00:11-1"}})
19:06:34 HTP: Hue API (/6197/lights/399729) from 192.168.2.66
19:06:34 /lights path=/lights/399729
19:06:34 HTP: Hue Result ({"state":{"on":true,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"tree-plug","modelid":"Gosund WP3","manufacturername":"Tasmota","uniqueid":"f4:cf:a2:00:61:97:00:11-1"})
19:06:34 HTP: Hue API (/6197/lights) from 192.168.2.66
19:06:34 HTP: Hue Result ({"399729":{"state":{"on":true,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"tree-plug","modelid":"Gosund WP3","manufacturername":"Tasmota","uniqueid":"f4:cf:a2:00:61:97:00:11-1"}})
19:06:36 UPP: Hue 3 response packets sent to 192.168.2.16:41302
19:06:44 UDP: Packet (101)
19:06:46 UPP: Hue 3 response packets sent to 192.168.2.16:41302
19:06:49 UDP: Packet (119)
19:06:49 UDP: Packet (119)
19:06:49 UDP: Packet (119)
19:06:49 UDP: Packet (119)
19:06:49 UDP: Packet (119)
19:06:49 UDP: Packet (119)
19:06:53 WIF: Checking connection...
19:06:54 UDP: Packet (101)
19:06:55 UPP: Hue 3 response packets sent to 192.168.2.16:41302
19:07:04 UDP: Packet (101)
19:07:05 UPP: Hue 3 response packets sent to 192.168.2.16:41302
19:07:13 WIF: Checking connection...
19:07:14 UDP: Packet (101)
```

hgschmie, 2020-12-14:
(I also tried with the plex server stopped (no UPnP packets from .16), but it makes no difference.

hgschmie, 2020-12-14:
so I upgraded the plug that would not discover (it was still running 8.5.1) to 9.1.0 and now it also discovers in the Hue emulation. I am not sure how the different pieces of the puzzle fit together.

s-hadinger, 2020-12-14:
This is still a mystery. The Hue emulation code has not changed between 8.5.1 and 9.1.0 (as far as I recall).

hgschmie, 2020-12-15:
I am currently using some other plugs where I want to use the tasmotas, so I can keep playing around with them. I am happy to help you debug this and figure it out; I am somewhat handy with networks and computers but I have not (yet) compiled the firmware myself. I have written some ESP8266 code, so there may be the right tool chain lying around. :-)

hgschmie, 2020-12-22:
Any news here? Can I help you to debug / hone in on that problem?

s-hadinger, 2020-12-22:
The problem most likely does not lie on Tasmota side. As said before, the Hue code has not changed between versions. This is most certainly due to your local configuration. 

We can't investigate unless you find a way to reproduce the issue.

hgschmie, 2020-12-23:
I would agree. That is why I asked "is there anything that I can help you to debug to understand these problems better?" I have these problems (well the hue emulation is working right now) and I am happy to help everyone to figure this out.  However, I am not a tasmota expert so I don't even know where to start.

lullerboi, 2020-12-23:
Hello,

I'm also having issues since using 9.1.0 / 9.2.0.

After the update, I've been testing the new WeMo multi device feature and at first thought it was working fine.... until I tried re-adding the devices to my Alexa Rooms. It wasn't possible any more. Alexa just would ignore my selections in the device list and add OTHER devices to the groups. I reverted to Hue emulation and re-add the devices as desired.

Now, since yesterday evening, two of my devices (LED Strips with both the same controller) won't work any more with Alexa. The webinterface is reachable and can be used to control the devices. Alexa claims the devices don't react. Restarting them has no effect. I can also still control them using diyHue, so I suppose it's not a network issue.

Can anyone else confirm these issues?

There apparently also was an Amazon update for my devices (FireTV Cube + 2 x Echo Dot 3rd Gen), which completely f*cked up my multiroom group for some reason (when listening to music, it stutters and cracks suddenly, has been working flawlessly for ~1 year before), maybe the problems are also related to that?!

I will try to delete the devices, change their names and so on and will report back. I hope I can make them work again...

AGMOI, 2020-12-25:
@lullerboi: I can confirm these issues. Since tasmota version 9.2.0.0 I have the same problems. Yesterday evening I installed the last developmend version, so alexa works "normal" if I add new devices in my rooms. But my other 3 non tasmota devices (shelly RGBW & shelly dimmer2) are still unknown for alexa.

peteakalad, 2020-12-25:
The multiple device wemo change broke adding to routines/groups as the serial number was an invalid format. The change to the serial number also caused duplicate devices if you upgraded and did a discovery. 

This should be fixed now with the latest 9.2.0 release, but will require you to delete any devices discovered with the initial wemo multiple broken builds and rediscover.

closed 2020-12-27 by ascillato2

hgschmie, 2020-12-28:
I updated my plugs to 9.2.0 and they still do not get discovered. I have renamed them, removed everything from my alexa account and went through all the motions again (setoption109 1 etc.). 

I see the devices being queries by one of the alexa devices but they never show up in the app.  So I do not consider this issue closed.

lullerboi, 2020-12-28:
Try disabling emulation and re-enable it. Remove ALL devices from Alexa (not only Tasmota devices).

The issues with my multiroom group I described were also somehow related to this. Had to re-add my Thermostats (Homematic IP) and even my Echo Dots as well.

lullerboi, 2021-01-01:
I'm still having issues. I deleted all Alexa groups and re-added all devices, but when re-creating the group "Flur" , Alexa kept assigning the device "Brunnen" (device marked <<< in below list) instead of the desired one (marked ***). I had to revert to Hue emulation again.

I have 4 plugs, is it possible the calculation of the "serial number" part still causes conflicts if you have devices with almost similar mac adresses?

192.168.0.26 | f4:cf:a2:03:25:e3 Espressif Inc. | eth0 | Smart_WZ_SW1.JJHomeLAN 
192.168.0.27 | 2c:f4:32:d0:eb:80 Espressif Inc. | eth0 | Smart_SZ_SW1.JJHomeLAN <<<
192.168.0.28 | f4:cf:a2:02:a2:60 Espressif Inc. | eth0 | Smart_KU_SW1.JJHomeLAN 
192.168.0.29 | f4:cf:a2:02:aa:ff Espressif Inc. | eth0 | Smart_FL_SW1.JJHomeLAN ***

I also set "Smart_WZ_SW1" and so on as hostname, maybe this is causing problems?

Original WeMo serial numbers seem to always have 14 characters, is the generated string longer and Alexa just chops off the rest?

I didn't delete my TV and Amplifier from Alexa because they are a pain in the a** to set up again with my Fire TV Cube.
So if you guys think the calculation looks solid I will also delete these and re-create all my Alexa groups and routines once more (yay!).

jenswes, 2021-02-18:
Hi,
i've the same issue with alexa's hue detection, Tasmota 9.20 and Tasmota 9.20.7. Having two Slitinto RGB Bulbs discovered first (few days ago), but alexa lost connection on one one hour later, rediscover worked once, but after some days not using them another not reachable by alexa.  i deleted that device but alexa could not dicover again...
Web and mqtt is always fine.
So i restore a backup from the woking bulb, renamed the backup-source and now no device is anymore discovered.

After reading above, setting to weblog 4 helps, both devices are discovered immediately.

Is there the hue code running faster as before and slow down with weblog 4 makes alexa happy again? just a guess

and thank you for tasmota, i love it :-)

lullerboi, 2021-02-18:
[quote]
**If anyone else has these kind of problems: Try disabling Hue emulation and enable it again.**
[/quote]

If that doesn't help please open a new topic. This one was dedicated to WeMo problems.