arendst/Tasmota #10106: Zigbee Bridge TLS values not overridden in user_config_override.h

tam481, 2020-12-10:
### PROBLEM DESCRIPTION
It is not possible to override **USE_MQTT_TLS_CA_CERT** and **USE_MQTT_AWS_IOT_LIGHT** when building "Sonoff Zigbee Bridge

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [*] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [* ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [*] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [*] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [*] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [*] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [*] Tasmota binary firmware version number used: 9.1.0.2
  - [ ] Pre-compiled
  - [*] Self-compiled
    - [ ] IDE / Compiler used: PlatformIO
- [*] Flashing tools used: Tasmota Web Page
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
 0:35:42 MQT: stat/zbBridge/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
20:35:42 MQT: stat/zbBridge/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
20:35:42 MQT: stat/zbBridge/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
20:36:16 MQT: stat/zbBridge/STATUS = {"Status":{"Module":75,"DeviceName":"Sonoff zbBridge","FriendlyName":["Sonoff zbBridge"],"Topic":"zbBridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1}}
20:36:16 MQT: stat/zbBridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"ZigbeeBridge","OtaUrl":"http://192.168.10.10:88/zbBridge.bin","RestartReason":"Power On","Uptime":"0T00:02:38","StartupUTC":"2020-12-10T20:33:38","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2020-12-09T22:41:18","SaveCount":9,"SaveAddress":"F8000"}}
20:36:16 MQT: stat/zbBridge/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(zbbridge)","BuildDateTime":"2020-12-10T20:24:25","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"415/699"}}
20:36:16 MQT: stat/zbBridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Hacienda","Hacienda"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000080E9","2805C8000100060000005A00000000000000","00000000","00206000","00000000"]}}
20:36:16 MQT: stat/zbBridge/STATUS4 = {"StatusMEM":{"ProgramSize":581,"Free":420,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1006C6","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,23,41","Sensors":""}}
20:36:16 MQT: stat/zbBridge/STATUS5 = {"StatusNET":{"Hostname":"zbBridge-3512","IPAddress":"192.168.10.15","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.10","Mac":"98:F4:AB:E2:AC:B9","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
20:36:16 MQT: stat/zbBridge/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.homeserver.lan","MqttPort":8883,"MqttClientMask":"zbBridge","MqttClient":"zbBridge","MqttUser":"mqtt_user","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:36:16 MQT: stat/zbBridge/STATUS7 = {"StatusTIM":{"UTC":"2020-12-10T20:36:16","Local":"2020-12-10T20:36:16","StartDST":"2020-03-29T01:00:00","EndDST":"2020-10-25T02:00:00","Timezone":99,"Sunrise":"07:55","Sunset":"15:50"}}
20:36:16 MQT: stat/zbBridge/STATUS10 = {"StatusSNS":{"Time":"2020-12-10T20:36:16"}}
20:36:16 MQT: stat/zbBridge/STATUS11 = {"StatusSTS":{"Time":"2020-12-10T20:36:16","Uptime":"0T00:02:38","UptimeSec":158,"Vcc":3.504,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Hacienda","BSSId":"84:A4:12:06:9A:AC","Channel":7,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Add the following to the user_config_override.h file

#ifdef USE_ZBBRIDGE_TLS                            // Enable TLS for ZbBridge
    #undef USE_MQTT_TLS_CA_CERT
    #undef USE_MQTT_AWS_IOT_LIGHT                 // Enable MQTT for AWS IoT in light mode, with user/password instead of private certificate
#endif                                             // USE_ZBBRIDGE_TLS

#ifdef USE_MQTT_TLS_CA_CERT
#undef USE_MQTT_TLS_CA_CERT
#endif 
#ifdef USE_MQTT_AWS_IOT_LIGHT
#undef USE_MQTT_AWS_IOT_LIGHT
#endif 

### EXPECTED BEHAVIOUR
TLS CA validation to be disabled

### SCREENSHOTS
Getting error 62 when connecting to MQTT with TLS

### ADDITIONAL CONTEXT
See forum post
https://groups.google.com/g/sonoffusers/c/nRYI3ndmuJA

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-12-11:
As designed. See https://tasmota.github.io/docs/Compile-your-build/#customize-your-build

ascillato2, 2020-12-11:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-11 by ascillato2

tam481, 2020-12-11:
Hi @ascillato2 and @Jason2866. Is it by design that when you attempt to override the enablement of TLS by using undefining the variables in "user_config_override.h" that it won't disable it? Or is it the case that the "user_config_override.h" will only work to override variables only for the default "Tasmota" build and not for example zbBridge?

Jason2866, 2020-12-12:
By design that changes with user_config_override.h only is for build Tasmota.
Using build Tasmota all customized configs for all use cases can be done.