arendst/Tasmota #10108: Rules not processing nested JSON data correctly

theOzzieRat, 2020-12-10:
### PROBLEM DESCRIPTION
I am trying to use a rule to publish json data to mqtt and either the rule fails to trigger or it publishes a null string.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 9.1.0(tasmota)
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used:
- [ ] Flashing tools used:
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
{"NAME":"Generic","GPIO":[1,3200,160,3232,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
{"Module":{"18":"Generic"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"160":"Switch1"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"225":"Relay2"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":79,"Free":432,"Rules":"ON system#boot DO Backlog baudrate 115200; seriallog 0; serialsend2 Start endon"}
{"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":74,"Free":437,"Rules":"on SerialReceived#Update do publish2 stat/garage_dropsaw/PIR %value% endon"}
{"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":72,"Free":439,"Rules":"on SerialReceived#Data do publish2 stat/garage_dropsaw/PIR %value% endon"}
```
- [x] Provide the output of this command: `Status 0`:
```
{"Status":{"Module":18,"DeviceName":"Garage Dropsaw","FriendlyName":["Garage Dropsaw","Tasmota2"],"Topic":"garage_dropsaw","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"garage_dropsaw","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":0}}
{"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://10.0.1.12:9541/data/firmwares/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:45:00","StartupUTC":"2020-12-10T22:52:18","Sleep":50,"CfgHolder":4617,"BootCount":96,"BCResetTime":"2020-08-06T05:23:54","SaveCount":458,"SaveAddress":"F5000"}}
{"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020-11-07T11:57:45","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"422/699"}}
{"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ProAtomic IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00400089","2805C8000100060000005A00000000000000","40000000","00006000","00000000"]}}
{"StatusMEM":{"ProgramSize":600,"Free":400,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
{"StatusNET":{"Hostname":"Sonoff-17","IPAddress":"10.0.10.17","Gateway":"10.0.10.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.10.1","Mac":"8C:AA:B5:0D:4C:00","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
{"StatusMQT":{"MqttHost":"10.0.1.12","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0D4C00","MqttUser":"sun-mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
{"StatusTIM":{"UTC":"2020-12-10T23:37:18","Local":"2020-12-11T00:37:18","StartDST":"2020-10-04T02:00:00","EndDST":"2020-04-05T03:00:00","Timezone":"+01:00","Sunrise":"08:34","Sunset":"16:52"}}
{"StatusSNS":{"Time":"2020-12-11T00:37:18","Switch1":"ON","Shutter1":{"Position":94,"Direction":0,"Target":94}}}
{"StatusSTS":{"Time":"2020-12-11T00:37:18","Uptime":"0T00:45:00","UptimeSec":2700,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","PWM":{"PWM1":476},"Wifi":{"AP":1,"SSId":"ProAtomic IoT","BSSId":"92:F6:52:A4:BB:9E","Channel":6,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}}
{"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":5,"Close":5,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"5"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
```
00:28:11 MQT: tele/garage_dropsaw/RESULT = {"SerialReceived":{"Update":{"Channel":2,"Value":688,"State":"Alarm Tamper"}}}
00:28:11 RUL: SERIALRECEIVED#UPDATE performs "publish2 stat/garage_dropsaw/PIR "
00:28:11 MQT: stat/garage_dropsaw/PIR =  (retained)
```

### TO REPRODUCE
Configure device as serial bridge and create rules as above.
Send the following json string from a serial console to the device.
```
{"Update":{"Channel":2,"Value":688,"State":"Alarm Tamper"}}
```

### EXPECTED BEHAVIOUR

1. I would expect rule2 to trigger %value% to contain the string {"Channel":2,"Value":688,"State":"Alarm Tamper"}. This rule does in fact trigger but %value% is a null string (no value).

2. I would expect rule3 to trigger and return the string {"Update":{"Channel":2,"Value":688,"State":"Alarm Tamper"}}. This rule does not trigger.

sfromis, 2020-12-11:
As mentioned on Discord, my expectation is that the current intention with JSON triggers is only for individual keys, not for treating whole JSON objects as a string, which would be outside the concept of parsing JSON. An alternate enhancement request could be to detect the serial received string as JSON, and omit the wrapping.

theOzzieRat, 2020-12-11:
Sorry, but I have to disagree. Currently it does trigger on any  token in the json string sent over the serial line. So in the example I used above of:
```
{"Update":{"Channel":2,"Value":673,"State":"Alarm Tamper"}}
```
All of the following triggers actually work:
```
SerialReceived#Update
SerialReceived#Update#channel
SerialReceived#Update#value
SerialReceived#Update#state
```
The only problem is that the first one doesn't return the string of the object. It has been parsed by the jsmn library, but the code just doesn't seem to include the use case of returning a json object.

Jason2866, 2020-12-11:
Serialbridge is not designed to support json processing.

theOzzieRat, 2020-12-20:
It is not actually required to process any nested data, but just needs to return the part of the string that the rule specifies. A JSON object is still just a string, just like any of the other types.

Jason2866, 2020-12-20:
It is not supported so not a issue. You can close this issue and open a feature request.

closed 2021-01-01 by github-actions[bot]

19cool53, 2021-11-27:
I compiled the standard 10.0 version, but it still does not seem to work.
From a Terminal I sent  a string without cr or lf:
  {"Data":"set","sid":"TM154","MyCmd":"power1 toggle"} 
Tasmota received:
RESULT = {"SSerialReceived":" {\"Data\":\"set\",\"sid\":\"TM154\"} "}
( The backslashes were added after update from v9.4 to 10.0 )
However, a rule on SSerialReceived#sid=TM154 still does not work. Can anyone explain why?
Ist there a hidden (undocumented) option i missed?
Does the received data need to be json encoded separate?
What is the difference to a Sonoff Bridge (seem to have json data

Regards, cool

barbudor, 2021-11-27:
Please do not hijack an old issue and open a new discussion in https://github.com/arendst/Tasmota/discussions/