arendst/Tasmota #10110: Json error in zigbee bridge

ionciubotaru, 2020-12-12:
### PROBLEM DESCRIPTION
Json error - comma missing between `DeviceVersion":0"InClusters":`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff zigbee bridge
- [ x] Tasmota binary firmware version number used: 9.1.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

12:53:03 MQT: stat/zb_bridge/RESULT = {"NAME":"Sonoff ZHABridge","GPIO":[320,5088,0,5120,323,322,0,0,0,0,0,0,32,0],"FLAG":0,"BASE":18}
12:53:03 MQT: stat/zb_bridge/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
12:53:03 MQT: stat/zb_bridge/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```
12:44:20 CMD: status 0
12:44:20 MQT: stat/zb_bridge/STATUS = {"Status":{"Module":75,"DeviceName":"zb_bridge","FriendlyName":["zb_bridge"],"Topic":"zb_bridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":0,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:44:20 MQT: stat/zb_bridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"8T03:53:47","StartupUTC":"2020-12-04T06:50:33","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2020-11-27T14:20:31","SaveCount":40,"SaveAddress":"F8000"}}
12:44:20 MQT: stat/zb_bridge/STATUS2 = {"StatusFWR":{"Version":"9.1.0(zbbridge)","BuildDateTime":"2020-11-07T12:01:19","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"380/699"}}
12:44:20 MQT: stat/zb_bridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AVrpi","vlaicuD"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800A","2805C8000100060000005A00000000000000","00008000","00006002","00000000"]}}
12:44:20 MQT: stat/zb_bridge/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,23,41","Sensors":""}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
12:57:45 MQT: tele/zb_bridge/RESULT = {"ZbState":{"Status":33,"Endpoint":"0x01","ProfileId":"0x0104","DeviceId":"0x040A","DeviceVersion":0"InClusters":["0x0000","0x0001","0x0003","0x0500","0x0B05"],"OutClusters":["0x0019"]}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-12-12:
Please, update to latest and try again.

ionciubotaru, 2020-12-12:
I have the latest stable version Tasmota v9.1.0 Imogen
Unfortunately I don't know how to compile Tasmota from source, so I will wait for the next stable release.

sfromis, 2020-12-12:
No need to compile, you can get latest precompiled from http://ota.tasmota.com/tasmota/

ionciubotaru, 2020-12-12:
Problem solved in version 9.1.0.2

closed 2020-12-12 by ionciubotaru