arendst/Tasmota #10112: Delay at ShutterStop / wrong position after several position commands

Bascht74, 2020-12-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hello, 
I run the actual version of Tasmota together with Shelly 2.5's to control German roller shutters. The roller shutter motors are from Somfy, so widely used.
I have installed the current version and calibrated the roller shutters cleanly: as soon as they are up or down, the Shelly/Tasmota stops the power supply at that exact moment. Great!

However, I have a problem: 
If I control the shutters several times with ShutterPosition (e.g. at position 10, 20, 30, 40, 50, 60, 70, 80), the shutters are already at position 80 allready at the top. 
In this respect, there seems to be a slight overrun, the motors move a little further after the power supply has been capped (This means that this entry would be a feature request). 

Or it could be that Tasmota delivers the stop command with a delay? 
What I notice is that the `ShutterStop` command, when I issue it e.g. as a web command / console command, always takes a moment to stop the motor (approx. 0.5 seconds), while the `shutterOpen` / `ShutterClose` command is executed almost immediately (in this case the entry might be a bug?). 
I have tried all of this 20-30 times and it always behaves the same: `shutterOpen` / `ShutterClose` -> immediately, `ShutterStop` -> delayed by about 0.5 seconds.

This behavior happens with both upward and downward movements.
This means that the positions are no longer correct if positions between 10-90 are used several times; an end position must first be approached, after which it is correct again for a while.

In principle, this is not a problem with Tasmota, but I would like to correct this behavior. 
There is a parameter `ShutterMotorDelay` for this purpose. 
I would like to use it for this, but at the moment it only allows positive values (according to the description). If I enter a negative value, the input is accepted, but the value is subtracted from the maximum length.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X]  Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly2.5
- [X] Tasmota binary firmware version number used: Tasmota v9.1.0 Imogen
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: tasmotizer-1.2.exe
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
12:10:54 CMD: Backlog Rule1; Rule2; Rule3
12:10:55 MQT: stat/tasmota_FB1089/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
12:10:55 MQT: stat/tasmota_FB1089/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
12:10:55 MQT: stat/tasmota_FB1089/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
  Rules output here:
no rules

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

12:08:08 CMD: status 0
12:08:08 MQT: stat/tasmota_FB1089/STATUS = {"Status":{"Module":0,"DeviceName":"ShellySwitch25-Rollo.EG.Wohnzimmer.Terrasse","FriendlyName":["Rollo.EG.Wohnzimmer.Terrasse","Tasmota2"],"Topic":"tasmota_FB1089","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:08:08 MQT: stat/tasmota_FB1089/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz","RestartReason":"External System","Uptime":"0T10:55:55","StartupUTC":"2020-12-12T00:12:13","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2020-12-09T10:40:56","SaveCount":142,"SaveAddress":"FA000"}}
12:08:08 MQT: stat/tasmota_FB1089/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2020.11.07 12:04:01","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"589/699"}}
12:08:08 MQT: stat/tasmota_FB1089/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Qar5tz09e589mn05a0",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A00000000000000","400000C0","00006000","00000000"]}}
12:08:08 MQT: stat/tasmota_FB1089/STATUS4 = {"StatusMEM":{"ProgramSize":601,"Free":400,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
12:08:08 MQT: stat/tasmota_FB1089/STATUS5 = {"StatusNET":{"Hostname":"Rollo.EG.Wohnzimmer.Terrasse","IPAddress":"192.168.175.42","Gateway":"192.168.175.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.175.1","Mac":"68:C6:3A:FB:10:89","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:08:08 MQT: stat/tasmota_FB1089/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.175.214","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FB1089","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:08:08 MQT: stat/tasmota_FB1089/STATUS7 = {"StatusTIM":{"UTC":"2020-12-12T11:08:08","Local":"2020-12-12T12:08:08","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:34","Sunset":"16:52"}}
12:08:08 MQT: stat/tasmota_FB1089/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:08:08 MQT: stat/tasmota_FB1089/STATUS10 = {"StatusSNS":{"Time":"2020-12-12T12:08:08","ANALOG":{"Temperature":60.4},"ENERGY":{"TotalStartTime":"2020-12-09T09:59:34","Total":0.023,"Yesterday":0.002,"Today":0.020,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":100,"Direction":0,"Target":100},"TempUnit":"C"}}
12:08:08 MQT: stat/tasmota_FB1089/STATUS11 = {"StatusSTS":{"Time":"2020-12-12T12:08:08","Uptime":"0T10:55:55","UptimeSec":39355,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Qar5tz09e589mn05a0","BSSId":"E0:63:DA:53:3C:3E","Channel":11,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:11"}}}
12:08:08 MQT: stat/tasmota_FB1089/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":300,"Close":280,"50perc":65,"Delay":236,"Opt":"0000","Calib":[1800,420,4020,900,1000],"Mode":"0"}}}

```
[... 208 lines cut ...]
(Maybe a different processing is the cause here? If with shutteropen/close the "relay" "on" command is triggered first and then all sorts of other things happen, and with shutterstop mqtt and other things are done first and the relay is set to "off" last, this could explain it. Especially the significantly longer runtime of shutterstop in connection with webdebug 4 indicates that the internal processing of the commands seems to take place in a different order).

2. 
Alternatively, a negative value for ShutterMotorDelay could be allowed, so that one can also take into account an overrun of the motor and not only a start-up time. However, if Tasmota is the cause of the delay, then this should of course be changed first, if possible. Then I can check again whether the problem is then "gone".

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

./.

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Many thanks for the great work to Tasmota. 
[... cut ...]
A lot of hard work has certainly gone into it and it shows. Of course, it is not so easy to get started because of the large scope, but I find everything quite well solved and the documentation is just great!

By the way, this is the first ticket I've given for Tasmota. 
I took a lot of trouble to observe everything and do it right. Please let me know if something is wrong or needs to be corrected. I will do it quickly. Likewise, if there is anything else I can try to narrow down the problem further!

Greetings

Sebastian

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2020-12-12:
Hi @Bascht74,
see #9906 
Joerg

ascillato2, 2020-12-12:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-12 by ascillato2

Bascht74, 2020-12-13:
Hi @schrej-zz , 

I saw #9906 but I don't think it is the same problem there. It has something to do with position calculation.

At #9906 there seems to be something wrong if you use the power commands, but I am using the `shutterclose/open` commands. They are working just fine and the position will be updated event after `shutterstop`. 
(I don't know why #9906 is using relay on/off commands as for shutters there are shutterclose/shutteropen commands that control the relays. I don't know if it is designed to work that way, even that it seems to word before)

But in my case when I use `shutterstop` the cover is not stopping right away and moving further (much further if weblog 4 enabled, but even with weblog 2 it moves 0.5s more thank it should). The final position is updated even before (!) the relay stops the motor...

What I found out via the logs above is that something must be wrong with the code:

Here the act.pos of the shutter is: 692106 (it is closing)
`12:03:45 SHT: time: 151, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 4, min_runtime_ms 0, act.pos 692106, next_stop 0, target: 0,  max_vel_change 0, dir: -1`

Then the shutterstop command comes in:
```
12:03:45 CMD: shutterstop
12:03:45 SRC: WebConsole from 192.168.175.172
12:03:45 CMD: Gruppe 0, Index 1, Befehl "SHUTTERSTOP", Daten ""
12:03:45 SHT: Stop moving 1: dir: -1
12:03:45 SHT: Pos. in: payload  (0), payload 121, idx 1, src 7

```
This seems to work that a target position ist set, which should be the same as the actual position, and here is the problem. The target should be the real position in this case to stop right away:
`12:03:45 SHT: lastsource 7:, real 691035, target 600000, payload 100   `

then the MQTT message is sent, that the Shutter should stop:
`12:03:45 MQT: stat/tasmota_FB1089/RESULT = {"ShutterStop1":100}
`
But the shutter is moving to the new target 600000, which is wrong:
`12:03:46 SHT: time: 174, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 4, min_runtime_ms 0, act.pos 667473, next_stop 0, target: 600000,  max_vel_change 0, dir: -1
`
and stops there at 600000: 
```
...
12:03:50 SHT: time: 236, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 4, min_runtime_ms 0, act.pos 601071, next_stop 0, target: 600000,  max_vel_change 0, dir: -1
12:03:50 SHT: time: 237, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 4, min_runtime_ms 0, act.pos 600000, next_stop 0, target: 600000,  max_vel_change 0, dir: -1
12:03:50 SHT: Stop Shutter 0. Switchmode 0
12:03:50 ADE: 6104446, 4473, [1773, -77, 644, -3], [45334, 15971, 16492, 4048]
12:03:50 ADE: U 6104446, C 4473, I 0 + 45334 = 45334, P 0 + 15971 = 15971
12:03:50 SRC: Shutter
12:03:50 SHT: Switched relay: 2 by Shutter
12:03:50 SHT: Shutter 1: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 2, manual change 0
***** 12:03:50 MQT: stat/tasmota_FB1089/RESULT = {"POWER2":"OFF"}
12:03:50 MQT: stat/tasmota_FB1089/POWER2 = OFF
12:03:50 SHT: Shutter 1 Real 600000, Start 600000, Stop 600000, Dir 0, Delay 236, Rtc 11.85 [s], Freq 0, PWM 0
```

As you can see, the position is updated...

I would like to look into that problem some more.

Could you reopen the issue it?

I found the code at `https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_27_shutter.ino`
and trying to figure out what is going on...

Sebastian

Bascht74, 2020-12-13:
Seems that at `CmndShutterStop` ...
`
int32_t temp_realpos = ShutterCalculatePosition(i);
        XdrvMailbox.payload = ShutterRealToPercentPosition(temp_realpos, i);
`

...is the place to start looking at.

It seems to have something todo with `Shutter[i].motordelay` and `Settings.shutter_set50percent[i] != 50`

Sebastian

schrej-zz, 2020-12-13:
@Bascht74,
please have also a look at the issues, that were closed by @ascillato2 and referred to in this issue.
I think, your issue is also covered by one of these issues.
I'm also not happy about the closing of my issue and refrerring to ths issue, but there relly seems to be a general issue with the changes, that @stefanbode implemented since version 8.3.1.

Bascht74, 2020-12-15:
Hello @schrej-zz 

I've found out some things now: 
1. I had unintentionally configured `ShutterMotorDelay`, which I now deleted. This led to the strong delay in my log above.
2. the wrong positioning seems to be handled in bug #10033, I'll give it a try.
3. the delay at `shutterstop` seems to be quite "normal". In the code I see that `shutterstop` determines the position and then also triggers `shutterposition` with this position. 

```
void CmndShutterStop(void)
{
  [...]
        int32_t temp_realpos = ShutterCalculatePosition(i);
        XdrvMailbox.payload = ShutterRealToPercentPosition(temp_realpos, i);
        TasmotaGlobal.last_source = SRC_WEBGUI;
        CmndShutterPosition();
      } [...]
}

```
Since it seems to work with percentage values (0-100), the determined percentage deviates from the real positioning, so there is a small lag. With low percentage values (<5%), this can apparently even lead to the motor moving a little in the other direction when it stops.