arendst/Tasmota #10116: Support GitHub Discussions?

Frogmore42, 2020-12-12:
**Have you looked for this feature in other issues and in the docs?**  
Yes, there are avenues for this that were created to keep discussions out of issues.  I guess they work, since I see far fewer discussions in issues now.  But, I also don't participate in the discussions/forums because they are not here.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No place to discuss things on GitHub.  Have to leave and signup for an account elsewhere to participate.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Enable GitHub discussions for this repo/project. 
https://docs.github.com/en/free-pro-team@latest/discussions/quickstart#

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
There are solutions for this today outside of GitHub, but nothing else in GitHub.

**Additional context**  
_Add any other context or screenshots about the feature request here._
Still in Beta, but looks like it is good.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-12-12:
Hi,

Your idea is good, but so far, what is actively mantained by the Tasmota community (and for free, just for fun) are:

* A Discord chat for support
* Github issues for bug reports.

There were several attempts in the past for forums but the main issue with them, is the need of maintenance. Most of those attempts have failed because it is a hard work and noone wants to do it for free. So, those have ended on being left behind and full of trolls. 

I personally don't think that is going to work by just using "another platform" (like github discussions in your proposal). To make that work, you should enter everyday to the forum, answers questions, kick trolls, etc etc. Opening it and just leave it isn't useful in my opinion.

ascillato, 2020-12-12:
Check in older issues like in https://github.com/arendst/Tasmota/issues/2059#issuecomment-415982348

ascillato2, 2020-12-13:
@Frogmore42 

Hi,

I have done some tests on the "GitHub Discussions" feature and it is very well integrated. So, may be this new platform could work. We will need your help and the help of others to make it work ( so as to avoid the most common complaint of _my question/problem is not answered_ ).

Let's make a poll, adding :+1: or :-1: in the first comment.

arendst, 2020-12-13:
I guess it makes finding solutions to earlier discussion easier than discord does?

ascillato2, 2020-12-13:
Cool, let's test it.

closed 2020-12-13 by ascillato2