arendst/Tasmota #10119: RFRecv not capturing signal correctly

bilogic, 2020-12-13:
### PROBLEM DESCRIPTION
I have a 32bit 433Mhz RF signal that is not being recognized by `RFRecv`
Here is a sample of the received signal
```
15:27:26 MQT: tele/tasmota_158C3C/RESULT = {"Time":"2020-12-13T15:27:26","RfReceived":{"Data":"0xA19","Bits":15,"Protocol":5,"Pulse":621}}
```
My hardware works with `RC-Switch`, which is why I'm reporting the issue here.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 9.1.0(sensors)
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
15:36:25 MQT: stat/tasmota_158C3C/RESULT = {"NAME":"RF433+RGB","GPIO":[1152,1,1120,1,1376,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:36:25 MQT: stat/tasmota_158C3C/RESULT = {"Module":{"0":"RF433+RGB"}}
15:36:26 MQT: stat/tasmota_158C3C/RESULT = {"GPIO0":{"1152":"RFrecv"},"GPIO1":{"0":"None"},"GPIO2":{"1120":"RFSend"},"GPIO3":{"0":"None"},"GPIO4":{"1376":"WS2812"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
15:32:03 MQT: stat/tasmota_158C3C/STATUS = {"Status":{"Module":0,"DeviceName":"rfrgb","FriendlyName":["RGB Light"],"Topic":"tasmota_158C3C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:32:03 MQT: stat/tasmota_158C3C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:04:51","StartupUTC":"2020-12-13T07:27:12","Sleep":50,"CfgHolder":4617,"BootCount":42,"BCResetTime":"2020-12-12T15:16:49","SaveCount":123,"SaveAddress":"F9000"}}
15:32:03 MQT: stat/tasmota_158C3C/STATUS2 = {"StatusFWR":{"Version":"9.1.0(sensors)","BuildDateTime":"2020-11-07T11:59:31","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"394/699"}}
15:32:03 MQT: stat/tasmota_158C3C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["rtfckut",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
15:32:04 MQT: stat/tasmota_158C3C/STATUS4 = {"StatusMEM":{"ProgramSize":632,"Free":368,"Heap":17,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1625C2","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
15:32:04 MQT: stat/tasmota_158C3C/STATUS5 = {"StatusNET":{"Hostname":"tasmota_158C3C-3132","IPAddress":"192.168.9.156","Gateway":"192.168.9.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.9.1","Mac":"B4:E6:2D:15:8C:3C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:32:04 MQT: stat/tasmota_158C3C/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.214.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_158C3C","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:32:04 MQT: stat/tasmota_158C3C/STATUS7 = {"StatusTIM":{"UTC":"2020-12-13T07:32:04","Local":"2020-12-13T15:32:04","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+08:00","Sunrise":"15:35","Sunset":"23:52"}}
15:32:04 MQT: stat/tasmota_158C3C/STATUS10 = {"StatusSNS":{"Time":"2020-12-13T15:32:04"}}
15:32:04 MQT: stat/tasmota_158C3C/STATUS11 = {"StatusSTS":{"Time":"2020-12-13T15:32:04","Uptime":"0T00:04:52","UptimeSec":292,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":100,"Color":"73FFE1","HSBColor":"167,55,100","Channel":[45,100,88],"Scheme":2,"Width":1,"Fade":"OFF","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"rtfckut","BSSId":"78:24:AF:E4:F8:F0","Channel":6,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
15:27:26 MQT: tele/tasmota_158C3C/RESULT = {"Time":"2020-12-13T15:27:26","RfReceived":{"Data":"0xA19","Bits":15,"Protocol":5,"Pulse":621}}
```

### TO REPRODUCE
1. Flash with `sonoff-sensor`
2. Transmit any 32bit 433Mhz RF signal
3. See the error output on console of tasmota

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expected Bits to be 32 and Protocol to be 1

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
Are there any parameters available to config RC switch?

**(Please, remember to close the issue when the problem has been addressed)**

bilogic, 2020-12-13:
I have `setReceiveTolerance()` set to 70, is this configurable in tasmota?

Jason2866, 2020-12-13:
Try latest development version and disable not needed protocols.
The rf433 rc-switch code is not active maintained. If you find a solution please provide a PR.

bilogic, 2020-12-13:
@Jason2866 ok, I will. But could you help me out a bit here. I cloned this repo and tried to compile in VSCode with platformio. What it did was compiled every iteration of firmware (>7GB), how do I work on just `tasmota-sensor.bin`? Or more specifically, what do I have to edit in `platformio.ini`? Thank you.

bilogic, 2020-12-13:
@Jason2866 I tried dev 9.1.0.2, `RfProtocol` is an unknown command.

Jason2866, 2020-12-13:
@bilogic  rename `Platformio_override_sample.ini` to `platformio_override.ini`
For customizing a build the base is ALWAYS build Tasmota. Edit user_config_override.h for your needs. See https://tasmota.github.io/docs/Compile-your-build/#customize-your-build

bilogic, 2020-12-13:
@Jason2866 I did an OTA of http://ota.tasmota.com/tasmota/tasmota-sensors.bin.gz, but `RfProtocol` is an unknown command.

bilogic, 2020-12-13:
@Jason2866 I also tried compiling both `default_envs = tasmota` and `default_envs = tasmota-sensors.bin`, both don't accept rfprotocol.

```
18:54:07 CMD: rfprotocol
18:54:07 MQT: stat/tasmota_158C3C/RESULT = {"Command":"Unknown"}
18:54:10 CMD: rfprotocol1
18:54:10 MQT: stat/tasmota_158C3C/RESULT = {"Command":"Unknown"}
```

It's bed time now, will try again when I wake.

bilogic, 2020-12-14:
@Jason2866 

I did the following:

1. Renamed `Platformio_override_sample.ini` to `platformio_override.ini` 
2. Added `#define USE_RC_SWITCH` in `user_config_override.h`
3. Went under `end:tasmota-sensors` to `Upload and Monitor`, `rfprotocol` unknown
4. Went under `end:tasmota` to `Upload and Monitor`, `rfprotocol` unknown
5. Added to `user_config_override.h` and under `Default`, `Upload and Monitor`, `rfprotocol` unknown
```
#ifndef USE_RC_SWITCH
#define USE_RC_SWITCH
#endif
```

I'm just not sure what I'm missing here, `my xdrv_17_rcswitch.ino` just keeps showing code that is "greyed" out.

bilogic, 2020-12-14:
I placed `USE_RC_SWITCH` literally everywhere, `user_config_override.h`, `my_user_config.h`, and even in `xdrv_17_rcswitch.ino` and compiling for both `env:tasmota` and `env:tasmota-sensors`, but `rfprotocol` and `rfsend` remain unknown on the console.

bilogic, 2020-12-14:
Ok, found the issue. Tasmota GPIOs have to be configured with `RFSend`/`RFRecv` before these commands will work. Testing now.

bilogic, 2020-12-14:
> Try latest development version and disable not needed protocols.
> The rf433 rc-switch code is not active maintained. If you find a solution please provide a PR.
> See [#10063 (comment)](https://github.com/arendst/Tasmota/pull/10063#issuecomment-740137505)

I disabled all but protocol 1 and it worked. Thank you.

closed 2020-12-14 by bilogic