arendst/Tasmota #10122: Tasmota ADC Value Range in Rule

kbroeker, 2020-12-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When setup the Tasmota ESP32 Software (Tasmota 9.1.0) and add a Rule to publish an Mqtt Message to the MQTT Broker on Value Change of an ADC the content of the %value% variable goes from 0 to 100. In this range the ADC ticks from 0 to 1000 and The Voltage at the ADC goes from 0 to 0.8V. 

My whole Ruhle is:
Rule1 ON analog#a1div10 DO publish stat/ESP32_1/analog/RESULT { "POWER1":"%value%"} ENDON 
    ON analog#a2div10 DO publish stat/ESP32_1/analog/RESULT { "POWER2":"%value%"} ENDON

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 9.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Setup the ESP32 with [https://tasmota.github.io/docs/ESP32/](https://tasmota.github.io/docs/ESP32/)
Adding an ADC on Pin 32
Put the middle pin of a potentiometer to the adc and the other two Pins to GND and 3V3
Setup the Rule 
Rule1 ON analog#a1div10 DO publish stat/ESP32_1/analog/RESULT { "POWER1":"%value%"} ENDON 
Then turn the potentiometer an read the MQTT messages

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The Content of the %value% variable should go from 0 to 100 in a Range from 0 to 3.3V. Or the content should go from 0 to 4096.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-12-13:
> The Content of the %value% variable should go from 0 to 100 in a Range from 0 to 3.3V. Or the content should go from 0 to 4096.

Hi,

That type of configuration is already implemented. Check the command `AdcParam` in the docs at https://tasmota.github.io/docs/Commands/#sensors

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-13 by ascillato2

lunek1234, 2020-12-13:
Hey, 
First thanks for the fast answer.
I allready read the documentation about ADCParam. But i thought thats not the right thing for me to do. In the documentation is written under "6 = ADC linear range remapping parameters" "The range remapping perform the following calculation on the ADC value [0..1023]". But on the ESP32 the Value Range ist [0.. 4095]. 
Now i tried to do this range remapping with the statement:
``` 
adcparam1 6,0,4095,0,3300
```
```
MQT: stat/Tasmota/ESP32_1/RESULT = {"AdcParam1":[6,0,4095,0,3300]}
```
(for a remapping to mV values)
After executing this statement the value in the tasmota MainDisplay ist remapped. But then my rules doesnt send mqtt messages anymore. After a device restart the esp32 starts the wifi manager and the settings in ADCparam1 are deletet.
So the Statement
```
adcparam1
```
lead to an answer of
```
MQT: stat/Tasmota/ESP32_1/RESULT = {"AdcParam1":[1,0,4095,0]}
```

What am i doing wrong?

Best Regards