arendst/Tasmota #10123: MCU based Garage Door Open losses MCU connecting after some time.

vajonam, 2020-12-13:
### PROBLEM DESCRIPTION
MCU based Garage Door Open losses MCU connecting after some time. It works correctly but after a while it drops connection to the MCU. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): WOFEA Garage Door Opener - Model 88
- [x] Tasmota binary firmware version number used: OTA 9.1.0.2 (reset flash)
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used:I ntially TuyaConvert
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

08:38:18 MQT: stat/garagedoor_01/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
08:38:18 MQT: stat/garagedoor_01/RESULT = {"Module":{"54":"Tuya MCU"}}
08:38:19 MQT: stat/garagedoor_01/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
08:38:56 MQT: stat/garagedoor_01/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":137,"Free":374,"Rules":"on switch2#state=1 do publish2 stat/garagedoor_01/status Open endon on switch2#state=0 do publish2 stat/garagedoor_01/status Closed endon"}
08:38:57 MQT: stat/garagedoor_01/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
08:38:57 MQT: stat/garagedoor_01/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
08:39:18 MQT: stat/garagedoor_01/STATUS = {"Status":{"Module":54,"DeviceName":"garagedoor_01","FriendlyName":["garagedoor_01"],"Topic":"garagedoor_01","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,1,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:39:18 MQT: stat/garagedoor_01/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:46:16","StartupUTC":"2020-12-13T12:53:02","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2020-12-13T13:48:34","SaveCount":45,"SaveAddress":"F7000"}}
08:39:18 MQT: stat/garagedoor_01/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020-12-13T08:46:27","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"415/699"}}
08:39:18 MQT: stat/garagedoor_01/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["obfus-hw",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
08:39:18 MQT: stat/garagedoor_01/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
08:39:18 MQT: stat/garagedoor_01/STATUS5 = {"StatusNET":{"Hostname":"garagedoor_01-1643","IPAddress":"10.1.1.156","Gateway":"10.1.1.253","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.253","Mac":"D8:F1:5B:A3:E6:6B","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:39:18 MQT: stat/garagedoor_01/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.mydomain.com","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A3E66B","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:39:18 MQT: stat/garagedoor_01/STATUS7 = {"StatusTIM":{"UTC":"2020-12-13T13:39:18","Local":"2020-12-13T08:39:18","StartDST":"2020-03-08T02:00:00","EndDST":"2020-11-01T02:00:00","Timezone":99,"Sunrise":"02:35","Sunset":"10:52"}}
08:39:18 MQT: stat/garagedoor_01/STATUS10 = {"StatusSNS":{"Time":"2020-12-13T08:39:18"}}
08:39:18 MQT: stat/garagedoor_01/STATUS11 = {"StatusSTS":{"Time":"2020-12-13T08:39:18","Uptime":"0T00:46:16","UptimeSec":2776,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"obfus-hw","BSSId":"92:2A:A8:1A:31:3D","Channel":3,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

Dec 12 03:26:20 garagedoor_01 garagedoor_01 ESP-TYA: fnId=0 is set for dpId=7
Dec 12 03:26:20 garagedoor_01 garagedoor_01 ESP-TYA: RX value 0 from dpId 7
Dec 12 03:26:21 garagedoor_01 garagedoor_01 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 12 03:26:21 garagedoor_01 garagedoor_01 ESP-TYA: fnId=0 is set for dpId=7
Dec 12 03:26:21 garagedoor_01 garagedoor_01 ESP-TYA: RX value 0 from dpId 7
Dec 12 03:26:22 garagedoor_01 garagedoor_01 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 12 03:26:22 garagedoor_01 garagedoor_01 ESP-TYA: fnId=0 is set for dpId=7
Dec 12 03:26:22 garagedoor_01 garagedoor_01 ESP-TYA: RX value 0 from dpId 7
Dec 12 03:26:23 garagedoor_01 garagedoor_01 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 12 03:26:23 garagedoor_01 garagedoor_01 ESP-TYA: fnId=0 is set for dpId=7
Dec 12 03:26:23 garagedoor_01 garagedoor_01 ESP-TYA: RX value 0 from dpId 7
Dec 12 03:26:24 garagedoor_01 garagedoor_01 ESP-TYA: Send "55aa00000000ff"
Dec 12 03:26:24 garagedoor_01 garagedoor_01 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 12 03:26:24 garagedoor_01 garagedoor_01 ESP-TYA: fnId=0 is set for dpId=7
Dec 12 03:26:24 garagedoor_01 garagedoor_01 ESP-TYA: RX value 0 from dpId 7
Dec 12 03:26:33 garagedoor_01 garagedoor_01 ESP-WIF: Checking connection...
Dec 12 03:26:35 garagedoor_01 garagedoor_01 ESP-TYA: Send "55aa00000000ff"
Dec 12 03:26:46 garagedoor_01 garagedoor_01 ESP-TYA: Send "55aa00000000ff"
Dec 12 03:26:53 garagedoor_01 garagedoor_01 ESP-WIF: Checking connection...
Dec 12 03:26:57 garagedoor_01 garagedoor_01 ESP-TYA: Send "55aa00000000ff"
Dec 12 03:27:08 garagedoor_01 garagedoor_01 ESP-TYA: Send "55aa00000000ff"
Dec 12 03:27:13 garagedoor_01 garagedoor_01 ESP-WIF: Checking connection...
Dec 12 03:27:19 garagedoor_01 garagedoor_01 ESP-TYA: Send "55aa00000000ff"
Dec 12 03:27:30 garagedoor_01 garagedoor_01 ESP-TYA: Send "55aa00000000ff"
Dec 12 03:27:33 garagedoor_01 garagedoor_01 ESP-WIF: Checking connection...
Dec 12 03:27:41 garagedoor_01 garagedoor_01 ESP-TYA: Send "55aa00000000ff"
Dec 12 03:27:52 garagedoor_01 garagedoor_01 ESP-TYA: Send "55aa00000000ff"
Dec 12 03:27:53 garagedoor_01 garagedoor_01 ESP-WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Just leave it running. Only way to get it running again is a power cycle. 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

MCU stays connected and responds to heartbeat 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
None. 

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

#8408 - seems to be the same behavior as this.. do I am now thinking its not hardware related. 

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-12-14:
Tasmota try to connect to the MCU trough serial every 10 seconds and is an hard coded function. As far as I can see your device simply stop responding after a while and that could means a defective unit or some strange Tuya internal function I'm not aware of.  
Try to set `SerialLog` to `4` and add a rule forcing the heartbeat to ~5 seconds to see if it improve: 
```py
ON system#boot DO ruletimer1 5 ENDON ON rules#timer=1 DO backlog serialsend5 55aa00000000ff; ruletimer1 5 ENDON
```
but please note even if is working I'll not change the heartbeat timing or it could cause issue with device with several dpid.

vajonam, 2020-12-14:
That commands looks like its missing a "#". 

```
11:21 CMD: Group 0, Index 1, Command "ON", Data "system#boot DO ruletimer1 5 ENDON ON rules#timer=1 DO backlog serialsend5 55aa00000000ff; ruletimer1 5 ENDON"
11:11:21 MQT: stat/garagedoor_01/RESULT = {"Command":"Unknown"}
```

vajonam, 2020-12-14:
Sorry. was a bit dense. :-) got the rule running. Will provide feedback shortly

vajonam, 2020-12-14:
@effelle 

> but please note even if is working I'll not change the heartbeat timing or it could cause issue with device with several dpid.

The downside of sending the serial heartbeat command more often is that we are likely to miss a response from the reed-switch that is on dpID=101 is that correct?

effelle, 2020-12-14:
Yes, there is the possibility to lost some packet between two heartbeat. Keep an eye on it.

vajonam, 2020-12-14:
I a different brand of garage door opener arrived today, i have it running in the same environment (cold in the garage). the new one doesn't use extra heartbeat message. will keep an eye on both and report back.

vajonam, 2020-12-14:
So with the opener-01 (the one with mcu disconnect issue)

I see repeated. 
```
Dec 14 14:27:15 172.17.0.1 garagedoor_01-1643 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 14 14:27:15 172.17.0.1 garagedoor_01-1643 ESP-TYA: fnId=0 is set 
```

What is the MCU sending back repeatedly? 

However with new opener-02 (which just arrived in the mail) 

I only see

```
20:28:32 TYA: Send "55aa00000000ff"
20:28:32 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
20:28:32 TYA: Heartbeat
20:28:39 WIF: Checking connection...
20:28:43 TYA: Send "55aa00000000ff"
20:28:43 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
```

vajonam, 2020-12-14:
Here a log snippet for mcu/opener-01, should I assign a function to dpId=7 ? 

```
Dec 14 14:32:30 172.17.0.1 garagedoor_01-1643 ESP-RUL: RULES#TIMER=1 performs "backlog serialsend5 55aa00000000ff; ruletimer1 5"
Dec 14 14:32:30 172.17.0.1 garagedoor_01-1643 ESP-SRC: Rule
Dec 14 14:32:30 172.17.0.1 garagedoor_01-1643 ESP-CMD: Group 0, Index 1, Command "BACKLOG", Data "serialsend5 55aa00000000ff; ruletimer1 5"
Dec 14 14:32:30 172.17.0.1 garagedoor_01-1643 ESP-SRC: Backlog
Dec 14 14:32:30 172.17.0.1 garagedoor_01-1643 ESP-CMD: Group 0, Index 5, Command "SERIALSEND", Data "55aa00000000ff"
Dec 14 14:32:30 172.17.0.1 garagedoor_01-1643 ESP-MQT: stat/garagedoor_01/RESULT = {"SerialSend":"Done"}
Dec 14 14:32:30 172.17.0.1 garagedoor_01-1643 ESP-{"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
Dec 14 14:32:30 172.17.0.1 garagedoor_01-1643 ESP-TYA: Heartbeat
Dec 14 14:32:30 172.17.0.1 garagedoor_01-1643 ESP-SRC: Backlog
Dec 14 14:32:30 172.17.0.1 garagedoor_01-1643 ESP-CMD: Group 0, Index 1, Command "RULETIMER", Data "5"
Dec 14 14:32:30 172.17.0.1 garagedoor_01-1643 ESP-MQT: stat/garagedoor_01/RESULT = {"T1":5,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
Dec 14 14:32:31 172.17.0.1 garagedoor_01-1643 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 14 14:32:31 172.17.0.1 garagedoor_01-1643 ESP-TYA: fnId=0 is set for dpId=7
Dec 14 14:32:31 172.17.0.1 garagedoor_01-1643 ESP-TYA: RX value 0 from dpId 7 
Dec 14 14:32:32 172.17.0.1 garagedoor_01-1643 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 14 14:32:32 172.17.0.1 garagedoor_01-1643 ESP-TYA: fnId=0 is set for dpId=7
Dec 14 14:32:32 172.17.0.1 garagedoor_01-1643 ESP-TYA: RX value 0 from dpId 7 
Dec 14 14:32:33 172.17.0.1 garagedoor_01-1643 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 14 14:32:33 172.17.0.1 garagedoor_01-1643 ESP-TYA: fnId=0 is set for dpId=7
Dec 14 14:32:33 172.17.0.1 garagedoor_01-1643 ESP-TYA: RX value 0 from dpId 7 
Dec 14 14:32:34 172.17.0.1 garagedoor_01-1643 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 14 14:32:34 172.17.0.1 garagedoor_01-1643 ESP-TYA: fnId=0 is set for dpId=7
Dec 14 14:32:34 172.17.0.1 garagedoor_01-1643 ESP-TYA: RX value 0 from dpId 7 
Dec 14 14:32:34 172.17.0.1 garagedoor_01-1643 ESP-TYA: Send "55aa00000000ff"
Dec 14 14:32:34 172.17.0.1 garagedoor_01-1643 ESP-{"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
Dec 14 14:32:34 172.17.0.1 garagedoor_01-1643 ESP-TYA: Heartbeat
Dec 14 14:32:35 172.17.0.1 garagedoor_01-1643 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 14 14:32:35 172.17.0.1 garagedoor_01-1643 ESP-TYA: fnId=0 is set for dpId=7
Dec 14 14:32:35 172.17.0.1 garagedoor_01-1643 ESP-TYA: RX value 0 from dpId 7 
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-TYA: fnId=0 is set for dpId=7
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-TYA: RX value 0 from dpId 7 
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-RUL: RULES#TIMER=1 performs "backlog serialsend5 55aa00000000ff; ruletimer1 5"
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-SRC: Rule
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-CMD: Group 0, Index 1, Command "BACKLOG", Data "serialsend5 55aa00000000ff; ruletimer1 5"
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-SRC: Backlog
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-CMD: Group 0, Index 5, Command "SERIALSEND", Data "55aa00000000ff"
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-MQT: stat/garagedoor_01/RESULT = {"SerialSend":"Done"}
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-{"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-TYA: Heartbeat
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-SRC: Backlog
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-CMD: Group 0, Index 1, Command "RULETIMER", Data "5"
Dec 14 14:32:36 172.17.0.1 garagedoor_01-1643 ESP-MQT: stat/garagedoor_01/RESULT = {"T1":5,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
Dec 14 14:32:37 172.17.0.1 garagedoor_01-1643 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 14 14:32:37 172.17.0.1 garagedoor_01-1643 ESP-TYA: fnId=0 is set for dpId=7
Dec 14 14:32:37 172.17.0.1 garagedoor_01-1643 ESP-TYA: RX value 0 from dpId 7 
Dec 14 14:32:38 172.17.0.1 garagedoor_01-1643 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 14 14:32:38 172.17.0.1 garagedoor_01-1643 ESP-TYA: fnId=0 is set for dpId=7
[... cut ...]
Dec 14 14:32:57 172.17.0.1 garagedoor_01-1643 ESP-TYA: fnId=0 is set for dpId=7
Dec 14 14:32:57 172.17.0.1 garagedoor_01-1643 ESP-TYA: RX value 0 from dpId 7 
Dec 14 14:32:58 172.17.0.1 garagedoor_01-1643 ESP-{"TuyaReceived":{"Data":"55AA0307000807020004000000001E","Cmnd":7,"CmndData":"0702000400000000","DpType2Id7":0,"7":{"DpId":7,"DpIdType":2,"DpIdData":"00000000"}}}
Dec 14 14:32:58 172.17.0.1 garagedoor_01-1643 ESP-TYA: fnId=0 is set for dpId=7
Dec 14 14:32:58 172.17.0.1 garagedoor_01-1643 ESP-TYA: RX value 0 from dpId 7 
```

effelle, 2020-12-14:
Adding function is not mandatory, you could manage a device using serialsend5 without adding a single dpid.

Tasmota is receiving a packet from the MCU telling the type2 dpid 7 is sending a 0 value. 
You need to discover what is the dpid to understand why is sending it out continuously.

vajonam, 2020-12-14:
I have listed logs for many instances over many days. the `"DpIdData":"00000000"` never chages..

vajonam, 2020-12-14:
I think the old hardware in indeed defective or flashed incorrectly. The new device doesn't do that constant send of data from the MCU closing this issue for now Thanks for your pointers!

closed 2020-12-14 by vajonam

effelle, 2020-12-14:
Try to do a `reset 1` and then flash it again with the latest dev or master. Perhaps you will be lucky enough and fix it and will costs you nothing more than some minutes... 
Good luck!

vajonam, 2020-12-14:
Already did that once. Will try it again.

vajonam, 2020-12-15:
@effelle Looks like the extra heartbeat helps. My theory is that because the MCU is constantly sending back information, if that happens to coincide with a usual 10-second HB packet, that gets missed and MCU shuts down, adding some HB packets make sure that we are withing the second window. So far no missed dpId 101 events either.. will keep monitoring

effelle, 2020-12-15:
Good!
Still, you need to understand what is dpid 7 and why it is continuously reporting. What's the brand and model? 
And what's the result of `TuyaSend8`?

vajonam, 2020-12-15:
TuyaSend8 - response below.

```
07:18:11 CMD: Group 0, Index 8, Command "TUYASEND", Data ""
07:18:11 TYA: Read MCU state
07:18:11 TYA: Send "55aa0001000000"
07:18:11 MQT: stat/tasmota_A3E66B/RESULT = {"TuyaSend":"Done"}
07:18:11 {"TuyaReceived":{"Data":"55AA0301002A7B2270223A226231756B346674647561716F70787631222C2276223A22312E302E30222C226D223A307D19","Cmnd":1,"CmndData":"7B2270223A226231756B346674647561716F70787631222C2276223A22312E302E30222C226D223A307D"}}
07:18:11 TYA: MCU Product ID: {"p":"b1uk4ftduaqopxv1","v":"1.0.0","m":0}
07:18:11 TYA: Send "55aa0002000001"
07:18:11 {"TuyaReceived":{"Data":"55AA0302000004","Cmnd":2}}
07:18:11 TYA: RX MCU configuration Mode=0
07:18:11 TYA: Read MCU state
07:18:11 TYA: Send "55aa0008000007"
```

This is WOFEA WG-088. The dpId 7, seem to send integrer values when the reed-switch is tripped it goes from 5, 2, 1 0

vajonam, 2020-12-15:
Almost seems like a countdown timer to "toggle" the relay.. 

If set the dpid 7 to fnid=21 (integer) it can see values in the log on of non zero.  othewise it never shows non-zero values in the logging that could be a logging issue.

effelle, 2020-12-15:
https://github.com/arendst/Tasmota/issues/6319

vajonam, 2020-12-15:
Yes. Only one previous incident of mcu losing contact. But no solutions. This is as close as I have gotten.

vajonam, 2020-12-15:
@effelle I think I have figured it out. dpid 7 ,is a integer value that triggers a countdown timer. So if I send (using TuyaMCU bookmarklet), 5 it counts down from 5 (updating the count, 4, 3, 2, 1 etc) and then triggers the relay so I take it this is how the stock firmware toggled the relay.

effelle, 2020-12-15:
Then if is a timer you could also configure it. Have a look to my last edit on dev channel #10004 .

vajonam, 2020-12-15:
@effelle Nice! `Backlog TuyaMCU 11,1; TuyaMCU 81,7; TuyaMCU 2,101;` that works to configure the timer, I guess I can set a 20 minute timer on open and then close the garage door after that or something like that. 

I suspect the MCU that send back the timer status might also have the same issue of a missed heartbeat due to a timer update, if the timing is exact. What are your thoughts on making the TuyaMCU Heartbeat configurable? Something like `TuyaMCU 5` or a special fnID for heatbeat?

effelle, 2020-12-15:
No, sorry, as I said the heartbeat must complain with the Tuya protocol (10 seconds) to avoid issues. Your only solution is a rule. 
Let me know if the timer is working fine.

vajonam, 2020-12-15:
I am not sure how to set the timer from the WebUI (readonly timer1), but 
`TuyaSend2 7,60` starts a 60 second timer and then triggers relay. So that is working.

effelle, 2020-12-15:
You can set it just on console.

vajonam, 2020-12-15:
It worked for about 13 hours, but then just stopped again so the heartbeat didn't fix it.. think my hardware is got some issue. anyhow at least we found the timer function for dpid=7

bshm, 2021-04-28:
@vajonam @effelle 
I experienced the same behavior with the same garage opener, I actually thought about rewiring the MCU reset pin to one of the GPIOs to be able to recover the MCU without disconnecting the power. But it turned out that simply disabling the heartbeat sending made the problem go away.

```
diff --git a/tasmota/xdrv_16_tuyamcu.ino b/tasmota/xdrv_16_tuyamcu.ino
index b3586909d..2fabbd4c8 100644
--- a/tasmota/xdrv_16_tuyamcu.ino
+++ b/tasmota/xdrv_16_tuyamcu.ino
@@ -71,7 +71,9 @@ struct TUYA {
   uint8_t cmd_checksum = 0;               // Checksum of tuya command
   uint8_t data_len = 0;                   // Data lenght of command
   uint8_t wifi_state = -2;                // Keep MCU wifi-status in sync with WifiState()
+#if 0
   uint8_t heartbeat_timer = 0;            // 10 second heartbeat timer for tuya module
+#endif  
 #ifdef USE_ENERGY_SENSOR
   uint32_t lastPowerCheckTime = 0;        // Time when last power was checked
 #endif // USE_ENERGY_SENSOR
@@ -1110,7 +1112,9 @@ void TuyaInit(void)
 
     }
   }
+#if 0  
   Tuya.heartbeat_timer = 0; // init heartbeat timer when dimmer init is done
+#endif  
 }
 
 void TuyaSerialInput(void)
@@ -1450,11 +1454,13 @@ bool Xdrv16(uint8_t function)
       case FUNC_EVERY_SECOND:
         if (TuyaSerial && Tuya.wifi_state != TuyaGetTuyaWifiState()) { TuyaSetWifiLed(); }
         if (!Tuya.low_power_mode) {
+#if 0          
           Tuya.heartbeat_timer++;
           if (Tuya.heartbeat_timer > 10) {
             Tuya.heartbeat_timer = 0;
             TuyaSendCmd(TUYA_CMD_HEARTBEAT);
           }
+#endif          
 #ifdef USE_TUYA_TIME
           if (!(TasmotaGlobal.uptime % 60)) {
             TuyaSetTime();
```
```

wernerhp, 2021-10-15:
I find that once the timer runs out it doesn't reset.
Setting it to `TuyaSend2 7,1` seems to trigger the relay after 1 second and then the timer stops.
I stop receiving `TYA` messages as soon as the timer runs out.
Even after a reboot I don't get a timer message.