arendst/Tasmota #10134: Multiple frequencies for RFRecv, rc-switch

bilogic, 2020-12-14:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
`RFRecv` supports only 1 receiver per microprocessor today, this makes it impossible to support multiple frequencies at the same time.

**Describe the solution you'd like**  
Make `RFRecv` support multiple receivers, e.g. 1 x 315Mhz and 1x 433Mhz together

**Describe alternatives you've considered**  
Using 2 microprocessors, which is quite a waste.

**Additional context**  
There is a working proof of concept here https://github.com/BCsabaEngine/RCSwitch and I'm ready to make the PR, but need some help understanding Tasmota code

1. I'm reviewing https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_17_rcswitch.ino and have some trouble understanding how to work out which pin `RFRecv` is attached to.

I may have more questions along the way. Thank you.

**(Please, remember to close the issue when the problem has been addressed)**

bilogic, 2020-12-14:
https://github.com/arendst/Tasmota/blob/0bf82f51ddb30991bec802257eb70addcc6b8029/tasmota/xdrv_17_rcswitch.ino#L236

1. I need to call `RfInit(void)` to fire for each RX pin, possible?

arendst, 2020-12-14:
> RFRecv supports only 1 receiver per microprocessor today, this makes it impossible to support multiple frequencies at the same time.

How about adding another wemos with another type of receiver? This is just a hardware change (using 434MHz or ie. 858MHz). I use it too on different devices.

Adding more interfaces to this software is more like a challenge and a can-do solution than adding anything usefull to it. It over complicates selecting user input and I feel it is of no use.

As for me, don't waste your time screwing up the code.

closed 2020-12-14 by ascillato2

bilogic, 2020-12-14:
Yea, I'm starting to understand the code. I just figured out the part about finding all pins that are `RFRecv`. Sorry, I'm very new to Tasmota.

1. I assumed that it was possible to multiple `WS2812B` on different GPIOs, is this correct?
2. I wanted to do the same for `RFRecv`
3. Do you mean it is not possible to have 2 x `RFRecv`?

ascillato, 2020-12-14:
> I assumed that it was possible to have multiple WS2812B on different GPIOs, is this correct?

Sorry. No. Only one led strip is supported.

> I wanted to do the same for RFRecv

Sorry, won't be added for now. https://github.com/arendst/Tasmota/issues/10134#issuecomment-744499854

bilogic, 2020-12-14:
Guys, it's done! I have `RFRecv` on D3 and D7 of my `tasmota-sensor.bin` in less than 20 lines of code. And everything is backward compatible! Please do have a look at my repo/PR shortly. Thank you.

bilogic, 2020-12-15:
Hi,

This was what I did to allow multiple definitions of `RFRecv`
https://github.com/bilogic/Tasmota/commit/432f25302770c85bcc6d4973364b16e75d6113ea

Is this something you guys will pull? If yes, I will merge my `RCSwitch.cpp` with the changes from https://github.com/arendst/Tasmota/pull/10063

Thank you.