arendst/Tasmota #10138: Smartplug LSC 970760 button not working properly in 9.1

ErPo, 2020-12-14:
### PROBLEM DESCRIPTION
After updating to version 9.10, the button doesn't funtion properly. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _Sonoff Basic (1)____
- [x] Tasmota binary firmware version number used: _9.10____
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: __OTA___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:03:05 SRC: Backlog
21:03:05 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
21:03:05 MQT: stat/tasmota_5C3F17/RESULT = {"NAME":"Generic","GPIO":[0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
21:03:06 SRC: Backlog
21:03:06 CMD: Group 0, Index 1, Command "MODULE", Data ""
21:03:06 MQT: stat/tasmota_5C3F17/RESULT = {"Module":{"1":"Sonoff Basic"}}
21:03:06 SRC: Backlog
21:03:06 CMD: Group 0, Index 1, Command "GPIO", Data "255"
21:03:06 MQT: stat/tasmota_5C3F17/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"57":"Led2i"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"18":"Button2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

21:04:00 SRC: WebConsole from 192.168.178.222
21:04:00 CMD: Group 0, Index 1, Command "STATUS", Data "0"
21:04:00 MQT: stat/tasmota_5C3F17/STATUS = {"Status":{"Module":1,"DeviceName":"SW1-TAS","FriendlyName":["Switch1Tasmota"],"Topic":"tasmota_5C3F17","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:04:00 MQT: stat/tasmota_5C3F17/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"2T07:32:24","StartupUTC":"2020-12-12T12:31:36","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2020-12-10T17:04:57","SaveCount":104,"SaveAddress":"FA000"}}
21:04:00 MQT: stat/tasmota_5C3F17/STATUS2 = {"StatusFWR":{"Version":"8.5.1(lite)","BuildDateTime":"2020-10-02T10:10:35","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"397/699"}}
21:04:00 MQT: stat/tasmota_5C3F17/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Ziggo4337128",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
21:04:00 MQT: stat/tasmota_5C3F17/STATUS4 = {"StatusMEM":{"ProgramSize":473,"Free":528,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F000783","04108001","00000006","010001C0","00000000","00004004","00000000"],"Drivers":"1,2,3,4,9,10,16,20,21,37","Sensors":"3"}}
21:04:00 MQT: stat/tasmota_5C3F17/STATUS5 = {"StatusNET":{"Hostname":"tasmota_5C3F17-7959","IPAddress":"192.168.178.37","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"213.46.228.196","Mac":"84:0D:8E:5C:3F:17","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:04:00 MQT: stat/tasmota_5C3F17/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.59","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5C3F17","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:04:00 MQT: stat/tasmota_5C3F17/STATUS7 = {"StatusTIM":{"UTC":"2020-12-14T20:04:00","Local":"2020-12-14T21:04:00","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:36","Sunset":"16:53"}}
21:04:00 MQT: stat/tasmota_5C3F17/STATUS10 = {"StatusSNS":{"Time":"2020-12-14T21:04:00"}}
21:04:00 MQT: stat/tasmota_5C3F17/STATUS11 = {"StatusSTS":{"Time":"2020-12-14T21:04:00","Uptime":"2T07:32:24","UptimeSec":199944,"Vcc":3.499,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"ON","Wifi":{"AP":1,"SSId":"Ziggo4337128","BSSId":"3C:84:6A:95:FE:56","Channel":4,"RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
NA

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Upgrade LSC 970760 using webinterface from 8.51 to 9.10. Press the button on the device. Martplug will not toggle pwer on/off but only power on shortly.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Instead of toggling between power on/off, it will just quickly switch on and back off.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
NA

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**
[image]

ascillato, 2020-12-14:
Hi, 

Without logs there isn't much information to help you. Please, complete the template using latest version and specially on the part:

> Provide the output of the Console log output when you experience your issue; if applicable:
> (Please use weblog 4 for more debug information)

ErPo, 2020-12-14:
Strange behaviour. Last week I went through all button and switch settings and I didn't get it to work. Now, after upgrading to version 9.10, the update changes GPIO14 to switch1 which causes the button to act as a breaker switch instead of toggling between power on and off. Changing it to button_n 2, makes it function properly again.....For now. :-)

[image]

closed 2020-12-14 by ErPo