arendst/Tasmota #10148: Bad ADC temp calculation for ESP32

vslovak, 2020-12-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

ESP32 ADC temp measuring, standard 10K thermistor used, NTC Beta Coefficient 3450, connected 2 pcs to GPIOs 33 and 34. Bridge resistor 6K8. AdcParam 2,5250,10000,4100 setup for correct two measured points 25 and 100 C for GPIO 33. Big difference for value at GPIO34 – app 40 C. Tested more combination for different ports – result is same. I looks like only for first ADC temp is changed Adcparam used, for others default value is used. Also incorrect values for Adcparam must be setup for right output temp value.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:04:00 MQT: stat/kotel3/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
21:04:00 MQT: stat/kotel3/RESULT = {"Module":{"1":"ESP32-DevKit"}}
21:04:00 MQT: stat/kotel3/RESULT = {"GPIO0":{"32":"Tlačítko1"},"GPIO1":{"0":"Není"},"GPIO2":{"0":"Není"},"GPIO3":{"0":"Není"},"GPIO4":{"0":"Není"},"GPIO5":{"0":"Není"},"GPIO9":{"0":"Není"},"GPIO10":{"0":"Není"},"GPIO12":{"224":"Relé1"},"GPIO13":{"448":"PWM_i1"},"GPIO14":{"544":"LedLink"},"GPIO15":{"225":"Relé2"},"GPIO16":{"0":"Není"},"GPIO17":{"0":"Není"},"GPIO18":{"0":"Není"},"GPIO19":{"0":"Není"},"GPIO20":{"0":"Není"},"GPIO21":{"640":"I2C SDA"},"GPIO22":{"608":"I2C SCL"},"GPIO23":{"0":"Není"},"GPIO24":{"0":"Není"},"GPIO25":{"2560":"MX31855 DO"},"GPIO26":{"2496":"MX31855 CS"},"GPIO27":{"2528":"MX31855 CLK"},"GPIO28":{"0":"Není"},"GPIO29":{"0":"Není"},"GPIO30":{"0":"Není"}," ...

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:04:56 CMD: status 0
21:04:56 MQT: stat/kotel3/STATUS = {"Status":{"Module":1,"DeviceName":"kotel4","FriendlyName":["kotel4","Tasmota2","Tasmota3","Tasmota4"],"Topic":"kotel3","ButtonTopic":"0","Power":8,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:04:56 MQT: stat/kotel3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Vbat power on reset","Uptime":"0T00:12:46","StartupUTC":"2020-12-15T19:52:10","Sleep":50,"CfgHolder":4617,"BootCount":73,"BCResetTime":"2020-11-20T19:21:52","SaveCount":185}}
21:04:56 MQT: stat/kotel3/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(display)","BuildDateTime":"2020-11-29T16:40:29","Core":"1_0_4_2","SDK":"v3.3.4-232-g44ec7972b","CpuFrequency":80,"Hardware":"ESP32","CR":"505/699"}}
21:04:56 MQT: stat/kotel3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["vlsiot",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000500060000005A00000000000000","00804000","00007000","00000000"]}}
21:04:56 MQT: stat/kotel3/STATUS4 = {"StatusMEM":{"ProgramSize":1089,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000405","07AA858E","000C1FA1","000000C3","00080000","00000080","00000020","00000000"],"Drivers":"1,2,4,5,8,9,10,13","Sensors":"1,2,5,6,39"}}
21:04:56 MQT: stat/kotel3/STATUS5 = {"StatusNET":{"Hostname":"kotel4","IPAddress":"192.168.8.210","Gateway":"192.168.8.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.8.1","Mac":"A8:03:2A:D6:87:A0","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
21:04:56 MQT: stat/kotel3/STATUS6 = {"StatusMQT":{"MqttHost":"rpi","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D687A0","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:04:56 MQT: stat/kotel3/STATUS7 = {"StatusTIM":{"UTC":"2020-12-15T20:04:56","Local":"2020-12-15T21:04:56","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:36","Sunset":"15:47"}}
21:04:56 MQT: stat/kotel3/STATUS10 = {"StatusSNS":{"Time":"2020-12-15T21:04:56","ANALOG":{"Temperature1":22.2,"Temperature2":null,"Temperature3":null,"Temperature4":null},"MAX6675":{"ProbeTemperature":0.0,"ReferenceTemperature":0.0,"Error":0},"TempUnit":"C"}}
21:04:56 MQT: stat/kotel3/STATUS11 = {"StatusSTS":{"Time":"2020-12-15T21:04:56","Uptime":"0T00:12:46","UptimeSec":766,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":44,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Dimmer":70,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"vlsiot","BSSId":"A8:5E:45:8F:AB:99","Channel":5,"RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

20:44:44 CMD: adcparam
20:44:44 MQT: stat/kotel3/RESULT = {"AdcParam1":[2,5250,10000,4100]}
20:44:47 MQT: tele/kotel3/STATE = {"Time":"2020-12-15T20:44:47","Uptime":"0T00:16:12","UptimeSec":972,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":44,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Dimmer":70,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"vlsiot","BSSId":"A8:5E:45:8F:AB:99","Channel":5,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:44:47 MQT: tele/kotel3/SENSOR = {"Time":"2020-12-15T20:44:47","ANALOG":{"Temperature1":22.5,"Temperature2":-18.0,"Temperature3":null},"DS18B20":{"Id":"0301A279D46B","Temperature":22.3},"MAX6675":{"ProbeTemperature":0.0,"ReferenceTemperature":0.0,"Error":0},"TempUnit":"C"}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-12-16:
Just tried to emulate your issue but cannot reproduce.

I have configured three temp adc inputs like this:

[image]

Using fix resistors in a neywork like
```
3V3 -- 5k6 -- ADCinput -- 6k8 -- Gnd
```
I see changes on specific inputs if I manupulate the related ``adcparamx`` parameters. So input GPIO33, configured as ``ADC Temp2`` reacts on changes using ``adcparam2``.

I did fix the output of command ``gpio 255`` so the last gpios are now also visible. Could you try to execute this command again with the latest dev version.

vslovak, 2020-12-16:
In Tasmota Docs is documented just only AdcParam, no AdcParamX. It is now clear, yes it works, I test it.  But frustrated is big difference for nominal temperature 25C. I use network: 3V3 -- 6k8 Br – ADCinput-- 10k Th – Gnd so and I have nominal temperature app 18C. Difference is big. I know there is parameter TempOffset, but it affects all temp sensors, so I have mix of digital and analog sensors in one equipment. I think it will be useful extend AdcParamX with temp offset parameter. It should solve different linearity of ADC ESP32.

ascillato, 2020-12-16:
> big difference for nominal temperature 25C. I use network: 3V3 -- 6k8 Br – ADCinput-- 10k Th – Gnd so and I have nominal temperature app 18C.

Please, calibrate the parameters of the adcparam command to match your measurements. In the docs st commands, there is a link to the formula used.

ascillato2, 2020-12-18:
Please, when you have time, tell us if you were able to calibrate your ADC with adcparam command. Thanks.

closed 2020-12-18 by ascillato2

lalo-uy, 2023-10-06:
I’m not sure if to comment here or open a new issue.
I try to buld a 4 ntc sensor with a esp-32s.
The diference at 23C are 3 to 4 deg.
After mutch investigation I belive the problem is not calibration, but wrong values from the dac. Error of 0,2v some times.
Found several blog posts about that. They talk about a  function esp_adc_cal_raw_to_voltage() , but don’t see it in tasmota code.