arendst/Tasmota #10150: If statement support in rules

Ruebezahl11, 2020-12-16:
### PROBLEM DESCRIPTION
Nested if statements in rules donot work

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): dev board
- [x] Tasmota binary firmware version number used: 9.1.0(display)-2_7_4_5

  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used:atformio
- [x] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
09:06:10 RSL: stat/tasmota_302151/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":417,"Free":94,"Rules":"on rotary1#pos1 do if (%value%>0) displaytext [zc1l1f0s3]%value%; var1=%value%*12-1; var2=(var1-var1%12)/12; var3=(var1%12)*10; displaydimmer 100 else displaydimmer 0 endif endon on time#second do if (%var1%==1) var1=0; displaydimmer 0; power on; delay 5; power off elseif (%var1%>0) displaytext [zc1l1f0s3p0-2]%var2%; displaytext [c4l1s2p0-2]%var3%; var1=var1-1; var2=(var1-var1%12)/12; var3=(var1%12)*5 endif  endon"}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
09:10:17 CMD: status 0
09:10:17 RSL: stat/tasmota_302151/STATUS = {"Status":{"Module":0,"DeviceName":"Wecker","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_302151","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:10:17 RSL: stat/tasmota_302151/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T01:28:20","StartupUTC":"2020-12-16T06:41:57","Sleep":50,"CfgHolder":4617,"BootCount":80,"BCResetTime":"2020-11-13T21:55:26","SaveCount":400,"SaveAddress":"F6000"}}
09:10:17 RSL: stat/tasmota_302151/STATUS2 = {"StatusFWR":{"Version":"9.1.0(display)","BuildDateTime":"2020-12-16T07:41:10","Boot":6,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"415/699"}}
09:10:17 RSL: stat/tasmota_302151/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mrt150n","mrt150n_EXT"],"TelePeriod":300,"Resolution":"55818000","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00016000","00000000"]}}
09:10:17 RSL: stat/tasmota_302151/STATUS4 = {"StatusMEM":{"ProgramSize":571,"Free":432,"Heap":27,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAAA59E","04049FA1","000000C3","00000000","00004080","00000020","00000000"],"Drivers":"1,2,4,5,8,9,10,13,16","Sensors":"1,2,5,6"}}
09:10:17 RSL: stat/tasmota_302151/STATUS5 = {"StatusNET":{"Hostname":"tasmota_302151-0337","IPAddress":"192.168.3.57","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer":"(IP unset)","Mac":"C8:2B:96:30:21:51","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:10:17 RSL: stat/tasmota_302151/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_302151","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
09:10:17 RSL: stat/tasmota_302151/STATUS7 = {"StatusTIM":{"UTC":"2020-12-16T08:10:17","Local":"2020-12-16T09:10:17","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:38","Sunset":"16:53"}}
09:10:17 RSL: stat/tasmota_302151/STATUS10 = {"StatusSNS":{"Time":"2020-12-16T09:10:17"}}
09:10:17 RSL: stat/tasmota_302151/STATUS11 = {"StatusSTS":{"Time":"2020-12-16T09:10:17","Uptime":"0T01:28:20","UptimeSec":5300,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"mrt150n","BSSId":"80:1F:02:EA:7A:8E","Channel":11,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:08:34:10 RUL: TIME#SECOND performs "if(165.000>0) if(27.500<10) displaytext [zc1l1f0s3]027.500 else displaytext [zc1l1f0s3]27.500 endif; displaytext [c4l1s2]30.000; var1=var1-1; var2=var1/6; var3=(var1%6)*10 endif; if(165.000==1)  displaydimmer 0; power1 on; delay 5; power1 off endif"
08:34:10 RSL: stat/tasmota_302151/RESULT = {"If":"Done"}
08:34:10 RSL: stat/tasmota_302151/RESULT = {"If":"Done"}
08:34:10 RSL: stat/tasmota_302151/RESULT = {"DisplayText":"[c4l1s2]30.000"}
08:34:11 RSL: stat/tasmota_302151/RESULT = {"Var1":"164.000"}
08:34:11 RSL: stat/tasmota_302151/RESULT = {"Var2":"27.333"}
08:34:11 RSL: stat/tasmota_302151/RESULT = {"Var3":"20.000"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
create a rule like mine: time#second is feature I am just testing, use time#minute instead. Use whatever input (button) you have in place of the rotary encoder to set var1 to some value

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
when the rule executes, neither if nor else part of the nested if statement is executed.
One of them should, depending on variable state

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I believe that setting insertPosition = 0 inside ExecuteCommandBlock should not happen on a nested IF statement

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2020-12-16:
@Ruebezahl11,
did you compile with
#define SUPPORT_IF_STATEMENT
enabled?
IF statement is not enabled by default.
See documentation of rules.
Joerg

Ruebezahl11, 2020-12-16:
Sure,I did...
it is a custom compile because I wanted ROTARY_V1 and USE_DISPLAY together

Jason2866, 2020-12-16:
If in rules does work. What is not working as described in the documentation?
Please be more specific or close this.

ascillato, 2020-12-16:
Hi,

Your rules have syntax error. The structure for IF is 

IF (<logical-expression>) <statement-list> {ELSEIF (<logical-expression>) <statement-list>} [ELSE <statement-list>] ENDIF

So, IF something IF something is incorrect. You must use ELSEIF

Please, recheck the docs. If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-16 by ascillato2

Ruebezahl11, 2020-12-16:
Hi oscillato,
this is supposed to be a nested if:
if (<logical-expression>) _then_
    if (<logical-expression>) _then_ <statement> else <statement> endif ;
    <statement> ; <statement> ; <statement>
endif
To make the intention more clear, I have added a _then_ where some other languages would use it. 

Acccording to the log, neither the if nor the else part of the nested if execute

ascillato, 2020-12-16:
Hi,

> this is supposed to be a nested if:
> if () then
> if () then else endif ;

Exactly, that syntax is not supported. You have to use elseif or other decision structure.

Ruebezahl11, 2020-12-17:
too bad I just posted rather than previewed....
We have statement-lists, so that is statements separated by semicolon
if (cond) statement ; statement ; statement endif
We have nested IF,so one of the statements could be another if block
if (cond) **if(cond2) statement1 else statement2 endif** ;  statement ; statement endif
I hope it is now clear what I wanted to achieve. The script matches the formal grammar

ascillato, 2020-12-17:
It is clear what you want to achieve. The issue is that the syntax you want is not supported in rules. To achieve that logic you need to use the structure supported by Tasmota. 
May be would be better if you move your rules to scripting feature of Tasmota that supports more complex logic structures.