arendst/Tasmota #10154: Shelly Dimmer 2 - PowerRetain does not work

dennyfmn, 2020-12-16:
### PROBLEM DESCRIPTION

Shelly Dimmer 2 running devel 9.1.0.2 and the STM code Release v51.5. PowerRetain does not work.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
14:08:40 CMD: Backlog Template; Module; GPIO 255
14:08:40 MQT: stat/shellydimmer1/RESULT = {"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}
14:08:40 MQT: stat/shellydimmer1/RESULT = {"Module":{"0":"Shelly Dimmer 2"}}
14:08:41 MQT: stat/shellydimmer1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"5568":"SHD Boot 0"},"GPIO5":{"5600":"SHD Reset"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"193":"Switch_n2"},"GPIO13":{"0":"None"},"GPIO14":{"192":"Switch_n1"},"GPIO15":{"0":"None"},"GPIO16":{"320":"Led_i1"},"GPIO17":{"4736":"ADC Temp"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

(No Rules)

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
14:10:09 CMD: status 0
14:10:09 MQT: stat/shellydimmer1/STATUS = {"Status":{"Module":0,"DeviceName":"ShellyDimmer-1","FriendlyName":["Shelly Dimmer One"],"Topic":"shellydimmer1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
14:10:09 MQT: stat/shellydimmer1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin","RestartReason":"External System","Uptime":"0T00:47:56","StartupUTC":"2020-12-16T19:22:13","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2020-12-16T19:10:32","SaveCount":62,"SaveAddress":"FB000"}}
14:10:09 MQT: stat/shellydimmer1/STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020-12-16T14:00:10","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"451/699"}}
14:10:09 MQT: stat/shellydimmer1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["microwap2",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008129","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
14:10:09 MQT: stat/shellydimmer1/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
14:10:09 MQT: stat/shellydimmer1/STATUS5 = {"StatusNET":{"Hostname":"shellydimmer1-6885","IPAddress":"192.168.1.157","Gateway":"192.168.1.240","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.240","Mac":"40:F5:20:00:DA:E5","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:10:09 MQT: stat/shellydimmer1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_00DAE5","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
14:10:09 MQT: stat/shellydimmer1/STATUS7 = {"StatusTIM":{"UTC":"2020-12-16T20:10:09","Local":"2020-12-16T14:10:09","StartDST":"2020-03-08T02:00:00","EndDST":"2020-11-01T02:00:00","Timezone":99,"Sunrise":"07:44","Sunset":"16:32"}}
14:10:09 MQT: stat/shellydimmer1/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
14:10:09 MQT: stat/shellydimmer1/STATUS10 = {"StatusSNS":{"Time":"2020-12-16T14:10:09","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":101.7},"ENERGY":{"TotalStartTime":"2020-12-16T19:10:32","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":8},"TempUnit":"F"}}
14:10:09 MQT: stat/shellydimmer1/STATUS11 = {"StatusSTS":{"Time":"2020-12-16T14:10:09","Uptime":"0T00:47:56","UptimeSec":2876,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":87,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"microwap2","BSSId":"10:C3:7B:D1:FE:D0","Channel":6,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Have the load ON. Remove the power from the dimmer. Restore the power to the dimmer. The load is OFF.

The GUI says the load is ON, but it is OFF. Toggling OFF/ON turns the load ON.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

With RetainPower 1, The load should be returned to the state it was in when the power was lost.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

dennyfmn, 2020-12-18:
Flashed a second Shelly Dimmer 2 with devel 9.2.0.1 using Tasmotizer instead of mgos-to-tasmota used on the first unit. Flashed shelly-dimmer-stm32_v51.5.bin. Updated the first unit OTA from devel 9.1.0.2 to devel 9.2.0.1. PowerRetain does not work on either unit.

arendst, 2020-12-18:
It's a timing issue between tasmota and the mcu; at restart tasmota is quicker than the mcu telling it to set power to a known state.

Workaround, just don't remove power from the dimmer.

dennyfmn, 2020-12-18:
Ha! Power outages are somewhat unpredictable. Is there a chance this can be fixed in the future?

dennyfmn, 2020-12-18:
I'm thinking about a possible work-around using a rule. Since the GUI has the correct state on power up, can that be used in a rule? Not sure on how to code this.
On power up
wait 5 seconds
if power1 is on
   set power1 on
Thanks!

ascillato, 2020-12-18:
> ...a possible work-around using a rule...

You can use for example:

```lua
Rule1 1
Rule1 on power1#boot do backlog var1 %value%;ruletimer1 5 endon on rules#timer=1 do power1 %var1% endon
```

dennyfmn, 2020-12-18:
Thanks! It almost works. Here is what I have
 
14:06:45.884 CMD: rule1
14:06:45.893 MQT: stat/shellydimmer2/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":97,"Free":414,"Rules":"on power1#boot do backlog var1 %value%;ruletimer1 5 endon on rules#timer=1 do power1 %var1% endon"}

When it powers up, the load is OFF when it should be ON. The GUI says it is ON.

If I go to the console, and type Power1 ON, nothing happens, the load is still OFF.
If I type Power1 OFF then Power1 ON the load turns ON.

Maybe the rule needs to say:
On power up
if the GUI says Power1 is ON
  wait 5 seconds
  Power1 OFF
  Power1 ON

Thanks for your help!

ascillato, 2020-12-18:
Please set weblog to 4 and then post the output of the console when your device starts? Thanks

dennyfmn, 2020-12-18:
Hmmm...
It looks like the IF statement is not included in the precompiled binaries. Is there any way to code a rule that only activates on  power up when Power1 State = ON without using IF? 

I changed the rule to:
Rule1 on power1#boot do backlog var1 %value%;ruletimer1 10 endon on rules#timer=1 do backlog power1 off; power1 on endon

(Added backlog power1 off; power1 on)

This works for the case when the load is ON when power is lost, but also turns the load on if it power was lost and the load was OFF. In other words, it always turns the load ON after a short delay on power up. Not good.

Just saw your request for weblog 4, will do...

Changed back to the original suggested rule.

00:00:00.049 CFG: Loaded from flash at F5, Count 95
00:00:00.054 QPC: Count 1
00:00:00.056 CFG: CR 462/699, Busy 0
00:00:00.058 CFG: CR 462/699, Busy 0
00:00:00.061 SRC: Restart
00:00:00.063 Project tasmota ShellyDimmer-2 Version 9.2.0.1(tasmota)-2_7_4_9(2020-12-17T16:43:33)
00:00:00.064 SNS: Hardware Serial
00:00:00.167 SHD: Shelly Dimmer Co-processor Version v51.5
00:00:00.168 CFG: CR 462/699, Busy 0
00:00:00.182 RUL: POWER1#BOOT performs "backlog var1 1;ruletimer1 5"
00:00:00.182 SRC: Rule
00:00:00.183 CMD: Group 0, Index 1, Command "BACKLOG", Data "var1 1;ruletimer1 5"
00:00:00.226 SRC: Backlog
00:00:00.227 CMD: Group 0, Index 1, Command "VAR", Data "1"
00:00:00.231 RSL: stat/shellydimmer2/RESULT = {"Var1":"1"}
00:00:00.437 SRC: Backlog
00:00:00.439 CMD: Group 0, Index 1, Command "RULETIMER", Data "5"
00:00:00.442 RSL: stat/shellydimmer2/RESULT = {"T1":5,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
00:00:00.448 WIF: Checking connection...
00:00:00.449 WIF: Attempting connection...
00:00:00.785 WIF: Connecting to AP1 microwap2 Channel 6 BSSId 10:C3:7B:D1:FE:D0 in mode 11n as shellydimmer2-2197...
00:00:01.828 WIF: Checking connection...
00:00:01.830 WIF: Attempting connection...
00:00:02.828 WIF: Checking connection...
00:00:02.829 WIF: Attempting connection...
00:00:03.825 WIF: Checking connection...
00:00:03.826 WIF: Connected
00:00:04.074 HTP: Web server active on shellydimmer2-2197 with IP address 192.168.1.178
15:08:41.017 RTC: UTC 2020-12-18T21:08:41, DST 2020-03-08T02:00:00, STD 2020-11-01T02:00:00
15:08:42.075 MQT: Attempting connection...
15:08:42.127 MQT: Connected
15:08:42.130 MQT: tele/shellydimmer2/LWT = Online (retained)
15:08:42.132 MQT: cmnd/shellydimmer2/POWER = 
15:08:42.134 MQT: Subscribe to cmnd/shellydimmer2/#
15:08:42.136 MQT: Subscribe to cmnd/tasmotas/#
15:08:42.138 MQT: Subscribe to cmnd/DVES_002895_fb/#
15:08:42.140 MQT: Unsubscribe from homeassistant/status
15:08:42.143 MQT: tele/shellydimmer2/INFO1 = {"Module":"Shelly Dimmer 2","Version":"9.2.0.1(tasmota)","FallbackTopic":"cmnd/DVES_002895_fb/","GroupTopic":"cmnd/tasmotas/"}
15:08:42.146 MQT: tele/shellydimmer2/INFO2 = {"WebServerMode":"Admin","Hostname":"shellydimmer2-2197","IPAddress":"192.168.1.178"}
15:08:42.149 MQT: tele/shellydimmer2/INFO3 = {"RestartReason":"External System"}
15:08:42.151 MQT: stat/shellydimmer2/RESULT = {"POWER":"ON"}
15:08:42.154 MQT: stat/shellydimmer2/POWER = ON
15:08:42.167 RUL: RULES#TIMER=1 performs "power1 1"
15:08:42.169 SRC: Rule
15:08:42.171 CMD: Group 0, Index 1, Command "POWER", Data "1"
15:08:42.177 MQT: stat/shellydimmer2/RESULT = {"POWER":"ON"}
15:08:42.180 MQT: stat/shellydimmer2/POWER = ON
15:08:43.203 QPC: Reset
15:08:45.151 APP: Boot Count 18
15:08:45.885 CFG: Saved to flash at F4, Count 96, Bytes 4096
15:08:46.289 MQT: tele/shellydimmer2/STATE = {"Time":"2020-12-18T15:08:46","Uptime":"0T00:00:11","UptimeSec":11,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"microwap2","BSSId":"10:C3:7B:D1:FE:D0","Channel":6,"RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:08:46.298 MQT: tele/shellydimmer2/SENSOR = {"Time":"2020-12-18T15:08:46","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":111.1},"ENERGY":{"TotalStartTime":"2020-12-17T18:37:06","Total":0.000,"Yesterday":0.205,"Today":0.000,"Period":0,"Power":0},"TempUnit":"F"}
15:08:51.151 MQT: tasmota/discovery/40F520002895/config = {"ip":"192.168.1.178","dn":"ShellyDimmer-2","fn":["Shelly Dimmer Two",null,null,null,null,null,null,null],"hn":"shellydimmer2-2197","mac":"40F520002895","md":"Shelly Dimmer 2","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.2.0.1","t":"shellydimmer2","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[2,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0},"lk":1,"lt_st":1,"sho":[0,0,0,0],"ver":1} (retained)
15:08:51.159 MQT: tasmota/discovery/40F520002895/sensors = {"sn":{"Time":"2020-12-18T15:08:51","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":110.3},"ENERGY":{"TotalStartTime":"2020-12-17T18:37:06","Total":0.000,"Yesterday":0.205,"Today":0.000,"Power":0},"TempUnit":"F"},"ver":1} (retained)
15:09:01.326 WIF: Checking connection...
15:09:21.326 WIF: Checking connection...
15:09:41.329 WIF: Checking connection...
15:09:46.158 MQT: tele/shellydimmer2/STATE = {"Time":"2020-12-18T15:09:46","Uptime":"0T00:01:11","UptimeSec":71,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"microwap2","BSSId":" [... cut ...]
15:10:46.152 MQT: tele/shellydimmer2/STATE = {"Time":"2020-12-18T15:10:46","Uptime":"0T00:02:11","UptimeSec":131,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"microwap2","BSSId":"10:C3:7B:D1:FE:D0","Channel":6,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:10:46.163 MQT: tele/shellydimmer2/SENSOR = {"Time":"2020-12-18T15:10:46","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":104.3},"ENERGY":{"TotalStartTime":"2020-12-17T18:37:06","Total":0.000,"Yesterday":0.205,"Today":0.000,"Period":0,"Power":1},"TempUnit":"F"}
15:11:01.326 WIF: Checking connection...

dennyfmn, 2020-12-18:
Changed the rule to add a condition and shortened the timer; this seems to work.

Rule1 on power1#boot=1 do ruletimer1 2 endon on rules#timer=1 do backlog power1 off; power1 on endon

I originally was conflating the PowerRetain command with PowerOnState. I now have:
PowerRetain 0
PowerOnState 3 (the default - never had changed it)

I believe it it working correctly. The load goes ON if power was lost when it was ON, and stays OFF if the load was OFF when power was lost.

Thanks for all your help! It will take some study for me to learn about using rules in Tasmota.

ascillato, 2020-12-18:
> I believe it it working correctly. The load goes ON if power was lost when it was ON, and stays OFF if the load was OFF when power was lost.

Great, so everything was just a confusion? If there is no problem in Tasmota, please close this issue. Thanks

jamesturton, 2020-12-19:
@ascillato2 if you like you can assign this one to me. I will try to fix the sycronisation between the ESP and the STM at start up

dennyfmn, 2020-12-19:
Just to clarify, the issue is that PowerOnState is not honored. The rule above works around the issue. I only mentioned PowerRetain because I initially thought it had something to do with the problem; it does not. Sorry for the confusion.
The title should really say "Shelly Dimmer 2 - PowerOnState does not work". Thanks for all your great work and help. This is a learning experience for me. I will close this issue.

ascillato2, 2020-12-19:
@jamesturton  

Thanks. Very appreciated.

dennyfmn, 2020-12-19:
@jamesturton 
Thanks for all your work and picking this up...

PR #10182 (2020-12-19): Shelly Dimmer: Use global setting value for brightness and fade rate at startup
## Description:

Use global setting value for brightness and fade rate at startup. This way the Shelly Dimmer obeys PowerOnState.
**Related issue (if applicable):** fixes #10154

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [ ] The code change is tested and works on Tasmot [... cut]
closed 2020-12-19 by dennyfmn