arendst/Tasmota #10155: All relays assigned the same device group should be sync'ed [SO88=1]

gbg108, 2020-12-16:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

The SetOption88 feature provides support for multiple relays using different device groups however there is an issue when multiple relays are assigned the same device group name. In this case only the first relay will be sync'ed to other remote devices in the group and other relays will be ignored.

For example, if I have three devices, one device (D1) with four relays, a second (D2) with a single relay and a third (D3) with a single relay and I would like to keep two relays on D1 in sync with D2 and two relays on D1 in sync with D3. To do this I would set the following device group names:

D1:{"DevGroupName1":"flood_lights","DevGroupName2":"flood_lights","DevGroupName3":"christmas_lights","DevGroupName4":"christmas_lights"}

D2:
{"DevGroupName1":"flood_lights","DevGroupName2":"","DevGroupName3":"","DevGroupName4":""}

D3:
{"DevGroupName1":"christmas_lights","DevGroupName2":"","DevGroupName3":"","DevGroupName4":""}

With the current code when D2 is toggled, only relay1 on D1 will toggle. Similarly when D3 is toggled only relay3 on D1 will toggle.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

All relays on the local device which are assigned the same device group should be sync'ed with remote devices in the group. Moreover, even if there are no remote devices, I believe device groups should still be enforced on the local device. This means that toggling one relay in the webUI should also toggle all other relays sharing that device group on the device.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
A discussion on this topic as already occurred in https://github.com/arendst/Tasmota/pull/10140

As per @pcdiem 
> To summarize, we have two options: 1. Add a 16-bit mask to map relays to device groups. This will require a new command to set the mappings but adds the ability to map all eight relays. 2. Add logic to dynamically create a map at boot time to map device groups with the same name into the same device group index. This solution certainly has the smallest footprint but only supports four relays when SO88 is enabled.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

pcdiem, 2020-12-26:
I was wrong. I proceeded with option 2 to handle multiple devices groups with the same name but the code winds up being unreasonably large. The code to handle this with a setting to map out relays to device groups is quite small and will be more flexible. In order to map each relay to no device group or to device group 1 - 4, we'll need 3 bits per relay. This will also eliminate the need for the SO88 bit. If I can get 24 bits of setting space allocated, I can add this.

pcdiem, 2021-01-21:
I'm waiting for someone (Theo?) to allocate the 24-bits of setting space needed for this feature. If that's not going to happen, this can be closed.

arendst, 2021-01-21:
I'll add it now.

arendst, 2021-01-21:
Over to you @pcdiem

pcdiem, 2021-02-03:
I'm finally getting around to working on this again. I'm trying to understand again why we need this. The request is to allow multiple relays to be in a single device group when SO88 is enabled. As @gbg108 stated, if relays 1 and 2 on device D1 are in device group "flood_lights", he would expect toggling relay 1 on D1 to send a message to device group "flood_lights" and also toggle relay 2 on D1. Likewise, toggling relay 2 on D1 would send a message to device group "flood_lights" and also toggle relay 1 on D1. This can easily be accomplished by enabling SO88, setting DevGroupName1 to "flood_lights" and adding the following rule on D1:

RULE1 ON Power1#State DO Power2 %value% ENDON ON Power2#State DO Power1 %value% ENDON

This will keep relays 1 and 2 in sync on D1 and will send out a device group update when the state of either relay 1 or 2 changes. This will, wind up publishing an extra MQTT tele message and an extra device group update when relay 2 is toggled because the rule will make it also toggle relay1 even though relay 1 will already be in the new state and ExecuteCommandPower doesn't check to see if this is the case. There's really no harm in this though.

The only issue with this is that we currently have no way to say that a particular device group is not to be used. So, in the scenario above, toggling either relay 1 or 2 will also send a device group update to device group 2 (using either DevGroupName2 or the MQTT Group Topic with _2 appended). Again, there's no real harm in this but it would be nice to have a way to indicate that we a relay that shouldn't be associated with any device group.

We also have the issue that there is currently no check to make sure that a user doesn't use the same device group name for multiple device groups. I can add this if we think it's worth the space.

Am I missing something or is it true that the above rule will accomplish what is being asked for in this feature request?

pcdiem, 2021-02-06:
Also, in light of the fact that Tasmota now supports up to 28 relays, the 32-bit settings field won't support a relay-to-device-group map. My current thinking is that we should use the 32-bit settings field to be an array of four bytes (one per device group) with each byte indicating which relay the device group tracks. This will allow any one of up to 254 relays to be associated with a device group. If someone wants multiple relays to be associated with device group, they can use rule to keeps the relays in sync. This scheme will also allow us to specify that a relay is not part of any device groups. I can't think of a case where having a single relay associated with multiple devices groups would provide any real benefit so that will be disallowed.

Any thoughts?

arendst, 2021-02-07:
Sounds fine with me. You could also just support the first 8 relays only.

For ESP8266 the limit stays on 8 relays (for now and I'm quite sure to keep it that way.

PR #10898 (2021-02-10): Add DGR device map support
## Description:

Add support for the device groups device map. If relays in separate device groups (SO88) is enabled, the map used used to determine which device group changes to the device (relay) state are sent to. The DevGroupDevice<x> command sets which device is assigned to the device group. If a device is not assigned to any device group, no device group updates are sent when the device changes state.

This PR also includes a reduction in the DevGroupSend light channels item code size.

This PR adds 288 bytes of program code.

**Related issue (if applicable):** [... cut]
closed 2021-02-10 by arendst