arendst/Tasmota #10159: Self compiled version is using wrong core

helmar74, 2020-12-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [X ] Tasmota binary firmware version number used: _____9.2.0
  - [ ] Pre-compiled
  - [X ] Self-compiled
    - [ ] IDE / Compiler used: PlatformIO _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Compile tasmota with platformIO with "standard settings"
After Updating a device, it show's, that it is still running on Core 2.7.4.3:
3:41:03 MQT: stat/sonoffSC_1/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020.12.17 13:33:06","Boot":31,"Core":"2_7_4_3","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"405/699"}}

I would assume, that compiling the new version would upgrade core to 2.7.4.9?
What can I do to ensure, that the build is done with current core version (2.7.4.9)?

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-12-17:
You are using old platformio*.* files or you have not deleted (hidden) folder `.platformio` to get latest core

helmar74, 2020-12-17:
> You are using old platformio*.* files or you have not deleted (hidden) folder `.platformio` to get latest core

I unzipped the downloaded zipped, and unpacked it.
What do I have to delete to get the new core?

helmar74, 2020-12-17:
I updated the Libaries in PlatformIO, now it's working

closed 2020-12-17 by helmar74