arendst/Tasmota #10161: DS18B20 not working in Tasmota v 9.2.0

helmar74, 2020-12-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): _____Wemos Mini D with DS18B20
- [X ] Tasmota binary firmware version number used: _____v9.2.0
  - [ ] Pre-compiled
  - [X ] Self-compiled
    - [X ] IDE / Compiler used: _____PlatformIO
- [ ] Flashing tools used: _____OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Hi, I self compiled tasmota 9.2.0.
Everything is working, but DS18B20 on two different Wemos Mini D aren't recognized anymore.
I downgraded to 9.1.0. there it's working without an problem

Anything I can try?

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-12-17:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2020-12-17:
Have you tried the precompiled Tasmota-sensors.bin?

helmar74, 2020-12-17:
> Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you? [...]

Updated the template.

helmar74, 2020-12-17:
DS18x20 is configured on D4 GPIO2. But temperature isn't shown

ascillato2, 2020-12-17:
Please try the precompiled Tasmota-sensors.bin?

helmar74, 2020-12-17:
Tasmota-sensors.bin is working...

What could I have done wrong in my_user_config.h

ascillato2, 2020-12-17:
> What could I have done wrong in my_user_config.h

I don't know. You haven't post that file. Sorry.

As it is a self compilation issue, it is better to address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). Hope you don't mind as the chat is a better and more dynamic channel for helping you in these types of cases.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-17 by ascillato2