arendst/Tasmota #10162: DS18B20 not working with my_user_config.h change to my defaults

helmar74, 2020-12-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____Wemos Mini D and DS18x20 sensor
- [ ] Tasmota binary firmware version number used: _____9.2.0
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: _____PlatformIO
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I have a compile error.
I adapted my_user_config.h to "my defaults"
Strange behavoiur:
When I have set this:
.
#define MDNS_ENABLED           false             // [SetOption55] Use mDNS (false = Disable, true = Enable)
.
.
//#define USE_TX20_WIND_SENSOR                     // Add support for La Crosse TX20 anemometer (+2k6/0k8 code)

If I change one of this settings, the DS18x20 sensors aren'T initalized correctly...
Attached you can find two files. One with working configuration, one with non working configuration

Has anyone an idea, what's going wrong here?

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

helmar74, 2020-12-17:
Sorry, can't upload at the momen, getting an error

ascillato2, 2020-12-17:
Just for reference: #10161, #10159

arendst, 2020-12-17:
Stop editing my_user_config.h and learn how to use user_config_override.h

helmar74, 2020-12-17:
Currently learning to use user_config.override.h.
How do I disable a feature (e.g. USE_DOMOTICZ)?

Do I have to enter 
#undef USE_DOMOTICZ
in user_config.override.h.
?

ascillato, 2020-12-17:
Yes, exactly

helmar74, 2020-12-17:
Ah, great. one further question ;-)
#define USE_DISPLAY is used twice, once for SPI and once for I2C. If I want to use both, do I have to enter in override:

#ifdef USE_I2C
    #undef USE_DISPLAY
  #define USE_DISPLAY
#endif  // USE_I2C
.
.
#ifdef USE_SPI
    #undef USE_DISPLAY
     #define USE_DISPLAY
 #endif  // USE_SPI
 ?

Or simply one time ?

ascillato, 2020-12-17:
The Override file will add or remove features for the Tasmota standard version. So, i2c is already defined in the standard Tasmota. If you want to use i2c you dont need to add it, but if you want to disable it, you must undef it.

So, undef just what you don't want and define all the features you want.

helmar74, 2020-12-17:
OK, so I created the override file and left all others default. Compilation worked
So far so good. ;-)
With this settings at the end of theFile the DS18B20 doesn't be initialized:
#undef USE_TX20_WIND_SENSOR
#define USE_TX20_WIND_SENSOR                     // Add support for La Crosse TX20 anemometer (+2k6/0k8 code)

[user_config_override.h.txt](https://github.com/arendst/Tasmota/files/5711534/user_config_override.h.txt)
#endif  // _USER_CONFIG_OVERRIDE_H_

If I change to:
#undef USE_TX20_WIND_SENSOR
//#define USE_TX20_WIND_SENSOR                     // Add support for La Crosse TX20 anemometer (+2k6/0k8 code)

#undef USE_DISCOVERY
#define USE_DISCOVERY                            // Enable mDNS for the following services (+8k code or +23.5k code with core 2_5_x, +0.3k mem)
#endif  // _USER_CONFIG_OVERRIDE_H_

The DS18B20 works.

I still thin enabling mDNS and enabling TX20 has a side effect to DS18B20 sensors...

Attached you find my override file with working settings. (Only renamed to .txt for uploading)
[user_config_override.h.txt](https://github.com/arendst/Tasmota/files/5711537/user_config_override.h.txt)

Maybe I made a mistake

helmar74, 2020-12-21:
I recompiled with the updated 9.2.0 version on 21.12.2020. Now the issue is solved.
Thanks a lot!

closed 2020-12-21 by helmar74