arendst/Tasmota #10163: Issues connecting device to AWS iot.  TSL error 1002

litzenberger, 2020-12-17:
### PROBLEM DESCRIPTION
I have followed directions and from AWS side I am able to connection using `test-invoke-authorizer` for the TasmotaAuth.

I used the output from AWS which is:
```
BackLog SetOption3 1; SetOption103 1; MqttHost XXXXX.iot.us-east-1.amazonaws.com; MqttPort 443; MqttUser tasmota?x-amz-customauthorizer-name=TasmotaAuth; MqttPassword XXXXXXXXXXXXXXXXX
```
Flashed the device according to docs.  Not sure what I am missing but the device gives:

17:57:49 MQT: TLS connection error: -1002
17:57:49 MQT: Connect failed to  XXXXXXX-ats.iot.us-east-1.amazonaws.com:443, rc -2. Retry in 10 sec

I have tried a coupe different ways to get this to work and same results.  Any ideas?

Thanks!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-12-17:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

litzenberger, 2020-12-17:
> Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you? [...]

yes, sorry about that.  Updated template

litzenberger, 2020-12-17:
the one things I cant verify is if the password is saved.  when I open the Configure MQTT, password is just three stars.  When I updated it to the current password and save it still shows three stars.  Is this expected?

ascillato, 2020-12-17:
Yes, for security your password is never sent from Tasmota to the web browser.

litzenberger, 2020-12-17:
I just tested my AWS on a nodeJS client and it is connecting.

s-hadinger, 2020-12-17:
Error -1002 means that the tcp connection failed. I suppose that the domain name is correct. 

Your Tasmota device can't reach the endpoint.  Either DNS failed or something is blocking the connection on your network (firewall?)

litzenberger, 2020-12-17:
yeah, that was my thought too nut I can reach the endpoint from my laptop connected to the same network.

litzenberger, 2020-12-17:
The last thing I did was overwrite the host in the configure MQTT screen and its now connecting.  Thanks Ill close this.

closed 2020-12-17 by litzenberger