arendst/Tasmota #10164: weblog >2 and syslog >2 generates a restart (at least with shelly2.5)

Bascht74, 2020-12-17:
### PROBLEM DESCRIPTION
If I use `syslog 3` `syslog 4` or `mqttlog 3` or `mqttlog 4` the device keeps restarting after a couple of seconds after rebooting until I chance the log level of syslog and/or mqttlog to 2, 1 or 0.
weblog seems not to be affected by this, It works flawesly.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly2.5
- [X] Tasmota binary firmware version number used: 9.2.0 20201217 10:10
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
18:15:23.443 SRC: WebConsole from 192.168.175.172
18:15:23.446 CMD: Gruppe 0, Index 1, Befehl "WEBLOG", Daten "2"
18:15:23.453 MQT: stat/tasmota_FB1089/RESULT = {"WebLog":2}
18:15:49.290 CMD: Backlog Template; Module; GPIO 255
18:15:49.357 MQT: stat/tasmota_FB1089/RESULT = {"NAME":"Shelly 2.5 (buttons)","GPIO":[320,0,34,0,224,129,0,0,640,128,608,225,3456,4736],"FLAG":0,"BASE":18}
18:15:49.594 MQT: stat/tasmota_FB1089/RESULT = {"Module":{"0":"Shelly 2.5 (buttons)"}}
18:15:49.802 MQT: stat/tasmota_FB1089/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"34":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"129":"Button_in2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"128":"Button_in1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ"},"GPIO17":{"4736":"ADC Temp"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
18:16:21.794 CMD: Backlog Rule1; Rule2; Rule3
18:16:21.863 MQT: stat/tasmota_FB1089/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":168,"Free":343,"Rules":"ON Clock#Timer=1 DO ShutterOpen1 ENDON ON Clock#Timer=2 DO ShutterClose1 ENDON ON Clock#Timer=3 DO ShutterPosition 10 ENDON ON Clock#Timer=4 DO ShutterPosition 50 ENDON"}
18:16:22.076 MQT: stat/tasmota_FB1089/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":"18:16:22.289 MQT: stat/tasmota_FB1089/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
18:16:57.466 CMD: Status 0
18:16:57.475 MQT: stat/tasmota_FB1089/STATUS = {"Status":{"Module":0,"DeviceName":"ShellySwitch25-Rollo.EG.Wohnzimmer.Terrasse","FriendlyName":["Rollo.EG.Wohnzimmer.Terrasse","Tasmota2"],"Topic":"tasmota_FB1089","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:16:57.511 MQT: stat/tasmota_FB1089/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-DE.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:02:02","StartupUTC":"2020-12-17T17:14:55","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2020-12-17T02:17:05","SaveCount":558,"SaveAddress":"F7000"}}
18:16:57.546 MQT: stat/tasmota_FB1089/STATUS2 = {"StatusFWR":{"Version":"9.2.0.1(tasmota)","BuildDateTime":"2020.12.17 15:44:44","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"582/699"}}
18:16:57.565 MQT: stat/tasmota_FB1089/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Qar5tz09e589mn05a0",""],"TelePeriod":300,"Resolution":"558180F0","SetOption":["0000C009","0105C80001000600003C5A00000000000000","400000C1","00006000","00000000"]}}
18:16:57.597 MQT: stat/tasmota_FB1089/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
18:16:57.629 MQT: stat/tasmota_FB1089/STATUS5 = {"StatusNET":{"Hostname":"Rollo.EG.Wohnzimmer.Terrasse","IPAddress":"192.168.175.42","Gateway":"192.168.175.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.175.1","Mac":"68:C6:3A:FB:10:89","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:16:57.654 MQT: stat/tasmota_FB1089/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.175.214","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FB1089","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:16:57.679 MQT: stat/tasmota_FB1089/STATUS7 = {"StatusTIM":{"UTC":"2020-12-17T17:16:57","Local":"2020-12-17T18:16:57","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:16","Sunset":"16:18"}}
18:16:57.697 MQT: stat/tasmota_FB1089/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:16:57.716 MQT: stat/tasmota_FB1089/STATUS10 = {"StatusSNS":{"Time":"2020-12-17T18:16:57.708","ANALOG":{"Temperature":58.6},"ENERGY":{"TotalStartTime":"2020-12-09T09:59:34","Total":0.021,"Yesterday":0.006,"Today":0.005,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":50,"Direction":0,"Target":50},"TempUnit":"C"}}
18:16:57.749 MQT: stat/tasmota_FB1089/STATUS11 = {"StatusSTS":{"Time":"2020-12-17T18:16:57.744","Uptime":"0T00:02:02","UptimeSec":122,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Qar5tz09e589mn05a0","BSSId":"E0:63:DA:53:3C:3E","Channel":11,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:17"}}}
18:16:57.782 MQT: stat/tasmota_FB1089/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":304,"Close":282,"50perc":65,"Delay":0,"Opt":"0100","Calib":[1830,425,4085,900,1000],"Mode":"0"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

18:17:41.041 CMD: weblog 4
18:17:41.048 MQT: stat/tasmota_FB1089/RESULT = {"WebLog":4}
18:17:41.244 CFG: in Flash gespeichert am F6, zählen 559, Bytes 4096
18:17:41.259 ADE: 6074231, 4475, [1773, -1, 641, 5], [1773, 7, 641, -2]
18:17:41.261 ADE: U 6074231, C 4475, I 0 + 0 = 0, P 0 + 0 = 0
18:17:42.208 ADE: 6074445, 4475, [1773, -35, 642, 11], [1773, -51, 641, -2]
18:17:42.211 ADE: U 6074445, C 4475, I 0 + 0 = 0, P 0 + 0 = 0
18:17:43.208 ADE: 6073737, 4475, [1773, 2, 641, 75], [1773, -9, 641, 24]
18:17:43.211 ADE: U 6073737, C 4475, I 0 + 0 = 0, P 0 + 0 = 0
18:17:44.204 ADE: 6073041, 4475, [1773, -49, 641, 22], [1773, -10, 641, 8]
[... 331 lines cut ...]
18:21:48.302 SRC: WebConsole from 192.168.175.172
18:21:48.305 CMD: Gruppe 0, Index 1, Befehl "WEBLOG", Daten "2"
18:21:48.311 MQT: stat/tasmota_FB1089/RESULT = {"WebLog":2}

after setting to mqttlog 0 no problems. 
Same thing with syslog >2, no problems with weblog >2
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just enter `mqttlog 4` (on a shelly2.4 device, as I don't have any other devices jet, but tried with different devices, they all face the same problem)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
no restarts after activating mqttlog > 2

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

One time I tried syslog 4 and the device restarted several times and after I changed it back to "0" the configuration of the device was wrong. The shelly was not anymore configured with two relays, the webpage showed only on. I restored a backup configuration of tasmobackup and everythin was good. I cannot reproduce this specific behaivior, but to write about it might help to get a picture, what is going on.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-12-17:
Bug confirmed with `mqttlog 4` on shelly 2.5

Thanks for reporting.

arendst, 2020-12-18:
Currently redesigning the logging environment from real-time to async solving issues with UDP and MQTT.

closed 2020-12-18 by ascillato2

Bascht74, 2020-12-19:
tested --> no more restarts with shelly2.5...

Great!

Sebastian