arendst/Tasmota #10165: 9.1.0.2 Wemo and Alexa group problem

Duffy77, 2020-12-17:
### PROBLEM DESCRIPTION
I have flashed some Devices with 9.1.0.2 and enabled WeMo Emulation.
Devices where found by Alexa with the right friendly Name (Stehlampe, Vitrine, Ventilator, Blume ...)
When i try to group the devices (Alexa App or at the PC) the wrong device is grouped and some devices are not possible to group.  

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Teckin sp22_
- [x] Tasmota binary firmware version number used: 9.1.0.2 and 9.2.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
1. let discover alexa new devices with WeMo emulation
2. min. 2 devices (i have used 6 devices) (Stehlampe, Vitrine, Ventilator, Blume,...)
2. go to alexa app or alexa.amazon.de and create a new group
3. put the first device in the new group and save (e.g. Stehlampe)
4. save the group 
5. open the group and look which devices are in there. if you have luck, then Stehlampe is in your group. But if you have no luck, another device (e.g. Vitrine) is in your group. in my szenario with 6 devices i only can put 2 devices in the group. the other devices are displayed but never put in the group.

so you can not put the device vitrine in one group, because Stehlampe is always used when you choose Vitrine.
seems, that there is a unique id in the device, which is discovered by alexa and when you group the device, the first device with that id is used...?

### EXPECTED BEHAVIOUR
i would like to group devices in rooms, but some devices are not groupable

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-12-17:
Closing this issue as it is **duplicated**. Sorry.

See #10105 

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

s-hadinger, 2020-12-17:
I've never seen this issue. Do your Tasmota devices have fixed IP addresses?

Duffy77, 2020-12-17:
Hi, no they have dhcp. But they kept their ip address cause of long lease time.I have checked and they have the same ip since first contact

Duffy77, 2020-12-17:
When I switch to hue emulation , i don't have the problem.It's a work around , But then all my switches are light items :(

Reopened #10165.

peteakalad, 2020-12-18:
I have same/similar issue unable to add wemos emu switch from tasmota to routine or group, yet it can be turned on and off from alexa.  New wemos emu from node-red works fine so there is something different about tasmota wemos emu.  Was fine until recent updates.   I have gen2 dot.  Tried setoption109 0 and 1 same issue.

peteakalad, 2020-12-18:
I rolled back and upgraded/tested each build from 8.5.1 deleting the device from alexa, updating tasmota build, discovering devices in alexa and then testing the newly added plug to add to a routine.

9.1.0 tasmota.bin was working ok so not seeing the same problem as the OP here on that version.  But 9.2.0 for some reason the wemos plug even though its listed in the main devices list, doesnt come up in the 'add to routine' dialog.  It does come up for group edits.

Perhaps not exactly the same issue as the OP (my device isnt showing up in routine adds vs group adds) but it would indicate some sort of issue with wemos emu. 

Something to do with https://github.com/arendst/Tasmota/commit/742171ca5eecdea451712a21cff9c34c2299cf84 perhaps?

s-hadinger, 2020-12-18:
I will have a look but apart from minot maintenance changes, wemo code hasn't changed for a while. The wemo multi is not enabled unless you self compile.

peteakalad, 2020-12-18:
Are you sure about multi being only self compile?  

Looks to me that USE_EMULATION_WEMO (flag that seems to be in most builds) was changed so that xdrv_21_wemo_multi.ino was included instead of xdrv_21_wemo.ino as part of this commit:

https://github.com/arendst/Tasmota/commit/742171ca5eecdea451712a21cff9c34c2299cf84

arendst, 2020-12-19:
Alas, multi is enabled by default as it was promissed to be a direct replacement for the single version.

How ignorent I could be....

peteakalad, 2020-12-19:
Maybe needs a new issue but..

I've tested with a bin from the original multi wemo PR discussion, and it too has the issue with the device not appearing in the routines list, confirming that it is the multi wemo causing my issue.
https://github.com/arendst/Tasmota/pull/9208#issuecomment-723645654 

I have also noted that a duplicate device appears during upgrades from original single wemo to multi wemo emu when doing discovery post upgrade.  I expect this duplicate device is because the multi code now uses the device id in the generation of the serial number and uuid.  That would suggest all users of wemo emu upgrading to 9.2.0, who do a subsquent Alexa discovery will end up with duplicate devices.  I have single relay/switch devices (mini, TH10).   

I would say the presented wemo serial/uuid needs to be consistent through upgrades - for single/first devices that would mean disabling the involvement of the device id in the serial number/uuid introduced by multi wemo emu PR.  Additionally this may fix the problem for device 1 not turning up in routines device list - it could be the serial number change from the standard wemo format causes probems.

peteakalad, 2020-12-19:
I confirm my suspicions are right.  

I compiled a binary from dev branch with the wemo multi code lines reverted so the setup xml friendly name, uuid and serial number no longer included the device id - i.e. the same generation of these strings as from single wemo.

This has resolved my issues, no duplicates following upgrade from < 9.1.0 and no issues adding the device to routines.  I'll raise a PR which does this only for 1st switch and for switch 2+ it'll have the current behaviour.  I think that the current behaviour is flawed given my routine problems, but its better than simply reverting back to single only I guess.

closed 2020-12-22 by ascillato2

ivanmarban, 2021-01-02:
9.2.0 is still having the problem described by @Duffy77 about trying to add various wemo emulated devices to a group in Alexa app. No matter you do, always only one and same device is added to whatever group.