arendst/Tasmota #10166: Shutters: The ShutterStop command triggers a reverse run of the shutter in certain situations

Bascht74, 2020-12-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I use the shutter function of Tasmota in conjunction with Shelly 2.5 and the latest DEV firmware. 
In certain situations the ShutterStop command triggers a reverse run of the shutter motor instead of simply stopping. Normally this should not happen according to @stefanbode, but the shutter should simply stop with some "inaccuracy". 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly2.5
- [X] Tasmota binary firmware version number used: 9.2.0.1(tasmota) 2020.12.17 15:44:44 2_7_4_9/2.2.2-dev(38a443e)
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
19:26:46.772 CMD: Backlog Template; Module; GPIO 255
19:26:46.857 MQT: stat/tasmota_FB1089/RESULT = {"NAME":"Shelly 2.5 (buttons)","GPIO":[320,0,34,0,224,129,0,0,640,128,608,225,3456,4736],"FLAG":0,"BASE":18}
19:26:47.098 MQT: stat/tasmota_FB1089/RESULT = {"Module":{"0":"Shelly 2.5 (buttons)"}}
19:26:47.308 MQT: stat/tasmota_FB1089/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"34":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"129":"Button_in2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"128":"Button_in1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ"},"GPIO17":{"4736":"ADC Temp"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
19:27:06.061 CMD: Backlog Rule1; Rule2; Rule3
19:27:06.142 MQT: stat/tasmota_FB1089/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":168,"Free":343,"Rules":"ON Clock#Timer=1 DO ShutterOpen1 ENDON ON Clock#Timer=2 DO ShutterClose1 ENDON ON Clock#Timer=3 DO ShutterPosition 10 ENDON ON Clock#Timer=4 DO ShutterPosition 50 ENDON"}
19:27:06.379 MQT: stat/tasmota_FB1089/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
19:27:06.591 MQT: stat/tasmota_FB1089/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:27:30.410 CMD: Status 0
19:27:30.422 MQT: stat/tasmota_FB1089/STATUS = {"Status":{"Module":0,"DeviceName":"ShellySwitch25-Rollo.EG.Wohnzimmer.Terrasse","FriendlyName":["Rollo.EG.Wohnzimmer.Terrasse","Tasmota2"],"Topic":"tasmota_FB1089","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:27:30.432 MQT: stat/tasmota_FB1089/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-DE.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:51:42","StartupUTC":"2020-12-17T17:35:48","Sleep":50,"CfgHolder":4617,"BootCount":1,"BCResetTime":"2020-12-17T18:34:59","SaveCount":597,"SaveAddress":"F8000"}}
19:27:30.440 MQT: stat/tasmota_FB1089/STATUS2 = {"StatusFWR":{"Version":"9.2.0.1(tasmota)","BuildDateTime":"2020.12.17 15:44:44","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"582/699"}}
19:27:30.445 MQT: stat/tasmota_FB1089/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Qar5tz09e589mn05a0",""],"TelePeriod":300,"Resolution":"558180F0","SetOption":["0000C009","0105C80001000600003C5A00000000000000","400000C1","00006000","00000000"]}}
19:27:30.457 MQT: stat/tasmota_FB1089/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
19:27:30.469 MQT: stat/tasmota_FB1089/STATUS5 = {"StatusNET":{"Hostname":"Rollo.EG.Wohnzimmer.Terrasse","IPAddress":"192.168.175.42","Gateway":"192.168.175.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.175.1","Mac":"68:C6:3A:FB:10:89","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:27:30.477 MQT: stat/tasmota_FB1089/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.175.214","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FB1089","MqttUser":"mqtt","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:27:30.486 MQT: stat/tasmota_FB1089/STATUS7 = {"StatusTIM":{"UTC":"2020-12-17T18:27:30","Local":"2020-12-17T19:27:30","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:16","Sunset":"16:18"}}
19:27:30.493 MQT: stat/tasmota_FB1089/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:27:30.507 MQT: stat/tasmota_FB1089/STATUS10 = {"StatusSNS":{"Time":"2020-12-17T19:27:30.499","ANALOG":{"Temperature":62.0},"ENERGY":{"TotalStartTime":"2020-12-09T09:59:34","Total":0.021,"Yesterday":0.006,"Today":0.006,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":49,"Direction":0,"Target":49},"TempUnit":"C"}}
19:27:30.519 MQT: stat/tasmota_FB1089/STATUS11 = {"StatusSTS":{"Time":"2020-12-17T19:27:30.514","Uptime":"0T00:51:42","UptimeSec":3102,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Qar5tz09e589mn05a0","BSSId":"E0:63:DA:53:3C:3E","Channel":6,"RSSI":98,"Signal":-51,"LinkCount":4,"Downtime":"0T00:01:25"}}}
19:27:30.528 MQT: stat/tasmota_FB1089/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":304,"Close":282,"50perc":65,"Delay":0,"Opt":"0100","Calib":[1830,425,4085,900,1000],"Mode":"0"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
20:15:18.496 CMD: weblog 4
20:15:18.507 MQT: stat/tasmota_FB1089/RESULT = {"WebLog":4}
20:15:18.724 CFG: in Flash gespeichert am F4, zählen 601, Bytes 4096
20:15:19.408 ADE: 6136484, 4478, [1773, -44, 648, 24], [1773, -21, 648, -4]
20:15:19.410 ADE: U 6136484, C 4478, I 0 + 0 = 0, P 0 + 0 = 0
20:15:19.931 WIF: Prüfe Verbindung...
20:15:20.408 ADE: 6135047, 4478, [1773, 6, 648, 88], [1773, 16, 648, -7]
20:15:20.410 ADE: U 6135047, C 4478, I 0 + 0 = 0, P 0 + 0 = 0
20:15:21.408 ADE: 6134957, 4479, [1773, 5, 648, 29], [1773, 1, 648, 19]
20:15:21.410 ADE: U 6134957, C 4479, I 0 + 0 = 0, P 0 + 0 = 0
20:15:22.408 ADE: 6135794, 4478, [1773, -14, 648, 52], [1773, 6, 648, 5]
[... 182 lines cut ...]
void CmndShutterStop(void)
{
  [...]
        int32_t temp_realpos = ShutterCalculatePosition(i);
        XdrvMailbox.payload = ShutterRealToPercentPosition(temp_realpos, i)-Shutter[i].direction;
        TasmotaGlobal.last_source = SRC_WEBGUI;
        CmndShutterPosition();
[...]
}
```

I think it might be because (if I observe this correctly in the log and understand the code correctly) ShutterRealToPercentPosition returns an absolute percentage. But for certain positions in the lower third (0-30%), shutterset50percent makes this range larger than in the range (31-100%). This results in values that deviate more from the current position and thus trigger these backward movements in some cases. 

[... cut ...]
I also think that without shutterset50percent the problem does not occur or occurs extremely rarely.

I hope you get what I mean. My mother language is German as yours, but I try to keep it in English, even it is difficult for me to write my thoughts...

P.S. I looked at the log an the theoretical runtime. I should run (theory 50ms * (144 - 27) = ) 5.850 ms and it did (see log) 5.836 ms, even with weblog 4. So pretty accurate stuff!

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-12-18:
@Bascht74 , yes, great and excellent analysis. There is no more investigation required on my side. Thx. I have an idea, how to change the shutterstop function to make it work. Will submit it later today.

closed 2020-12-18 by arendst

Bascht74, 2020-12-18:
@stefanbode Thanks! This is a very elegant way to solve it :-) I got the idea, much easier...
I will try it out after the dev firmware files are updated!

Bascht74, 2020-12-18:
Hi @stefanbode I tried it, and it worked! Great! `Shutterstop` / remote commands / Buttons are now much more responsive as well...

The only think that wonders me, now that we got the milliseconds ("Jammern auf hohem Niveau" / "Whining on a high level"):

Time 37 (.144) --> log says: act pos: 36000
at .218 --> shutterstop (what would be + 74 ms = act pos. 38000 (Time 39)
but it takes till .347 until the shutter logs the new stuff (Time 41) --> act pos: 40000 (.144 + 200ms = pos 36000 + 4000)

I don't know why it is not taking the time 39 or time 40 slot to process it ... (so that it reacts earlier)
There is also nothing in the log about time 38 - time 40...
So the position is of by 3000 ( = 150 ms)

If there is a cause for this (not checking / logging between Time 38 - 41) that can be optimized, the reaction to user inputs (e.g. push a button / remote button) would be a litte earlier (up to 150 ms)

Without understanding your code completely I right now think that might be because every 50ms only one shutter related code piece is executed. So it has to wait for the next slow until it will pickup the new stuff. 
Could you get around it that way, that you trigger the position-check code right after you executed `Shutter[i].target_position = Shutter[i].real_position;`?

This way it would be as responsive as the shutteropen/-close commands (that take 50 ms to switch on the relay)...

shutterstop: 
```

10:43:30.144 SHT: time: 37, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 36000, next_stop 0, target: 628000,  max_vel_change 0, dir: 1
10:43:30.210 ADE: 6195236, 4473, [44758, -16336, 16536, -2520], [1773, 6, 654, -14]
10:43:30.213 ADE: U 6195236, C 4473, I 44758 + 0 = 44758, P 16336 + 0 = 16336
10:43:30.218 CMD: shutterstop
10:43:30.224 SRC: WebConsole from 192.168.175.172
10:43:30.227 CMD: Gruppe 0, Index 1, Befehl "SHUTTERSTOP", Daten ""
10:43:30.231 SHT: Stop moving 1: dir: 1
10:43:30.238 MQT: stat/tasmota_FB1089/RESULT = {"ShutterStop":"Done"}
10:43:30.347 SHT: time: 41, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 40000, next_stop 0, target: 38000,  max_vel_change 0, dir: 1
10:43:30.350 SHT: Stop Shutter 0. Switchmode 0
10:43:30.353 SRC: Shutter
10:43:30.355 SHT: Switched relay: 1 by Shutter
10:43:30.358 SHT: Shutter 1: source: Shutter, powerstate_local 0, ShutterGlobal.RelayCurrentMask 1, manual change 0
10:43:30.363 MQT: stat/tasmota_FB1089/RESULT = {"POWER1":"OFF"}
10:43:30.370 MQT: stat/tasmota_FB1089/POWER1 = OFF (beibehalten)
10:43:30.876 SHT: Shutter 1 Real 41000, Start 41000, Stop 38000, Dir 0, Delay 0, Rtc 2.10 [s], Freq 0, PWM 0
10:43:30.881 MQT: stat/tasmota_FB1089/SHUTTER1 = 1 (beibehalten)
10:43:30.886 MQT: stat/tasmota_FB1089/RESULT = {"Shutter1":{"Position":1,"Direction":0,"Target":1}}
10:43:30.983 CFG: in Flash gespeichert am F8, zählen 662, Bytes 4096
```

shutteropen/close:
```
11:06:27.072 CMD: shutteropen
11:06:27.077 SRC: WebConsole from 192.168.175.172
11:06:27.080 CMD: Gruppe 0, Index 1, Befehl "SHUTTEROPEN", Daten ""
11:06:27.084 SHT: Pos. in: payload  (0), payload 100, idx 1, src 7
11:06:27.086 SHT: lastsource 7:, real 41000, target 608000, payload 100
11:06:27.089 SHT: Delay Start. var0 <99>=<>, max10s?
11:06:27.107 SHT: Delay Start. Done
11:06:27.110 SRC: Shutter
11:06:27.112 SHT: Switched relay: 1 by Shutter
11:06:27.115 SHT: Shutter 1: source: Shutter, powerstate_local 1, ShutterGlobal.RelayCurrentMask 1, manual change 0
11:06:27.121 MQT: stat/tasmota_FB1089/RESULT = {"POWER1":"ON"}
11:06:27.128 MQT: stat/tasmota_FB1089/POWER1 = ON (beibehalten)
11:06:27.134 MQT: stat/tasmota_FB1089/RESULT = {"ShutterOpen1":100}
11:06:28.431 SHT: time: 27, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 67000, next_stop 0, target: 628000,  max_vel_change 0, dir: 1

```

PR #10170 (2020-12-18): #10166 fixed
improved stop procedure to avoid overrun due to rounding issues between % and the real_position

## Description:

**Related issue (if applicable):** fixes #10166

## Checklist:
  - [ x] The pull request is done against the latest dev branch
  - [x ] Only relevant files were touched
  - [ x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [ x] The code change is tested and works on Tasmota core ESP32 V.1.0.4.2
  - [x ] I accept the [CLA](https://github.com/arendst/ [... cut]