arendst/Tasmota #10172: Set Option to control LWT messages

ortegafernando, 2020-12-18:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I have a door/windows sensor and I will like to avoid it publishing LWT message to broker. But this problem is also for every device with deepsleep functions.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
We need an option to make an initial connection to broker without LWT. 

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
None

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-12-18:
Hi,

LWT is part of the MQTT protocol. Is not going to be removed and adding a remove option will overcomplicate the setups adding a new source of error for users.

For deepsleep, you just need to config your sensors differently in your home automation software.

If you are struggling with deepsleep, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-18 by ascillato2

ortegafernando, 2020-12-19:
Ok I understand that this is a residual feature. Thanks a lot.