arendst/Tasmota #10173: Scripting subscribe nested JSON message

infoflo, 2020-12-18:
### PROBLEM DESCRIPTION
If you subscribe to MQTT Events, only 2 nexted hierarchies seem to be possible.

Works:
+>Subscribe A,foo/bla,directive.header
{"directive":{"header":"TurnOn"}}
A -> TurnOn

Does not work:
+>Subscribe A,foo/bla,directive.header.name
{"directive":{"header":{"name":"TurnOn"}}}
A -> empty

Alternativly how can JSON Messages handled in scripting itself, it could also be possible to Subscribe to the complete message, and parsing it later.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): not device dependent
- [x] Tasmota binary firmware version number used: 9.1.0 and 9.2.0
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: platformio
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Script output here:

>D 512
A=""

>S
; Mqtt Connected
if mqttc>0 {
  +>Subscribe A,alexa/command/cosolit/10882,directive.header
}

if upd[A]>0 {
  print A triggered
  print A %A%
}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
13:59:21 MQT: tasmota/DG/WR/Desk_Light/stat/STATUS = {"Status":{"Module":1,"DeviceName":"DG WR Desk Light","FriendlyName":["DG WR Desk Light"],"Topic":"DG/WR/Desk_Light","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":60,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1}}
13:59:21 MQT: tasmota/DG/WR/Desk_Light/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"DG","OtaUrl":"http://xxx/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:21:04","StartupUTC":"2020-12-18T12:38:17","Sleep":50,"CfgHolder":4617,"BootCount":44,"BCResetTime":"2020-10-16T15:31:07","SaveCount":95,"SaveAddress":"F7000"}}
13:59:21 MQT: tasmota/DG/WR/Desk_Light/stat/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-17T17:24:27","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"514/699"}}
13:59:21 MQT: tasmota/DG/WR/Desk_Light/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":1,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Frank 2.4GHz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000080A9","2805C8000100060000005A00000000000000","000000C0","00006000","00000000"]}}
13:59:21 MQT: tasmota/DG/WR/Desk_Light/stat/STATUS4 = {"StatusMEM":{"ProgramSize":478,"Free":524,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","83000583","04080001","00000006","010000C0","00000100","00000000","00000000"],"Drivers":"1,2,3,4,9,10,27","Sensors":"2,3"}}
13:59:21 MQT: tasmota/DG/WR/Desk_Light/stat/STATUS5 = {"StatusNET":{"Hostname":"DG-WR-Desk-Light","IPAddress":"192.168.178.119","Gateway":"192.168.178.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.11","Mac":"CC:50:E3:02:33:E3","Webserver":2,"WifiConfig":4,"WifiPower":20.0}}
13:59:21 MQT: tasmota/DG/WR/Desk_Light/stat/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.frank.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0233E3","MqttUser":"DG_WR_Desk_Light","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:59:21 MQT: tasmota/DG/WR/Desk_Light/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-12-18T12:59:21","Local":"2020-12-18T13:59:21","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"08:15","Sunset":"16:26"}}
13:59:21 MQT: tasmota/DG/WR/Desk_Light/stat/STATUS10 = {"StatusSNS":{"Time":"2020-12-18T13:59:21"}}
13:59:21 MQT: tasmota/DG/WR/Desk_Light/stat/STATUS11 = {"StatusSTS":{"Time":"2020-12-18T13:59:21","Uptime":"0T00:21:04","UptimeSec":1264,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxx","BSSId":"38:10:D5:0F:F6:9F","Channel":6,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Subscribe to MQTT JSON message with more than 2 nested hierarchies.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Get values out of nested JSON messages with more than 2 hierarchies, 3-5 or besser without restriction.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-12-19:
tosmota only supports 2 nested levels in rules or scripts. i don't have it on my timetable to expand this as it is rarely needed.
returning the complete message is also not a good option because scripter is limited in maximum string size to 48 chars.
what is your use case ? 
there is probably a different solution other than subscribe.

closed 2020-12-19 by ascillato2

infoflo, 2020-12-19:
I have a mqtt bridge to a external service, that generated mqtt messages on Alexa Smarthome Commands. These are json formatted. Currently I am using node-blue to translate these json messages into commands for tasmota devices. I thought, perhaps it would be possible to get rid of the node-blue part with tasmota scripting. Because I moved almost all automatisation into the tasmota devices itself by using scripting.

gemu2015, 2020-12-20:
ok, as this seems an interesting use case i implemented a deeper JSON path search
try with appended scripter version. i also enabled more debug to show whats going on.
i can not test my self since my broker does not support subscribe.

use this additional define
#define SUPPORT_MQTT_EVENT_MORE

[xdrv_10_scripter.ino.zip](https://github.com/arendst/Tasmota/files/5720917/xdrv_10_scripter.ino.zip)