arendst/Tasmota #10175: Scripting USE_SCRIPT_SUB_COMMAND not working

infoflo, 2020-12-18:
### PROBLEM DESCRIPTION
in Scripting USE_SCRIPT_SUB_COMMAND seeams not to work at all.

Example:
#vacuum(state)
=>print %state%

16:30:56 CMD: VACUUM 1
16:30:56 MQT: tasmota/DG/KU/IR/stat/RESULT = {"Command":"Unknown"}

According to the documentation is should word as command:
If #define USE_SCRIPT_SUB_COMMAND is included in your user_config_override.h, a subroutine may be invoked via the Console or MQTT using the subroutine's name. For example, a declared subroutine #SETLED(num) may be invoked by typing SETLED 1 in the Console. The parameter 1 is passed into the num argument. This also works with string parameters.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): independent of device
- [x] Tasmota binary firmware version number used: 9.1.0 and 9.2.0
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: platformio
- [x] Flashing tools used: ota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
>D
state=-1

>S
=#vacuum(1)

#vacuum(state)
=>print executed

```
- [ ] Provide the output of this command: `Status 0`:
```

16:30:56 CMD: VACUUM 1
16:30:56 MQT: tasmota/DG/KU/IR/stat/RESULT = {"Command":"Unknown"}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Executing Scripting Subroutine with compileflag USE_SCRIPT_SUB_COMMAND via console or mqtt cmnd.

### EXPECTED BEHAVIOUR
Subroutine should get executed and not "unknown command", according to documentation.

_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-12-18:
due to the fact that Tasmota capitalizes cmds your subroutine must be upper case.
since i coded this there was a change in Tasmota affecting the cmd numbering. will provide a fix soon.
currently a 1 is appended to the cmd so that you must name your subroutine

#VACUUM1(state)

and call it with 

VACUUM state

infoflo, 2020-12-18:
Thx, it is working with your hints. There is still a "unknown command" message after the execution of the subroutine, but it works and gets executed now.

closed 2020-12-19 by ascillato2