arendst/Tasmota #10176: Rule information updates on the display every second (instead of 5 minutes)

freexmonster, 2020-12-18:
### PROBLEM DESCRIPTION

I just want to update the sensor information every second and display this along with the time and seconds? How to do it?
I didn't understand the manual. I read it 10 times and did not understand anything.
I want to update this rule every second. Now the update takes place every 5 minutes (as I understand it by default).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
rule1  
      on tele-AHT1X-0x38 do DisplayText [z] [fp] [s2c2x15y0] [tS] endon 
      on tele-AHT1X-0x38#Temperature do DisplayText [s2c1x0y23]%value%C endon
      on tele-AHT1X-0x38#Humidity do DisplayText [s2c1x70y23]%value%% endon
      on tele-AHT1X-0x38#DewPoint do DisplayText [s1c1x45y50]%value%C endon

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-12-18:
`tele-` triggers only on TelePeriod which is 300 seconds by default.

```
rule1  
      on AHT1X-0x38 do DisplayText [z] [fp] [s2c2x15y0] [tS] endon 
      on AHT1X-0x38#Temperature do DisplayText [s2c1x0y23]%value%C endon
      on AHT1X-0x38#Humidity do DisplayText [s2c1x70y23]%value%% endon
      on AHT1X-0x38#DewPoint do DisplayText [s1c1x45y50]%value%C endon
```

freexmonster, 2020-12-18:
> `tele-` triggers only on TelePeriod which is 300 seconds by default.
>
> ```
> rule1
>       on AHT1X-0x38 do DisplayText [z] [fp] [s2c2x15y0] [tS] endon
>       on AHT1X-0x38#Temperature do DisplayText [...]

Yes it works, thank you for your prompt reply.
Found the line 141 my_user_config.h
[TelePeriod] Telemetry (0 = disable, 10 - 3600 seconds)
I set the value to 10.
This is certainly not realtime, but at least something worked.

meingraham, 2020-12-18:
Note - I removed `tele-` from the triggers so those modified rules should trigger every time the value changes. You can set TelePeriod back to the default.

BTW, you don't need to recompile to change Tasmota settings (e.g., `TelePeriod`). You can do that from the Console.

ascillato2, 2020-12-18:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-18 by ascillato2

stjakoer, 2022-09-15:
I've got a similar problem

`rule1 on tele-counter1#c1 do backlog var1 %value%; MULT1 0.0026; DisplayText[zs2l1c1] Liter: ; DisplayText [s2l2c1] %var1% endon`

When I use tele- it only triggers every 10 seconds. But I would like to have it at least every 2 seconds. When I write it without tele- nothing works.
Why is this?

barbudor, 2022-09-16:
Hooking on a 2 years old issue and claiming to have the same problem without providing much details is not the best way to get accurate answer
With 'tele-' the rule is executed at each teleperiod, minimum being 10 seconds
Without 'tele-' the rule is executed each time the counter changes with some caveats when the counter is above 2^23 because of float conversion (see https://github.com/arendst/Tasmota/issues/16008)