arendst/Tasmota #10180: Si7021 heater element query

liams1989, 2020-12-18:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
Humidity reading is high and have found on the wiki 

NOT SUPPORTED

Si7021 sensor contains an integrated resistive heating element that may be used to drive off condensation. It's recommended enabling the heater when the humidity readings are above 80%. Humidity readings above 80% become very unreliable. If the "Sonoff Si7021 sensor" is used at the dew point and condensation occurs on the sensor, the on-chip heater could be activated to dry the sensor and restore operation once the sensor is above the dew point. Turning on the heater will also reduce the tendency of the humidity sensor to accumulate an offset due to "memory" of sustained high humidity conditions. Six different power levels are available. The maximum source current of a GPIO pin on an ESP8266 is about 12mA. So the first power level of 3.09mA and second power level of 9.18mA would be theoretically usable.

Keep that in mind if you use this "Sonoff Si7021 sensor" outdoors!

So am confused is this an option to add a power setting to the GPIO pin?
**Describe the solution you'd like**  
Option to control power level of gpio pin 

**Describe alternatives you've considered**  
Unable to think of alternatives 

**Additional context**  
None
**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-12-18:
There is no need for any new option in Tasmota. Just define any free GPIO as output and turn it on when needed. You can add rules logic to do that automatically. 

Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-18 by ascillato2

liams1989, 2020-12-19:
> There is no need for any new option in Tasmota. Just define any free GPIO as output and turn it on when needed. You can add rules logic to do that automatically.
>
>
>
> Closing this issue as it has [...]

Hi thanks for your reply. Where do I assign the gpio pin and what do I assign it to?
Also where do I add rule logic?
Thanks for your help!

ascillato, 2020-12-19:
Gpio assignments are done from Tasmota web menu. From module configuration menu. It is very recommended to check the Tasmota docs for more information.

Please, for Tasmota usage check the docs, check discussions and/or the Tasmota users chat. Github issues are meant only for Software bug reports. Thanks.