arendst/Tasmota #10188: Tuya MCU Dimmer will not raise/lower via Tasmota, just ON/OFF

peepshow-21, 2020-12-19:
### PROBLEM DESCRIPTION
The device will not dim when issued command via Tasmota. Only ON/OFF works. On when ant dimmer value>1 is set. Off when 0 it set as dimmer. The light dim fine via the buttons on the switch.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and  [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____ Tuya MCU Dimmer
- [X] Tasmota binary firmware version number used: _____9.2.0
  - [ ] Self-compiled
- [X] Flashing tools used: _____tamotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
12:48:08 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
12:48:08 SRC: Backlog
12:48:08 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
12:48:08 MQT: stat/lounge-lights/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:48:08 SRC: Backlog
12:48:08 CMD: Group 0, Index 1, Command "MODULE", Data ""
12:48:08 MQT: stat/lounge-lights/RESULT = {"Module":{"54":"Tuya MCU"}}
12:48:08 SRC: Backlog
12:48:08 CMD: Group 0, Index 1, Command "GPIO", Data "255"
12:48:08 MQT: stat/lounge-lights/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"416":"PWM1"},"GPIO13":{"0":"None"},"GPIO14":{"290":"Led3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
12:48:11 TYA: Send "55aa00000000ff"

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
Not using Rules.

```
- [X] Provide the output of this command: `Status 0`:
```lua
18:06:05 MQT: stat/lounge-lights/STATUS = {"Status":{"Module":54,"DeviceName":"Lounge Lights","FriendlyName":["lounge-lights"],"Topic":"lounge-lights","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:06:05 MQT: stat/lounge-lights/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T05:52:22","StartupUTC":"2020-12-19T11:13:43","Sleep":50,"CfgHolder":4617,"BootCount":188,"BCResetTime":"2020-12-07T18:40:57","SaveCount":361,"SaveAddress":"FA000"}}
18:06:05 MQT: stat/lounge-lights/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-17T10:05:31","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"410/699"}}
18:06:05 MQT: stat/lounge-lights/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Error-404s",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280503000100060000005A00000000000000","00000000","00006000","00000000"]}}
18:06:05 MQT: stat/lounge-lights/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
18:06:05 MQT: stat/lounge-lights/STATUS5 = {"StatusNET":{"Hostname":"lounge-lights-0931","IPAddress":"172.17.0.117","Gateway":"172.17.0.5","Subnetmask":"255.255.0.0","DNSServer":"172.17.0.5","Mac":"C8:2B:96:0E:43:A3","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:06:05 MQT: stat/lounge-lights/STATUS6 = {"StatusMQT":{"MqttHost":"172.17.20.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0E43A3","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:06:05 MQT: stat/lounge-lights/STATUS7 = {"StatusTIM":{"UTC":"2020-12-19T17:06:05","Local":"2020-12-19T18:06:05","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:40","Sunset":"16:54"}}
18:06:05 MQT: stat/lounge-lights/STATUS10 = {"StatusSNS":{"Time":"2020-12-19T18:06:05"}}
18:06:05 MQT: stat/lounge-lights/STATUS11 = {"StatusSTS":{"Time":"2020-12-19T18:06:05","Uptime":"0T05:52:22","UptimeSec":21142,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Error-404s","BSSId":"00:22:33:44:AA:A0","Channel":10,"RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
12:52:24 TYA: Heartbeat
12:52:27 WIF: Checking connection...
12:52:35 TYA: Send "55aa00000000ff"
12:52:35 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
12:52:35 TYA: Heartbeat
12:52:36 SRC: WebGui from 172.17.20.1
12:52:36 CMD: Group 0, Index 1, Command "DIMMER", Data "40"
12:52:36 MQT: stat/lounge-lights/RESULT = {"POWER":"ON","Dimmer":40}
12:52:37 SRC: WebGui from 172.17.20.1
12:52:37 CMD: Group 0, Index 1, Command "DIMMER", Data "66"
12:52:37 MQT: stat/lounge-lights/RESULT = {"POWER":"ON","Dimmer":66}
12:52:37 CFG: Saved to flash at FB, Count 352, Bytes 4096
12:52:46 TYA: Send "55aa00000000ff"
12:52:46 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
12:52:46 TYA: Heartbeat
12:52:47 WIF: Checking connection...

```

### TO REPRODUCE
Just slide the slider and watch lights

### EXPECTED BEHAVIOUR
Expect light to dim and raise. they just stay the same

### SCREENSHOTS

Issue 1.

I have installed Tasmota (9.2.0) & configured it. I can see the slider;

[image]

My module config is;
[image]

If I move it, I see events in the logs;
[image]

But the lights do not change they just stay on.

If I set it to zero, it goes off. any other value is on (at whatever value the switch left it). Toggle button works fine too. The dimmer will go up and down using the hardware buttons on the switch, although it can be via sluggish in showing those events in the console.

I've turned weblog 4 on and used the hardware buttons to find the range;
[image]
As per the dimmer section here (https://tasmota.github.io/docs/TuyaMCU/)
Although the documentation show this;
[image]
which is a little different that what I get.

Which goes from 11 to 255, so I've done this;
*dimmerrange 11,255

No matter what I try, I can seem to get the light to do anything other than on/off via Tasmoto.

Issue 2.

If I press the ON button on the switch, it fades in up the values to come on slow. Same with OFF, if I press that, it fades down the values until off. This is VERY annoying, especially off. When you press it, it looks like nothing has happened so you press it again. Such a stupid implementation. Is there a config, so you press off and it just goes off? I think I'll remove them if there isn't, or wife will never let me add anything else in house EVER!! :)

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

peepshow-21, 2020-12-20:
Issue seem to be that there is no connection between the Tamsota control and dimmer.
The first part below is moving the slider in the webpage. you can see the value changing and the being reflect in the MQT messages sent, but *NOT*the value for dpid 2 (which seems to tbe the dimmer value) I see no messages related to TYA

btw, the first TYA message relate to moving the slider off 0, which seem to trigger the relay on dpld 1, which turns it on and that sets dpid 2 to value 11, which is it's lowest and the value it was last at when turned off. But after that initial part, no moving of the slider causes any changes in value, other than putting it to 0, which will turn the light off (dpId 1 again), nothing of the slider seems to do dpid 2!!!!!

```
17:01:46 TYA: fnId=11 is set for dpId=1
17:01:46 TYA: RX Relay-1 --> MCU State: On Current State:On
17:01:46 {"TuyaReceived":{"Data":"55AA00070008020200040000000B21","Cmnd":7,"CmndData":"020200040000000B","DpType2Id2":11,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000000B"}}}
17:01:46 TYA: fnId=21 is set for dpId=2
17:01:46 TYA: RX value 11 from dpId 2 
17:01:46 SRC: WebGui from 172.17.20.1
17:01:46 CMD: Group 0, Index 1, Command "DIMMER", Data "30"
17:01:46 MQT: stat/lounge-dimmer/RESULT = {"POWER":"ON","Dimmer":30}
17:01:47 SRC: WebGui from 172.17.20.1
17:01:47 CMD: Group 0, Index 1, Command "DIMMER", Data "46"
17:01:47 MQT: stat/lounge-dimmer/RESULT = {"POWER":"ON","Dimmer":46}
17:01:47 CFG: Saved to flash at F5, Count 684, Bytes 4096
17:01:47 SRC: WebGui from 172.17.20.1
17:01:47 CMD: Group 0, Index 1, Command "DIMMER", Data "63"
17:01:47 MQT: stat/lounge-dimmer/RESULT = {"POWER":"ON","Dimmer":63}
17:01:48 CFG: Saved to flash at F4, Count 685, Bytes 4096
17:01:50 TYA: Send "55aa00000000ff"
```

By comparison, If I press the buttons on the fitting, I message messages related to the TYA dpid2 value changing and also the MQT message related to this change.

```
17:03:21 TYA: fnId=11 is set for dpId=1
17:03:21 TYA: RX Relay-1 --> MCU State: On Current State:On
17:03:21 {"TuyaReceived":{"Data":"55AA00070008020200040000002C42","Cmnd":7,"CmndData":"020200040000002C","DpType2Id2":44,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000002C"}}}
17:03:21 TYA: fnId=21 is set for dpId=2
17:03:21 TYA: RX value 44 from dpId 2 
17:03:21 SRC: Switch
**17:03:21 CMD: Group 0, Index 3, Command "DIMMER", Data "17"
17:03:21 TYA: Send dim skipped value 41 for dpid 2
17:03:21 MQT: stat/lounge-dimmer/RESULT = {"POWER":"ON","Dimmer":17}**
17:03:21 CFG: Saved to flash at F9, Count 688, Bytes 4096
17:03:21 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
17:03:21 TYA: fnId=11 is set for dpId=1
17:03:21 TYA: RX Relay-1 --> MCU State: On Current State:On
17:03:21 {"TuyaReceived":{"Data":"55AA00070008020200040000002F45","Cmnd":7,"CmndData":"020200040000002F","DpType2Id2":47,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000002F"}}}
17:03:21 TYA: fnId=21 is set for dpId=2
**17:03:21 TYA: RX value 47 from dpId 2** 
17:03:22 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
17:03:22 TYA: fnId=11 is set for dpId=1
17:03:22 TYA: RX Relay-1 --> MCU State: On Current State:On
17:03:22 {"TuyaReceived":{"Data":"55AA00070008020200040000003248","Cmnd":7,"CmndData":"0202000400000032","DpType2Id2":50,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000032"}}}
17:03:22 TYA: fnId=21 is set for dpId=2
17:03:22 TYA: RX value 50 from dpId 2 
17:03:22 SRC: Switch
**17:03:22 CMD: Group 0, Index 3, Command "DIMMER", Data "20"
17:03:22 TYA: Send dim skipped value 46 for dpid 2
17:03:22 MQT: stat/lounge-dimmer/RESULT = {"POWER":"ON","Dimmer":20}**
17:03:22 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
17:03:22 TYA: fnId=11 is set for dpId=1
17:03:22 TYA: RX Relay-1 --> MCU State: On Current State:On
17:03:22 {"TuyaReceived":{"Data":"55AA0007000802020004000000354B","Cmnd":7,"CmndData":"0202000400000035","DpType2Id2":53,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000035"}}}
17:03:22 TYA: fnId=21 is set for dpId=2
17:03:22 TYA: RX value 53 from dpId 2 
17:03:22 CFG: Saved to flash at F8, Count 689, Bytes 4096
17:03:24 WIF: Checking connection...
17:03:29 TYA: Send "55aa00000000ff"
17:03:29 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
17:03:29 TYA: Heartbeat
```

I'm probably missing a setting somewhere, just no idea what!!

vajonam, 2020-12-20:
If your dimmer is complete controlled via MCU then in you module config you don’t need anything other than TuyaRX /TX

the TuyaMCU 11,1 and 21,2 will hook up dimmer and relay to the MCU. This should fix you problem. I am running a similar dimmer with just the settings

you might allow

SetOption20 and 54 to be able to adjust the dimmer without turning on the light.

vajonam, 2020-12-20:
[image]

peepshow-21, 2020-12-20:
@vajonam , thank you!!!! that worked!!

One other thing, when you press off on the switch, it slowly fade. This is VERY annoying as it so slow and you dont think its working for a few moments. I hate that.

Can it be made to just turn off/on, back to the last value? That would be much better

vajonam, 2020-12-20:
There are some options the fade it in our out. Start with Fade setting. But this might be a function of the MCU.  I don’t have any expertise with fading. You have to look at the settings. 

 `Fade 0`

Fair enough on the LED 3. For mine everything is controlled via MCU so really not much tasmota to control.

digiblur, 2020-12-20:
A lot of the dimmers have built in fade and you can't do anything short of writing your own code and flashing it to the secondary MCU the "TuyaMCU" itself.  

One good test to see if Tasmota is impacting anything.  For TuyaMCU you should be using `module 54` already, go to the console, set `module 18` This will put you as generic with nothing configured.  Now go to the actual dimmer face plate itself.  Turn it on and off, dim it up/down, etc whatever.  This will show you the exact function of the dimmer as designed.

ascillato2, 2020-12-27:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-27 by ascillato2