arendst/Tasmota #10189: WebUI Script Editor does not load script correctly when Zigbee enabled

martin072, 2020-12-19:
### PROBLEM DESCRIPTION
Compiling a binary with Zigbee and Scripting enabled causes the WebUI Script Editor to load the script incorrectly 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 R1
- [x] Tasmota binary firmware version number used: 9.2.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: Platform IO (Mac OS)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```l
<img width="965" alt="Screenshot 2020-12-19 at 19 00 45" src="https://user-images.githubusercontent.com/24524506/102697495-68de5300-4236-11eb-9d1b-c7c897c9214c.png">
ua
 20:07:36 MQT: home/sonoffs/stat/84:F3:EB:0C:44:70/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
20:07:36 Event: 2.00
20:07:36 MQT: home/sonoffs/stat/84:F3:EB:0C:44:70/RESULT = {"Module":{"0":"Generic"}}
20:07:36 Event: 2.00
20:07:36 MQT: home/sonoffs/stat/84:F3:EB:0C:44:70/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3584":"Zigbee Rx"},"GPIO2":{"0":"None"},"GPIO3":{"3552":"Zigbee Tx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
20:08:14 MQT: home/sonoffs/stat/84:F3:EB:0C:44:70/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota Zigbee","FriendlyName":["Tasmota Zigbee"],"Topic":"84:F3:EB:0C:44:70","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:08:14 MQT: home/sonoffs/stat/84:F3:EB:0C:44:70/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:37:06","StartupUTC":"2020-12-19T18:31:08","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2020-12-17T21:10:07","SaveCount":60,"SaveAddress":"3F8000"}}
20:08:14 MQT: home/sonoffs/stat/84:F3:EB:0C:44:70/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-19T15:56:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"424/699"}}
20:08:15 MQT: home/sonoffs/stat/84:F3:EB:0C:44:70/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UPC233367",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
20:08:15 MQT: home/sonoffs/stat/84:F3:EB:0C:44:70/STATUS4 = {"StatusMEM":{"ProgramSize":710,"Free":3364,"Heap":17,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87CA8587","046D8001","000004CF","050002C4","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,8,9,10,12,16,17,22,23,24,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,9,31,45"}}
20:08:15 MQT: home/sonoffs/stat/84:F3:EB:0C:44:70/STATUS5 = {"StatusNET":{"Hostname":"84:F3:EB:0C:44:70-1136","IPAddress":"192.168.2.127","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.4.4","Mac":"84:F3:EB:0C:44:70","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:08:15 MQT: home/sonoffs/stat/84:F3:EB:0C:44:70/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0C4470","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:08:15 MQT: home/sonoffs/stat/84:F3:EB:0C:44:70/STATUS7 = {"StatusTIM":{"UTC":"2020-12-19T19:08:15","Local":"2020-12-19T20:08:15","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:40","Sunset":"16:54"}}
20:08:15 MQT: home/sonoffs/stat/84:F3:EB:0C:44:70/STATUS10 = {"StatusSNS":{"Time":"2020-12-19T20:08:15"}}
20:08:15 MQT: home/sonoffs/stat/84:F3:EB:0C:44:70/STATUS11 = {"StatusSTS":{"Time":"2020-12-19T20:08:15","Uptime":"0T00:37:07","UptimeSec":2227,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"UPC233367","BSSId":"DE:0D:17:AD:89:74","Channel":5,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Compile version 9.2.0 with Zigbee AND Scripting enabled

### EXPECTED BEHAVIOUR
Script should load into the text frame

### SCREENSHOTS
Attached

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2020-12-19:
Hi
From what I understand in the code, the script text is sent to the web page through low-level function ` _WSContentSend(glob_script_mem.script_ram);` which bypass the buffer.
Thus if there is content in the buffer, the script gets in the page before the content of the buffer.
I tried adding `WSContentFlush();` before sending the cript content and it seems to solve the problem :

`xdrv_1_scripting.ino:5200` : 
```
    // script is to larg for WSContentSend_P
    WSContentFlush(); // flushing buffer before low-level direct send
    if (glob_script_mem.script_ram[0]) {
      _WSContentSend(glob_script_mem.script_ram);
    }
    WSContentSend_P(HTTP_FORM_SCRIPT1b);
```

If someone can confirm this is the right thing, I can do the PR.

gemu2015, 2020-12-20:
@barbudor 

yes that's the right place. strange enough that this didn't lead to problems earlier.

ill make a pr soon as i have another issue to solve

PR #10218 (2020-12-21): Scripter fix
fix scripter webeditor distortion when zigbee is enabled
enable larger json hirarchie at mqtt event

## Description:

**Related issue (if applicable):** fixes #10189

also adds larger JSON hierarchy see #10173
currently enabled by 
#define SUPPORT_MQTT_EVENT_MORE
until confirmed working without side effects.

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP826 [... cut]
closed 2020-12-21 by arendst