arendst/Tasmota #10199: Why did you remove XdrvCall(FUNC_SETTINGS_OVERRIDE) event call for drivers? Bug or on purpose?

mikep1998, 2020-12-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I was using the driver event FUNC_SETTINGS_OVERRIDE to initialize my own GPIO configuration. 
I have discovered the FUNC_SETTINGS_OVERRIDE event call has been removed. 
With patch Fix reset to defaults after 6 hours of DeepSleep the call was removed. 
Can you put it back? Or can you tell me another way to set the GPIOs in my drivers code?

  if (1 == RtcReboot.fast_reboot_count) {  // Allow setting override only when all is well
    UpdateQuickPowerCycle(true);
    XdrvCall(FUNC_SETTINGS_OVERRIDE);
  }

If are not going to support the event you should also remove it from the API.md

Thanks

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

  if (1 == RtcReboot.fast_reboot_count) {  // Allow setting override only when all is well
    UpdateQuickPowerCycle(true);
    XdrvCall(FUNC_SETTINGS_OVERRIDE);
  }

arendst, 2020-12-20:
It's not used in the current repository and therefor removed.

You'll have to dive into predefined template configuration or just load your required template once as it is persistent during release upgrades.

ascillato2, 2020-12-20:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-20 by ascillato2

mikep1998, 2020-12-21:
You provided this API in you documentation for other to use. Now you have removed it?

Are you going to removed other APIs that you provided?

arendst, 2020-12-21:
If they are not used in the current repository they will be removed.

mikep1998, 2020-12-21:
Why did you publish an API for people to use?

arendst, 2020-12-21:
At that time it looked a useful call. Better solutions are now available so the call is gone.

Anyway, you recompiled your own additions already so nothing stops you to add the call again.

mikep1998, 2020-12-21:
True I have added the code back in my compile. But I like to keep current with the latest code and not fall to far behind. That means I will have to add that snippet of code back to your files every time I upgrade. Which I was doing over a year ago everytime I updated. Then I found I could write my own driver and use the API. Even the API is not perfect after I load the code I need reboot the device one more time for the changes to take affect.