arendst/Tasmota #10202: Shelly Dimmer 2 not dimming with hardware button

DasLukas, 2020-12-20:
### PROBLEM DESCRIPTION
I flashed a Shelly Dimmer 2 with the newest Tasmota version 9.2.0.1 and the SMT v51.5 like described in this [issue](https://github.com/arendst/Tasmota/issues/9412). The Shelly is connected to a single wall button, in Germany we say "Taster". The Button can toggle the LED panel on and off but not dim it with a long press. Also the dim range probably not right. Lights turn of at least 54% (can be seen in the screenshot).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly Dimmer 2
- [x] Tasmota binary firmware version number used: 9.2.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: NodeMCU PyFlasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:38:00.145 RSL: stat/tasmota_552164/RESULT = {"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}
13:38:00.353 RSL: stat/tasmota_552164/RESULT = {"Module":{"0":"Shelly Dimmer 2"}}
13:38:00.561 RSL: stat/tasmota_552164/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"5568":"SHD Boot 0"},"GPIO5":{"5600":"SHD Reset"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"193":"Switch_n2"},"GPIO13":{"0":"None"},"GPIO14":{"192":"Switch_n1"},"GPIO15":{"0":"None"},"GPIO16":{"320":"Led_i1"},"GPIO17":{"4736":"ADC Temp"}}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
13:38:29.824 RSL: stat/tasmota_552164/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
13:38:30.026 RSL: stat/tasmota_552164/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
13:38:30.232 RSL: stat/tasmota_552164/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:40:58.346 RSL: stat/tasmota_552164/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_552164","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[11,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:40:58.352 RSL: stat/tasmota_552164/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:03:29","StartupUTC":"2020-12-20T12:37:29","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2020-12-20T12:43:53","SaveCount":70,"SaveAddress":"FA000"}}
13:40:58.357 RSL: stat/tasmota_552164/STATUS2 = {"StatusFWR":{"Version":"9.2.0.1(tasmota)","BuildDateTime":"2020-12-18T12:18:19","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"444/699"}}
13:40:58.364 RSL: stat/tasmota_552164/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box 6490 Cable",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","0A05C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
13:40:58.373 RSL: stat/tasmota_552164/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":25,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
13:40:58.378 RSL: stat/tasmota_552164/STATUS5 = {"StatusNET":{"Hostname":"tasmota_552164-0356","IPAddress":"192.168.17.151","Gateway":"192.168.17.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.17.3","Mac":"8C:AA:B5:55:21:64","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:40:58.382 RSL: stat/tasmota_552164/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_552164","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:40:58.389 RSL: stat/tasmota_552164/STATUS7 = {"StatusTIM":{"UTC":"2020-12-20T12:40:58","Local":"2020-12-20T13:40:58","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:40","Sunset":"16:54"}}
13:40:58.393 RSL: stat/tasmota_552164/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:40:58.399 RSL: stat/tasmota_552164/STATUS10 = {"StatusSNS":{"Time":"2020-12-20T13:40:58","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":26.3},"ENERGY":{"TotalStartTime":"2020-12-20T12:43:53","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0},"TempUnit":"C"}}
13:40:58.406 RSL: stat/tasmota_552164/STATUS11 = {"StatusSTS":{"Time":"2020-12-20T13:40:58","Uptime":"0T00:03:29","UptimeSec":209,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":54,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"FRITZ!Box 6490 Cable","BSSId":"2C:3A:FD:AD:EB:A6","Channel":1,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:10"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Flash a Shelly Dimmer 2 with Tasmota 9.2.0.1 and SMT v51.5. Connect a single button with the Shelly on SW1 and a LED panel with LED driver.

### EXPECTED BEHAVIOUR
The Shelly can be toggle and dim with a single button and the web ui.

### SCREENSHOTS
[image]
[image]
[image]
[image]

### ADDITIONAL CONTEXT
This is the log after configuration of the shelly:

-------------------------------Start the Shelly and configure it----------------------------------------
[... 9 lines cut ...]
12:47:48.486 RSL: tele/tasmota_552164/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_552164-0356","IPAddress":"192.168.17.151"}
12:47:48.488 RSL: tele/tasmota_552164/INFO3 = {"RestartReason":"Software/System restart"}
12:47:48.489 RSL: stat/tasmota_552164/RESULT = {"POWER":"OFF"}
12:47:48.491 RSL: stat/tasmota_552164/POWER = OFF
12:47:51.418 QPC: Reset
12:47:53.424 RSL: tele/tasmota_552164/STATE = {"Time":"2020-12-20T12:47:53","Uptime":"0T00:00:10","UptimeSec":10,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"***","BSSId":"***","Channel":6,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}
12:47:53.431 RSL: tele/tasmota_552164/SENSOR = {"Time":"2020-12-20T12:47:53","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":23.5},"ENERGY":{"TotalStartTime":"2020-12-20T12:43:53","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":14},"TempUnit":"C"}
12:48:38.164 CMD: Backlog SwitchMode1 11; SetOption32 10  Rule1 ON System#boot mem1 + ENDON ON switch1#state=2 DO cmnd/%topic%/POWER TOGGLE ENDON ON switch1#state=4 DO cmnd/%topic%/DIMMER %mem1% ENDON ON switch1#state=5 DO mem1 - ENDON ON switch1#state=6 DO mem1 + ENDON  Rule1 1
12:48:38.185 RSL: stat/tasmota_552164/RESULT = {"SwitchMode1":11}
12:48:38.418 RSL: stat/tasmota_552164/RESULT = {"SetOption32":10}
------------------------------Turn the light on and off with the hardware button--------------------------------
13:03:47.868 RSL: stat/tasmota_552164/RESULT = {"POWER":"ON"}
13:03:47.870 RSL: stat/tasmota_552164/POWER = ON
13:03:52.143 RSL: stat/tasmota_552164/RESULT = {"POWER":"OFF"}
13:03:52.145 RSL: stat/tasmota_552164/POWER = OFF
-------------------------------Dim the light in web ui------------------------------
[... cut ...]
13:04:26.086 RSL: stat/tasmota_552164/RESULT = {"POWER":"ON","Dimmer":35}
13:04:27.064 RSL: stat/tasmota_552164/RESULT = {"POWER":"ON","Dimmer":41}
13:04:27.998 RSL: stat/tasmota_552164/RESULT = {"POWER":"ON","Dimmer":47}
13:04:29.202 RSL: stat/tasmota_552164/RESULT = {"POWER":"ON","Dimmer":50}
13:04:29.897 RSL: stat/tasmota_552164/RESULT = {"POWER":"ON","Dimmer":55}
13:04:30.553 RSL: stat/tasmota_552164/RESULT = {"POWER":"ON","Dimmer":62}`

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-12-20:
It is not the default behaviour to do dimming with a long press. You have to configure.
See https://tasmota.github.io/docs/Buttons-and-Switches/

DasLukas, 2020-12-20:
> It is not the default behaviour to do dimming with a long press. You have to configure.
The dimming range has to be configured https://tasmota.github.io/docs/Commands/#dimmerrange
AND set ledtable 0

Dimming work fine now! Thanks.

> See https://tasmota.github.io/docs/Buttons-and-Switches/

I push this rule to the console:

```
Backlog SwitchMode1 11; SetOption32 10

Rule1
ON System#boot mem1 + ENDON
ON switch1#state=2 DO cmnd/%topic%/POWER TOGGLE ENDON
ON switch1#state=4 DO cmnd/%topic%/DIMMER %mem1% ENDON
ON switch1#state=5 DO mem1 - ENDON
ON switch1#state=6 DO mem1 + ENDON

Rule1 1
```

Isn't it right?

Jason2866, 2020-12-20:
Try
```on switch1#state=2 do Power Toggle endon
on switch1#state=4 do Dimmer %mem1% endon
on switch1#state=5 do mem1 - endon
on switch1#state=6 do mem1 + endon
```

DasLukas, 2020-12-20:
I try this direct in the console and get this output:

```
18:08:14.078 CMD: on switch1#state=2 do Power Toggle endon on switch1#state=4 do Dimmer %mem1% endon on switch1#state=5 do mem1 - endon on switch1#state=6 do mem1 + endon
18:08:14.086 MQT: stat/tasmota_552164/RESULT = {"Command":"Unknown"}
```

Getting Command:Unkown then i tried this:

```
Backlog SwitchMode1 11; SetOption32 10

Rule1
ON System#boot mem1 + ENDON
on switch1#state=2 do Power Toggle endon
on switch1#state=4 do Dimmer %mem1% endon
on switch1#state=5 do mem1 - endon
on switch1#state=6 do mem1 + endon

Rule1 1
```

Still no dimming. Is there no option to realize dimming with a single button? I thing its a normal way to dim lights or isn't it?

Jason2866, 2020-12-20:
Tasmota is a universal ESP82XX firmware. Do NOT expect to flash and everything is working out of the box. When using `switchmode 11` NO rule is needed.
Dont mix different solutions to reach the goal. Please read the link i provided about switches and buttons. Without reading the documentation of Tasmota you probably wont achieve what you want.

closed 2020-12-20 by ascillato2

DasLukas, 2020-12-20:
Finale i solve the problem. Red across the docs/Commands i red about `Backlog` and realize that i miss a semicolon while pushing the following config to the console:

```
Backlog SwitchMode1 11; SetOption32 10; <- missing semicolon

Rule1
ON System#boot mem1 + ENDON
ON switch1#state=2 DO cmnd/%topic%/POWER TOGGLE ENDON
ON switch1#state=4 DO cmnd/%topic%/DIMMER %mem1% ENDON
ON switch1#state=5 DO mem1 - ENDON
ON switch1#state=6 DO mem1 + ENDON

Rule1 1
```

I get the hint when I recognize that the Rule1 space was empty:
```
13:38:29.824 RSL: stat/tasmota_552164/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```

> When using switchmode 11 NO rule is needed.

SwitchMode11 without rule not works for me.

J4nsen, 2021-01-12:
I've had the same exact problem with the same hardware, and I'm still a little bit confused. Should it be working with just setting SwitchMode1 11 or are rules needed? Because the documentation reads like setting SwichMode 11 is enough.

DasLukas, 2021-01-12:
As I write in my post before I use switchmode 11 with the rule above. Now everything works fine. The documentation seems to be not right or the Firmware not works properly with the Shelly dimmer 2.

Maximus48p, 2021-03-03:
> Finale i solve the problem. Red across the docs/Commands i red about `Backlog` and realize that i miss a semicolon while pushing the following config to the console:
>
> ```
> Backlog SwitchMode1 11; [...]

Seems to work perfectly when using a push-button as switch.  ;)
But is there a similar rule when attaching an actual on/off switch and still be able to use the dimmer function?
Otherwise i have to change my on/off switches for push-switches. :(

jeroenst, 2021-05-26:
In the document

> Seems to work perfectly when using a push-button as switch. ;)
> But is there a similar rule when attaching an actual on/off switch and still be able to use the dimmer function?
> Otherwise i have to [...]

In the documentation the semicolon is still missing:

https://tasmota.github.io/docs/Rules/#control-a-dimmer-with-one-switch

Jason2866, 2021-05-26:
There is no `;` missing in the rule example.
Backlog does not need a `;` after last command

jeroenst, 2021-07-14:
> There is no `;` missing in the rule example.
> Backlog does not need a `;` after last command

Actually this doesn't work (Rule 1 isn't stored):

``18:43:31.885 CMD: Backlog SwitchMode 11; SetOption32 10 Rule1 on system#boot mem1 + ENDON ON switch1#state=2 DO power TOGGLE ENDON ON switch1#state=4 DO dimmer %mem1% ENDON ON switch1#state=5 DO mem1 - ENDON ON switch1#state=6 DO mem1 + ENDON  Rule1 1
18:43:31.992 MQT: 0002/TASMOTA_DIMMER01/stat/RESULT = {"SwitchMode1":11}
18:43:32.199 MQT: 0002/TASMOTA_DIMMER01/stat/RESULT = {"SetOption32":10}
18:43:35.853 CMD: rule 1
18:43:35.903 MQT: 0002/TASMOTA_DIMMER01/stat/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}``

But it works with the added semicolon:
``18:44:17.939 CMD: Backlog SwitchMode 11; SetOption32 10; Rule1 on system#boot mem1 + ENDON ON switch1#state=2 DO power TOGGLE ENDON ON switch1#state=4 DO dimmer %mem1% ENDON ON switch1#state=5 DO mem1 - ENDON ON switch1#state=6 DO mem1 + ENDON  Rule1 1
18:44:18.087 MQT: 0002/TASMOTA_DIMMER01/stat/RESULT = {"SwitchMode1":11}
18:44:18.292 MQT: 0002/TASMOTA_DIMMER01/stat/RESULT = {"SetOption32":10}
18:44:18.454 RUL: Stored uncompressed, would compress from 189 to 99 (-48%)
18:44:18.519 MQT: 0002/TASMOTA_DIMMER01/stat/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":189,"Free":322,"Rules":"on system#boot mem1 + ENDON ON switch1#state=2 DO power TOGGLE ENDON ON switch1#state=4 DO dimmer %mem1% ENDON ON switch1#state=5 DO mem1 - ENDON ON switch1#state=6 DO mem1 + ENDON  Rule1 1"}}
18:44:23.484 CMD: rule 1
18:44:23.563 MQT: 0002/TASMOTA_DIMMER01/stat/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":189,"Free":322,"Rules":"on system#boot mem1 + ENDON ON switch1#state=2 DO power TOGGLE ENDON ON switch1#state=4 DO dimmer %mem1% ENDON ON switch1#state=5 DO mem1 - ENDON ON switch1#state=6 DO mem1 + ENDON  Rule1 1"}}``

jeroenst, 2021-07-14:
I added a function for double press full brightness, can't test it at this moment (having no shelly dimmer 2 at my home), so maybe someone else can test it for now?

```
Backlog SwitchMode 11; SetOption32 10;

Rule1
on system#boot mem1 + ENDON
ON switch1#state=2 DO power TOGGLE ENDON
ON switch1#state=4 DO dimmer %mem1% ENDON
ON switch1#state=5 DO mem1 - ENDON
ON switch1#state=6 DO mem1 + ENDON
ON switch1#state=8 DO dimmer 100 ENDON

Rule1 1
```

sponz, 2021-10-06:
@jeroenst 
Hi I am facing the same issue. I have the last version 9.5.0 and I went through all the steps from here: 

https://templates.blakadder.com/shelly_Dimmer_2.html

I used your commands, but it does not work. 

How is your wiring? I am using a push-button between SW1 and L. 

Thanks!

jeroenst, 2021-10-06:
@sponz yes, my pushbutton is between sw1 and L.

Is your rule stored and enabled?

Check with entering Rule 1

barbudor, 2021-10-08:
```
18:44:23.563 MQT: 0002/TASMOTA_DIMMER01/stat/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":189,"Free":322,"Rules":"on system#boot mem1 + ENDON ON switch1#state=2 DO power TOGGLE ENDON ON switch1#state=4 DO dimmer %mem1% ENDON ON switch1#state=5 DO mem1 - ENDON ON switch1#state=6 DO mem1 + ENDON  Rule1 1"}}
```
This is wrong.
You have entered `Rule1 1` as part of the rule instead of a separated command needed to enable the rule

sponz, 2021-10-08:
@jeroenst Just more information based on @barbudor  comment....

I have tried with and without ";" and the result is the same:

15:09:19.155 CMD: Backlog SwitchMode 11; SetOption32 10
15:09:19.224 MQT: dimmer/3C6105E40556000303/RESULT = {"SwitchMode1":11}
15:09:19.434 MQT: dimmer/3C6105E40556000303/RESULT = {"SetOption32":10}
15:10:12.964 CMD: Rule1 on system#boot var1 + ENDON ON switch1#state=2 DO power TOGGLE ENDON ON switch1#state=4 DO dimmer %var1% ENDON ON switch1#state=5 DO var1 - ENDON ON switch1#state=6 DO var1 + ENDON
15:10:12.973 RUL: Stored uncompressed, would compress from 180 to 93 (-48%)
15:10:12.979 MQT: dimmer/3C6105E40556000303/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":180,"Free":331,"Rules":"on system#boot var1 + ENDON ON switch1#state=2 DO power TOGGLE ENDON ON switch1#state=4 DO dimmer %var1% ENDON ON switch1#state=5 DO var1 - ENDON ON switch1#state=6 DO var1 + ENDON"}}
15:13:40.984 CMD: Rule1 1
15:13:40.993 MQT: dimmer/3C6105E40556000303/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":180,"Free":331,"Rules":"on system#boot var1 + ENDON ON switch1#state=2 DO power TOGGLE ENDON ON switch1#state=4 DO dimmer %var1% ENDON ON switch1#state=5 DO var1 - ENDON ON switch1#state=6 DO var1 + ENDON"}}
15:13:41.123 RUL: SWITCH1#STATE=4 performs "dimmer "
15:13:41.130 MQT: dimmer/3C6105E40556000303/RESULT = {"Dimmer":1}
15:13:41.522 RUL: SWITCH1#STATE=4 performs "dimmer "
15:13:41.533 MQT: dimmer/3C6105E40556000303/RESULT = {"Dimmer":1}
15:13:41.931 RUL: SWITCH1#STATE=4 performs "dimmer "
15:13:41.939 MQT: dimmer/3C6105E40556000303/RESULT = {"Dimmer":1}
15:13:42.347 RUL: SWITCH1#STATE=4 performs "dimmer "
15:13:42.358 MQT: dimmer/3C6105E40556000303/RESULT = {"Dimmer":1}
15:13:42.748 RUL: SWITCH1#STATE=4 performs "dimmer "
15:13:42.756 MQT: dimmer/3C6105E40556000303/RESULT = {"Dimmer":1}
15:13:43.150 RUL: SWITCH1#STATE=4 performs "dimmer "
15:13:43.157 MQT: dimmer/3C6105E40556000303/RESULT = {"Dimmer":1}

The hardware is ok, as it was working fine with the original firmware!

jeroenst, 2021-10-08:
From your logging, I think you have a switch that is normally closed... Can you check wiring, or change gpio 12 or 14 (to switch instead of switch_n)..

sponz, 2021-10-09:
@jeroenst Thanks for you help. That was a good hint. I trusted on the push box and the reference was wrong! It is a normally closed :-O 

@barbudor Thanks also for your help. Just using Switchmode 12 instead of Switchmode 11 makes the magic!!

Do you still think that I have to use 'var1 +'?

Thanks a lot for your help and support!

barbudor, 2021-10-11:
Depending if you want that only from the switch or also from a remote MQTT POWER command
I think that the `Dimmer 80` command will necesserally power on the light so not sure this rule above will work as you expect
You may have to play with more rules, storing effectve power state into a var

samezrp, 2022-08-23:
Did anyone manage to run this on-button-dimming feature?
I try to do it as in examples, expecting to have:
-one click toggle on/off
-long click to have light brighter until I keep button pressed;
-another long click until I keep button pressed again;
This is standard shelly behavior.
With all all examples above I have all the time light lit. Pressing button for a short time starts to dimm to max or min to 100 or 1%, there is no way to switch it off, even thou I do it on web page it returns to dimmer 1% value.
BTW where can I find description of state=x switch states?

samezrp, 2022-08-23:
With your config my SONOFF dimmer does:
One press:
```
21:48:20.690 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:20.696 RSL: RESULT = {"POWER":"ON","Dimmer":90}
21:48:21.133 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:21.138 RSL: RESULT = {"POWER":"ON","Dimmer":80}
21:48:21.609 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:21.615 RSL: RESULT = {"POWER":"ON","Dimmer":70}
21:48:22.032 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:22.038 RSL: RESULT = {"POWER":"ON","Dimmer":60}
21:48:22.434 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:22.440 RSL: RESULT = {"POWER":"ON","Dimmer":50}
21:48:22.880 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:22.885 RSL: RESULT = {"POWER":"ON","Dimmer":40}
21:48:23.278 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:23.284 RSL: RESULT = {"POWER":"ON","Dimmer":30}
21:48:23.782 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:23.788 RSL: RESULT = {"POWER":"ON","Dimmer":20}
21:48:24.180 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:24.186 RSL: RESULT = {"POWER":"ON","Dimmer":10}
21:48:24.630 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:24.636 RSL: RESULT = {"POWER":"ON","Dimmer":1}
21:48:25.034 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:25.039 RSL: RESULT = {"POWER":"ON","Dimmer":1}
21:48:25.432 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:25.437 RSL: RESULT = {"POWER":"ON","Dimmer":1}
21:48:25.894 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:25.900 RSL: RESULT = {"POWER":"ON","Dimmer":1}
21:48:26.281 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:48:26.286 RSL: RESULT = {"POWER":"ON","Dimmer":1}
21:48:26.685 RUL: SWITCH1#STATE=4 performs "DIMMER -"
endless

```

Long press:
```
21:49:53.619 RUL: SWITCH1#STATE=7 performs "backlog event upordown=-"
21:49:53.637 RSL: RESULT = {"Event":"Done"}
21:49:53.693 RUL: EVENT#UPORDOWN=- performs "var1 +"
21:49:53.699 RSL: RESULT = {"Var1":"+"}
```

Release long press:
```
21:49:53.193 RUL: SWITCH1#STATE=4 performs "DIMMER -"
21:49:53.198 RSL: RESULT = {"POWER":"ON","Dimmer":1}
21:49:53.619 RUL: SWITCH1#STATE=7 performs "backlog event upordown=-"
21:49:53.637 RSL: RESULT = {"Event":"Done"}
21:49:53.693 RUL: EVENT#UPORDOWN=- performs "var1 +"
21:49:53.699 RSL: RESULT = {"Var1":"+"}
21:50:03.236 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:03.241 RSL: RESULT = {"POWER":"ON","Dimmer":11}
21:50:03.692 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:03.698 RSL: RESULT = {"POWER":"ON","Dimmer":21}
21:50:04.140 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:04.145 RSL: RESULT = {"POWER":"ON","Dimmer":31}
21:50:04.605 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:04.611 RSL: RESULT = {"POWER":"ON","Dimmer":41}
21:50:04.990 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:04.996 RSL: RESULT = {"POWER":"ON","Dimmer":51}
21:50:05.391 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:05.397 RSL: RESULT = {"POWER":"ON","Dimmer":61}
21:50:05.841 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:05.847 RSL: RESULT = {"POWER":"ON","Dimmer":71}
21:50:06.239 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:06.245 RSL: RESULT = {"POWER":"ON","Dimmer":81}
21:50:06.737 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:06.743 RSL: RESULT = {"POWER":"ON","Dimmer":91}
21:50:07.143 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:07.149 RSL: RESULT = {"POWER":"ON","Dimmer":100}
21:50:07.610 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:07.615 RSL: RESULT = {"POWER":"ON","Dimmer":100}
21:50:07.994 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:07.000 RSL: RESULT = {"POWER":"ON","Dimmer":100}
21:50:08.383 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:08.389 RSL: RESULT = {"POWER":"ON","Dimmer":100}
21:50:08.782 RUL: SWITCH1#STATE=4 performs "DIMMER +"
21:50:08.788 RSL: RESULT = {"POWER":"ON","Dimmer":100}
endless
```
And my rule:
```
21:51:37.199 CMD: Rule1
21:51:37.205 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":438,"Free":73,"Rules":"on system#boot do var1 + ENDON on system#boot do var2 0 ENDON on switch1#state=2 do POWER TOGGLE ENDON on switch1#state=2 do event powertoggle=%var2% ENDON on switch1#state=4 do DIMMER %var1% ENDON on switch1#state=7 do event upordown=%var1% ENDON on event#upordown=+ do var1 - ENDON on event#upordown=- do var1 + ENDON on event#powertoggle=0 do DIMMER 50 ENDON on event#powertoggle=0 do var1 - ENDON on power1#state do var2 %value% ENDON"}}
```

There is no way to toggle power at all :/
My GPIO is setup as SWITCH, Tasmote ver 12.1.0, I do not understand this behavior :(

jeroenst, 2022-08-23:
You need to invert the input by using switch_n

samezrp, 2022-08-23:
[image]
I have it set to _n :(

jeroenst, 2022-08-23:
Use template shelly dimmer instead

jeroenst, 2022-08-23:
Try without _n

samezrp, 2022-08-23:
Also done, no change.

jeroenst, 2022-08-23:
Ok, I don't have experience with sonoff d1 and tasmota unfortunately only shelly dimmer 2.

jsponz, 2022-08-23:
@jeroenst Could you share your tasmota configuration? Thanks!

samezrp, 2022-08-23:
I wish to use Shelly, but I have few of them and every one behaves different. In example one of them, without any scene, timer or whatsoever switches off everyday at 11pm sharp. Another ones after problem restore go to ON, even thou they are set to be OFF on power restore...
By configuration you mean dump? I'm not sure if it is safe with stored passwords.

jsponz, 2022-08-23:
have you flashed all of them with the same version? That happened to me with a Shelly 1L, but just one. Other are working fine. 
I meant this configuration: https://user-images.githubusercontent.com/33499054/186266817-5c3dcd00-2369-4d6f-b3b1-7334487731a7.png

samezrp, 2022-08-23:
[Samezrp.txt](https://github.com/arendst/Tasmota/files/9410764/Samezrp.txt)
My json config with password removed.

samezrp, 2022-08-23:
> have you flashed all of them with the same version? That happened to me with a Shelly 1L, but just one. Other are working fine. I meant this configuration: https://user-images.githubusercontent.com/33499054/186266817-5c3dcd00-2369-4d6f-b3b1-7334487731a7.png

But you meant shellies config? I did not try Tasmota with shellies, but you are right, maybe this is an option. I will try tomorrow. If not tomorrow than after 4th of Sept when I'm back from vacation. Thanks.

samezrp, 2022-08-24:
I think I have found reason of this behavior.

Based on:
https://youtu.be/jcdPRbWCGjE?t=409
I used following rule to check on switch state (very helpful to debug):

`rule1 on switch1#state do publish switch_test %value% endon`

On SONOFF D1 there is a bug (I believe) and switch state=4 under SwitchMode1 11 is triggered all the time, no matter if switch is pressed or not.

So in this case any action defined under:

`on switch1#state=4 do DIMMER %var1% ENDON`

will be triggered all the time, depending on dim+ or dim- till 1 or 100% dimming level.

One press gives:
```
**endless**
18:19:06.772 RSL: switch_test = 4
18:19:07.170 RUL: SWITCH1#STATE performs "publish switch_test 4"
18:19:07.174 RSL: switch_test = 4
18:19:07.644 RUL: SWITCH1#STATE performs "publish switch_test 4"
18:19:07.648 RSL: switch_test = 4
18:19:07.916 RUL: SWITCH1#STATE performs "publish switch_test 7"
18:19:07.919 RSL: switch_test = 7
18:19:08.114 RUL: SWITCH1#STATE performs "publish switch_test 5"
18:19:08.117 RSL: switch_test = 5
18:19:08.717 RUL: SWITCH1#STATE performs "publish switch_test 4"
**endless**
```
long press gives:
```
**endless**
18:21:07.274 RSL: switch_test = 4
18:21:07.670 RUL: SWITCH1#STATE performs "publish switch_test 4"
18:21:07.674 RSL: switch_test = 4
18:21:08.072 RUL: SWITCH1#STATE performs "publish switch_test 4"
18:21:08.076 RSL: switch_test = 4
18:21:08.466 RUL: SWITCH1#STATE performs "publish switch_test 6"
18:21:08.470 RSL: switch_test = 6
18:21:08.484 RUL: SWITCH1#STATE performs "publish switch_test 7"
18:21:08.488 RSL: switch_test = 7
18:21:09.470 RUL: SWITCH1#STATE performs "publish switch_test 6"
18:21:09.474 RSL: switch_test = 6
**stop until button is pressed**
18:21:16.022 RUL: SWITCH1#STATE performs "publish switch_test 4"
18:21:16.026 RSL: switch_test = 4
18:21:16.418 RUL: SWITCH1#STATE performs "publish switch_test 4"
18:21:16.422 RSL: switch_test = 4
18:21:16.817 RUL: SWITCH1#STATE performs "publish switch_test 4"
18:21:16.821 RSL: switch_test = 4
**endless**
```
so there is no way to have this SwitchMode1 11 working with endless switch_test = 4 as described in manual.

SwitchMode1 12 could be an option as there is no this constant state = 4 action, however it does not create switch status = 5, so there is no way to change dim direction. It could be done via status=10, however it is not natural and would give another step when dimmed.

It looks like:
18:01:51.130 RSL: RESULT = {"SwitchMode1":12}

**press**
```
18:03:15.114 RUL: SWITCH1#STATE performs "publish switch_test 2"
18:03:15.118 RSL: switch_test = 2
18:03:15.714 RUL: SWITCH1#STATE performs "publish switch_test 10"
18:03:15.718 RSL: switch_test = 10
18:03:15.731 RUL: SWITCH1#STATE performs "publish switch_test 6"
18:03:15.735 RSL: switch_test = 6

```
**long press**
```
18:03:56.032 RSL: switch_test = 4
18:03:56.428 RUL: SWITCH1#STATE performs "publish switch_test 4"
18:03:56.432 RSL: switch_test = 4
18:03:56.827 RUL: SWITCH1#STATE performs "publish switch_test 4"
18:03:56.831 RSL: switch_test = 4
18:03:57.227 RUL: SWITCH1#STATE performs "publish switch_test 4"
18:03:57.231 RSL: switch_test = 4
18:03:57.698 RUL: SWITCH1#STATE performs "publish switch_test 4"
18:03:57.702 RSL: switch_test = 4
18:03:58.098 RUL: SWITCH1#STATE performs "publish switch_test 4"
18:03:58.102 RSL: switch_test = 4
**until button pressed, then**
18:03:58.199 RUL: SWITCH1#STATE performs "publish switch_test 7"
18:03:58.203 RSL: switch_test = 7
18:03:59.198 RUL: SWITCH1#STATE performs "publish switch_test 6"
18:03:59.202 RSL: switch_test = 6
```

samezrp, 2022-08-25:
>[jeroenst](https://github.com/jeroenst)
> This is what I use:

Your, script works perfectly with SONOFF D1 with following settings:
Backlog SwitchMode1 12
SetOption32 10
switch_n
Thank you!

jeroenst, 2022-08-26:
@samezrp nice it works fine, great to hear!