arendst/Tasmota #10204: mDNS: broken i 9.2.0/9.2.0.1 - when self compiling with "#define MDNS_ENABLED true"

Anders-Nielsen, 2020-12-20:
### PROBLEM DESCRIPTION
mDNS not working i 9.2.0/9.2.0.1 when self-compiling it with "#define MDNS_ENABLED true" included in user_config_override.h - SetOption55 is set to 1, but mDNS fails to initialise and does not find a MQTT host. When setting MQTTHost = 0, the device resets.
V. 9.1.0 works without problems

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP-12E/nodeMCU 
- [ ] Tasmota binary firmware version number used: 9.2.0/9.2.0.1
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: PlatformIO/Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:34:50.877 RSL: stat/tasmota_AB44CD/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:34:51.094 RSL: stat/tasmota_AB44CD/RESULT = {"Module":{"0":"Generic"}}
18:34:51.297 RSL: stat/tasmota_AB44CD/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:35:45.877 RSL: stat/tasmota_AB44CD/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_AB44CD","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:35:45.904 RSL: stat/tasmota_AB44CD/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:08:51","StartupUTC":"2020-12-20T17:26:54","Sleep":50,"CfgHolder":4699,"BootCount":3,"BCResetTime":"2020-12-20T18:26:06","SaveCount":6,"SaveAddress":"F6000"}}
18:35:45.937 RSL: stat/tasmota_AB44CD/STATUS2 = {"StatusFWR":{"Version":"9.2.0.1(tasmota)","BuildDateTime":"2020-12-20T18:24:30","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"351/699"}}
18:35:45.958 RSL: stat/tasmota_AB44CD/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AnWireless",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000020","00006000","00000000"]}}
18:35:45.989 RSL: stat/tasmota_AB44CD/STATUS4 = {"StatusMEM":{"ProgramSize":568,"Free":432,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F8AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,8,9,10,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
18:35:46.000 RSL: stat/tasmota_AB44CD/STATUS5 = {"StatusNET":{"Hostname":"tasmota_AB44CD-1229","IPAddress":"192.168.13.132","Gateway":"192.168.13.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.13.100","Mac":"68:C6:3A:AB:44:CD","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:35:46.022 RSL: stat/tasmota_AB44CD/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AB44CD","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:35:46.043 RSL: stat/tasmota_AB44CD/STATUS7 = {"StatusTIM":{"UTC":"2020-12-20T17:35:46","Local":"2020-12-20T18:35:46","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:40","Sunset":"16:54"}}
18:35:46.065 RSL: stat/tasmota_AB44CD/STATUS10 = {"StatusSNS":{"Time":"2020-12-20T18:35:46"}}
18:35:46.074 RSL: stat/tasmota_AB44CD/STATUS11 = {"StatusSTS":{"Time":"2020-12-20T18:35:46","Uptime":"0T00:08:52","UptimeSec":532,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Wifi":{"AP":1,"SSId":"AnWireless","BSSId":"F0:79:59:76:52:80","Channel":3,"RSSI":36,"Signal":-82,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
I just resets everytime MQTTHost is set to 0 or "

00:00:00.050 CFG: Loaded from flash at FA, Count 10
00:00:00.055 QPC: Count 1
00:00:00.057 CFG: CR 351/699, Busy 0
00:00:00.060 SRC: Restart
00:00:00.061 Project tasmota Tasmota Version 9.2.0.1(tasmota)-2_7_4_9(2020-12-20T18:24:30)
00:00:00.217 WIF: Checking connection...
00:00:00.218 WIF: Attempting connection...
00:00:00.554 WIF: Connecting to AP1 XYXYXYXYXY Channel 3 BSSId XYXYXYXYXYXY in mode 11n as tasmota_AB44CD-1229...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1: enable mDNS in under_config_override.h - #define MDNS_ENABLED true
2: compile and upload
3: run
4: Setoption55 1
5: mqtthost 0

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Query the mDNS for a MQTT Host

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-12-20:
As designed. Your user_config_override.h has not all needed entrys.
You have to add too
```
#define USE_DISCOVERY                            // Enable mDNS for the following services (+8k code or +23.5k code with core 2_5_x, +0.3k mem)
```

Anders-Nielsen, 2020-12-20:
Thank you - that did the job - my bad. I will close the issue.

I need mDNS, as I am not in control of the DHCP and the MQTT host changes ip now and then.... I do not know which other options I have.

closed 2020-12-20 by Anders-Nielsen

Jason2866, 2020-12-20:
Getting a fixed IP for the mqqt broker would be good. Maybe you can talk with the admin of the DHCP server to get a fixed IP for the broker.