arendst/Tasmota #10217: SYMFONISK Sound Controller and Sonoff Zb Bridge

jmorris644, 2020-12-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Issues with pairing the SYMFONISK Sound Controller and receiving messages correctly from the device

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): SYMFONISK Sound Controller and Sonoff ZB Bridge
- [ ] Tasmota binary firmware version number used: 9.2.0
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:49:14 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
13:49:14 SRC: Backlog
13:49:14 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
13:49:14 MQT: stat/ZbBridge/RESULT = {"NAME":"Sonoff ZbBridge","GPIO":[320,3552,0,3584,5312,0,0,0,640,576,608,0,32,0],"FLAG":0,"BASE":75}
13:49:14 SRC: Backlog
13:49:14 CMD: Group 0, Index 1, Command "MODULE", Data ""
13:49:14 MQT: stat/ZbBridge/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
13:49:14 SRC: Backlog
13:49:14 CMD: Group 0, Index 1, Command "GPIO", Data "255"
13:49:14 MQT: stat/ZbBridge/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
13:50:10 ZIG: {"ZbEZSPReceived2":"251C94012300000151330746F1FEFFCCCCCC04"}
13:50:10 ZIG: ZbEZSPSentRaw 83
13:50:10 ZIG: {"ZbEZSPReceived":"2300000151330746F1FEFFCCCCCC04"}
13:50:10 ZIG: {"ZbEZSPReceived2":"351C9001240051330746F1FEFFCCCCCC00030000"}
13:50:10 ZIG: ZbEZSPSentRaw 84
13:50:10 ZIG: {"ZbEZSPReceived":"240051330746F1FEFFCCCCCC00030000"}
13:50:10 MQT: tele/ZbBridge/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0xCCCCCCFFFEF14607","ShortAddr":"0x3351","ParentNetwork":"0x0000","Status":0,"Decision":3}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Pair device by clicking the reset button 4 times
2. Any physical action on the device will produce the exact same message in the log.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

1. Need to receive different messages for different physical actions wit the device. ie: moving dial, single, double, triple click

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Initial pairing would not work. I would see the pairing request but it seemed as if the bridge would not reply correctly. After trying a number of things I relflashed the ncp-uart-sw_6.7.6_115200 file. After that I was able to pair the device. But since that time I am not able to get it to function properly

**(Please, remember to close the issue when the problem has been addressed)**

jmorris644, 2020-12-21:
I was able to get it working. I saw a post from someone who had difficulty with another device that was previously paired with a different ZB network. Their solution was to zbforget the device. They tried it a few times and finally they got it to work. I also remembered reading about the Sound Controller that it comes already paired with the IKEA hub for ease of installation. So I tried zbforgetting the device and repairing the device. After two tries it did not successfully pair. So I tried it one more time and not only did it pair but it also now works. Here is the weblog form the last pairing
```

21:30:30 ZIG: {"ZbEZSPReceived2":"472E94012300000051330746F1FEFFCCCCCC04"}
21:30:30 ZIG: ZbEZSPSentRaw 85
21:30:30 ZIG: {"ZbEZSPReceived":"2300000051330746F1FEFFCCCCCC04"}
21:30:30 ZIG: {"ZbEZSPReceived2":"572E9001240051330746F1FEFFCCCCCC0203FFFF"}
21:30:30 ZIG: ZbEZSPSentRaw 86
21:30:30 ZIG: {"ZbEZSPReceived":"240051330746F1FEFFCCCCCC0203FFFF"}
21:30:30 WIF: Checking connection...
21:30:40 ZIG: {"ZbEZSPReceived2":"672E90012300000151330746F1FEFFCCCCCC04"}
21:30:40 ZIG: ZbEZSPSentRaw 87
21:30:40 ZIG: {"ZbEZSPReceived":"2300000151330746F1FEFFCCCCCC04"}
21:30:40 ZIG: {"ZbEZSPReceived2":"772E9001240051330746F1FEFFCCCCCC01000000"}
21:30:40 ZIG: ZbEZSPSentRaw 80
21:30:40 ZIG: {"ZbEZSPReceived":"240051330746F1FEFFCCCCCC01000000"}
21:30:40 MQT: tele/ZbBridge/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0xCCCCCCFFFEF14607","ShortAddr":"0x3351","ParentNetwork":"0x0000","Status":1,"Decision":0}}
21:30:42 ZIG: Zigbee Devices Data saved in EEPROM (16 bytes)
21:30:42 ZIG: {"ZbEZSPReceived2":"072E90012300000151330746F1FEFFCCCCCC04"}
21:30:42 ZIG: ZbEZSPSentRaw 81
21:30:42 ZIG: {"ZbEZSPReceived":"2300000151330746F1FEFFCCCCCC04"}
21:30:43 ZIG: {"ZbEZSPReceived2":"172E9001240051330746F1FEFFCCCCCC01000000"}
21:30:43 ZIG: ZbEZSPSentRaw 82
21:30:43 ZIG: {"ZbEZSPReceived":"240051330746F1FEFFCCCCCC01000000"}
21:30:43 MQT: tele/ZbBridge/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0xCCCCCCFFFEF14607","ShortAddr":"0x3351","ParentNetwork":"0x0000","Status":1,"Decision":0}}
21:30:43 ZIG: {"ZbEZSPReceived2":"272E940162000746F1FEFFCCCCCC"}
21:30:43 ZIG: ZbEZSPSentRaw 83
21:30:43 ZIG: {"ZbEZSPReceived":"62000746F1FEFFCCCCCC"}
21:30:43 ZIG: {"ZbEZSPReceived2":"372E9001450004000013000000000400007AFFDE5133FFFF0C8151330746F1FEFFCCCCCC8002"}
21:30:43 ZIG: ZbEZSPSentRaw 84
21:30:43 ZIG: {"ZbEZSPReceived":"450004000013000000000400007AFFDE5133FFFF0C8151330746F1FEFFCCCCCC8002"}
21:30:43 MQT: tele/ZbBridge/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0xCCCCCCFFFEF14607","ShortAddr":"0x3351","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
21:30:43 ZIG: ZbEZSPSend 3400005133000005000000400100000D01030D5133
21:30:43 ZIG: adding packet to_send, to_ack:7, to_send:7, to_end:7
21:30:43 ZIG: Something to_send, to_ack:7, to_send:7, to_end:0
21:30:43 ZIG: ZbEZSPSentRaw 742F00013400005133000005000000400100000D01030D5133
21:30:43 ZIG: {"ZbEZSPReceived2":"402F80013400009A"}
21:30:43 ZIG: new ack/data received, was 7 now 0
21:30:43 ZIG: freeing packet 7 from memory
21:30:43 ZIG: ZbEZSPSentRaw 85
21:30:43 ZIG: {"ZbEZSPReceived":"3400009A"}
21:30:43 ZIG: {"ZbEZSPReceived2":"502F90013F0006FDFF000013000000000400007A000000"}
21:30:43 ZIG: ZbEZSPSentRaw 86
21:30:43 ZIG: {"ZbEZSPReceived":"3F0006FDFF000013000000000400007A000000"}
21:30:44 ZIG: ZbEZSPSentRaw 87
21:30:44 ZIG: {"ZbEZSPReceived":"450000000005800000400100007BFFDE5133FFFF060D005133010102"}
21:30:44 MQT: tele/ZbBridge/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
21:30:44 ZIG: ZbEZSPSend 34000051330401000001014001000001010700010004000500
21:30:44 ZIG: adding packet to_send, to_ack:0, to_send:0, to_end:0
21:30:44 ZIG: Something to_send, to_ack:0, to_send:0, to_end:1
21:30:44 ZIG: ZbEZSPSentRaw 0730000134000051330401000001014001000001010700010004000500
21:30:44 ZIG: {"ZbEZSPReceived2":"702F90013F00005133000005000000400100009A010000"}
21:30:44 ZIG: ZbEZSPSentRaw 80
21:30:44 ZIG: {"ZbEZSPReceived":"3F00005133000005000000400100009A010000"}
21:30:44 ZIG: {"ZbEZSPReceived2":"013080013400009B"}
21:30:44 ZIG: new ack/data received, was 0 now 1
21:30:44 ZIG: freeing packet 0 from memory
21:30:44 ZIG: ZbEZSPSentRaw 81
21:30:44 ZIG: {"ZbEZSPReceived":"3400009B"}
21:30:45 ZIG: {"ZbEZSPReceived2":"11309001450000040100000101400100007CFFDE5133FFFF35180101040000420E494B4541206F662053776564656E050000421A53594D464F4E49534B20536F756E6420436F6E74726F6C6C657202"}
21:30:45 ZIG: ZbEZSPSentRaw 82
21:30:45 ZIG: {"ZbEZSPReceived":"450000040100000101400100007CFFDE5133FFFF35180101040000420E494B4541206F662053776564656E050000421A53594D464F4E49534B20536F756E6420436F6E74726F6C6C657202"}
21:30:45 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0x3351","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":139,"securityuse":0,"seqnumber":124,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"040000420E494B4541206F662053776564656E050000421A53594D464F4E49534B20536F756E6420436F6E74726F6C6C6572"}}
21:30:45 ZIG: ZbZCLRawReceived: {"0x3351":{"0000/0004":"IKEA of Sweden","0000/0005":"SYMFONISK Sound Controller","Endpoint":1,"LinkQuality":139}}
21:30:45 MQT: tele/ZbBridge/SENSOR = {"ZbReceived":{"0x3351":{"Device":"0x3351","Manufacturer":"IKEA of Sweden","ModelId":"SYMFONISK Sound Controller","Endpoint":1,"LinkQuality":139}}}
21:30:45 ZIG: {"ZbEZSPReceived2":"213090013F00005133040100000101400100009B010000"}
21:30:45 ZIG: ZbEZSPSentRaw 83
[... cut ...]
21:30:51 ZIG: {"ZbEZSPReceived2":"743390013F0000513300002100000040010000A1010000"}
21:30:51 ZIG: ZbEZSPSentRaw 80
21:30:51 ZIG: {"ZbEZSPReceived":"3F0000513300002100000040010000A1010000"}
21:30:51 ZIG: {"ZbEZSPReceived2":"043390013F0006FFFF0401030001FF0001000083000000"}
21:30:51 ZIG: ZbEZSPSentRaw 81
21:30:51 ZIG: {"ZbEZSPReceived":"3F0006FFFF0401030001FF0001000083000000"}
21:30:51 ZIG: {"ZbEZSPReceived2":"143390013F0006FCFF0000360000000000000082000000"}
21:30:51 ZIG: ZbEZSPSentRaw 82
21:30:51 ZIG: {"ZbEZSPReceived":"3F0006FCFF0000360000000000000082000000"}
```

ascillato2, 2020-12-22:
> I was able to get it working...

Great to know that now it is working for you.

closed 2020-12-22 by ascillato2

s-hadinger, 2020-12-22:
Great to hear it now works. Can you please share some commands sent by the device? `Weblog 2` would be enough.

jmorris644, 2020-12-23:
To the best of my knowledge there are 5 different actions possible. !,2, and 3 clicks and rotate clockwise and rotate counterclockwise. Below are the messages received for each using weblog 2

Single click
```
15:50:24 MQT: tele/ZbBridge/SENSOR = {"ZbReceived":{"0x3351":{"Device":"0x3351","0006!02":"","Power":2,"Endpoint":1,"LinkQuality":168}}}
```
Double Click
```
15:50:31 MQT: tele/ZbBridge/SENSOR = {"ZbReceived":{"0x3351":{"Device":"0x3351","0008!02":"000100000000","DimmerStepUp":1,"Endpoint":1,"LinkQuality":168}}}
```

Triple Click
```
15:50:37 MQT: tele/ZbBridge/SENSOR = {"ZbReceived":{"0x3351":{"Device":"0x3351","0008!02":"010100000000","DimmerStepDown":1,"Endpoint":1,"LinkQuality":168}}}
```

Rotate Clockwise
```
15:50:41 MQT: tele/ZbBridge/SENSOR = {"ZbReceived":{"0x3351":{"Device":"0x3351","0008!01":"00C30000","DimmerMove":0,"Endpoint":1,"LinkQuality":94}}}
15:50:41 MQT: tele/ZbBridge/SENSOR = {"ZbReceived":{"0x3351":{"Device":"0x3351","0008!03":"0000","DimmerStop":true,"Endpoint":1,"LinkQuality":100}}}
```

Rotate Counterclockwise
```
15:50:47 MQT: tele/ZbBridge/SENSOR = {"ZbReceived":{"0x3351":{"Device":"0x3351","0008!01":"01C30000","DimmerMove":1,"Endpoint":1,"LinkQuality":94}}}
15:50:47 MQT: tele/ZbBridge/SENSOR = {"ZbReceived":{"0x3351":{"Device":"0x3351","0008!03":"0000","DimmerStop":true,"Endpoint":1,"LinkQuality":100}}}
```

s-hadinger, 2020-12-23:
Nice! Thanks for sharing.

jmorris644, 2020-12-23:
I also posted an update request to the blakadder site.

nagyrobi, 2021-02-17:
Strangely the manual included with a SYMFONISK Sound Controller purchased last week states that for pairing you need to press the reset button for longer than 10 seconds. Didn't see any mention about pressing 4 times under 5 seconds variant, but that made it pair.
However, I got no MQTT messages from the device. Turned out it was very simple to fix: remove the battery and re-insert.
Hope this saves a few hours of hair-pulling for someone.

nagyrobi, 2021-03-01:
> remove the battery and re-insert

Turns out not a permament fix. Returning the controller to Ikea because it's not reliable in this environment.
Maybe a firmware update would fix it but that is nearly impossible.