arendst/Tasmota #10221: Prometheus metrics parse error on tasmota_sensors_totalstarttime

sbelectronics, 2020-12-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Prometheus reports "strconv.ParseFloat: parsing "2020-12-22T04:58:04": invalid syntax". It appears this value is returned by the tasmota_sensors_totalstarttime metric.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): Sonoff S31
- [X ] Tasmota binary firmware version number used: 9.2.0
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: web
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
07:59:45 CMD: Backlog Template; Module; GPIO 255
07:59:45 RSL: stat/tasmota_07ED08/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
07:59:45 RSL: stat/tasmota_07ED08/RESULT = {"Module":{"41":"Sonoff S31"}}
07:59:45 RSL: stat/tasmota_07ED08/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3072":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3104":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

08:00:37 RSL: stat/tasmota_07ED08/STATUS = {"Status":{"Module":41,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_07ED08","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:00:37 RSL: stat/tasmota_07ED08/STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:06:16","StartupUTC":"2020-12-22T06:54:21","Sleep":50,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2020-12-22T04:58:04","SaveCount":39,"SaveAddress":"F8000"}}
08:00:37 RSL: stat/tasmota_07ED08/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T20:49:15","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"372/699"}}
08:00:37 RSL: stat/tasmota_07ED08/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["GUARDIAN","DONOTUSE"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100068000005A00000000000000","00000000","00006000","00000000"]}}
08:00:37 RSL: stat/tasmota_07ED08/STATUS4 = {"StatusMEM":{"ProgramSize":589,"Free":412,"Heap":12,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00104004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,75"}}
08:00:37 RSL: stat/tasmota_07ED08/STATUS5 = {"StatusNET":{"Hostname":"tasmota_07ED08-3336","IPAddress":"192.168.2.201","Gateway":"192.168.2.108","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.108","Mac":"C8:2B:96:07:ED:08","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:00:37 RSL: stat/tasmota_07ED08/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_07ED08","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:00:37 RSL: stat/tasmota_07ED08/STATUS7 = {"StatusTIM":{"UTC":"2020-12-22T07:00:37","Local":"2020-12-22T08:00:37","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:41","Sunset":"16:55"}}
08:00:37 RSL: stat/tasmota_07ED08/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
08:00:37 RSL: stat/tasmota_07ED08/STATUS10 = {"StatusSNS":{"Time":"2020-12-22T08:00:37","ENERGY":{"TotalStartTime":"2020-12-22T04:58:04","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":118,"Current":0.000}}}
08:00:37 RSL: stat/tasmota_07ED08/STATUS11 = {"StatusSTS":{"Time":"2020-12-22T08:00:37","Uptime":"0T00:06:16","UptimeSec":376,"Heap":12,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"GUARDIAN","BSSId":"82:2A:A8:D7:E9:38","Channel":6,"RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:03"}}}
08:00:37 RSL: stat/tasmota_07ED08/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40213019","4023ffd3","40237173","402444d6","4024575c","40215812","402134dc","40248552","402485b0","401000e1","40213514","40244731","40232d3e","401011ca","40232e5b","402494d0","40233012","40233038","40246aed","401011ca","4020fbf5","40230ba2","40100a40","40245874","40101e49"]}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

08:01:34 CMD: weblog 4
08:01:34 RSL: stat/tasmota_07ED08/RESULT = {"WebLog":4}
08:01:34 DMP: F2 5A 02 C6 F0 00 06 04 00 3B F1 04 D7 46 4B A3 48 2C D4 13 61 00 61 1A
08:01:34 HTP: Prometheus
08:01:34 DMP: F2 5A 02 C6 F0 00 06 04 00 3B F1 04 D7 46 4B A3 48 2E 6E 77 61 00 61 1A
08:01:34 DMP: F2 5A 02 C6 F0 00 06 04 00 3B F1 04 D7 46 4B A3 48 30 07 DC 61 00 61 1A
08:01:34 DMP: F2 5A 02 C6 F0 00 06 04 00 3B F1 03 72 05 4B A3 48 31 A1 41 61 00 61 73
08:01:35 DMP: F2 5A 02 C6 F0 00 06 04 00 3B F1 03 72 05 4B A3 48 33 39 A7 61 00 61 73
08:01:35 DMP: F2 5A 02 C6 F0 00 06 04 00 3B F1 03 72 05 4B A3 48 34 D3 0C 61 00 61 73
08:01:35 CFG: Saved to flash at F7, Count 40, Bytes 4096
08:01:35 DMP: F2 5A 02 C6 F0 00 06 04 00 3B F1 03 72 05 4B A3 48 36 6D 70 61 00 61 73
08:01:35 DMP: F2 5A 02 C6 F0 00 06 04 00 3B F1 03 72 05 4B A3 48 38 06 D5 61 00 61 73
08:01:35 DMP: F2 5A 02 C6 F0 00 06 04 00 3B F1 03 72 05 4B A3 48 39 A0 3A 61 00 61 73
08:01:35 DMP: F2 5A 02 C6 F0 00 06 04 00 3B F1 03 72 05 4B A3 48 3B 38 A0 61 00 61 73
08:01:35 DMP: F2 5A 02 C6 F0 00 06 02 00 3B F1 03 72 05 4B A3 48 3C D2 05 61 00 61 71
08:01:35 DMP: F2 5A 02 C6 F0 00 06 02 00 3B F1 03 72 05 4B A3 48 3E 6B 6A 61 00 61 71
[... 74 lines cut ...]
08:01:50 DMP: F2 5A 02 C6 F0 00 06 01 00 3B F1 04 1C 9A 4B A3 48 C0 A8 96 61 00 62 9C
08:01:50 DMP: F2 5A 02 C6 F0 00 06 01 00 3B F1 04 1C 9A 4B A3 48 C2 41 FB 61 00 62 9C
08:01:50 DMP: F2 5A 02 C6 F0 00 06 01 00 3B F1 04 1C 9A 4B A3 48 C3 DB 60 61 00 62 9C
08:01:50 DMP: F2 5A 02 C6 F0 00 06 01 00 3B F1 04 1C 9A 4B A3 48 C5 75 C4 61 00 62 9C
08:01:50 DMP: F2 5A 02 C6 F0 00 06 01 00 3B F1 04 1C 9A 4B A3 48 C7 0F 29 61 00 62 9D
08:01:50 DMP: F2 5A 02 C6 F0 00 06 01 00 3B F1 04 1C 9A 4B A3 48 C8 A8 8E 61 00 62 9C
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Compiled in prometheus module, used with prometheus 2.22.1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Metrics should be processed without error.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sbelectronics, 2020-12-23:
A secondary problem. I patched to code to avoid emitting the problem line (Prometheus now successfully pulls data from the device), but am occasionally seeing exceptions:

```
08:41:10 RSL: tele/tasmota_07ED08/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40213019","40240003","4023718f","40244506","40244c88","402134dc","40248582","402485e0","401000e1","40213514","40244761","40232d3e","402475d4","40232e5b","40106341","40233012","40233038","40246b1d","401011ca","4020fbf5","40230ba2","40100a40","402458a4","40101e49"]}}
```

If I can figure out how to translate these addresses into code references, I'll try to dig up the context of the failure. One of my modules appeared to reboot approximately 160 times over the last day. It caused the power to cycle to the connected outlet each time.

arendst, 2020-12-23:
Disable Prometheus to see if the issue is in there.

If so it is likely caused by a string dedicated to "%d" or a number dedicated to "%s".

Happy hunting.

sbelectronics, 2020-12-23:
I'm trying it now with Prometheus turned off. The crash didn't happen on the first prometheus scrape -- I received valid data several times. Assuming the problem does go away with prometheus disabled, I'll try to perform some manual scrapes with `curl` and see if I can reproduce it. Maybe it's a memory leak or some other cumulative problem.

sbelectronics, 2020-12-23:
I think it is a memory issue. It fails repeatedly every 60 scrapes or so, and I can watch the "Free Memory" on the status page tick down from 27K to around 5K at which point it becomes nonresponsive. I'd guess it's losing around 300 bytes per scrape.

I'm new to the code, so maybe I'm missing something obvious, but in `xsns_75_prometheus.ino`, I see calls like this:

```
                const char *sensor = FormatMetricName(key2.getStr());                                                                                        //cleanup sensor name
                const char *type = FormatMetricName(key3.getStr());    
```

`FormatMetricName` performs a `malloc`, but I'm not seeing the corresponding `free`.

sbelectronics, 2020-12-23:
Added a few `free()` in the right places and the exceptions are gone. 

Diff to fix both issues is here:

```
--- a/tasmota/xsns_75_prometheus.ino
+++ b/tasmota/xsns_75_prometheus.ino
@@ -61,7 +61,7 @@ const char *UnitfromType(const char *type)  // find unit for measurment type
   return "";
 }
 
-const char *FormatMetricName(const char *metric)  // cleanup spaces and uppercases for Prmetheus metrics conventions
+char *FormatMetricName(const char *metric)  // cleanup spaces and uppercases for Prmetheus metrics conventions
 {
   char *formated = (char *)malloc(strlen(metric)+1);
   uint32_t cnt = 0;
@@ -161,10 +161,12 @@ void HandleMetrics(void)
               const char *value = key3.getValue().getStr(nullptr);
               if (value != nullptr && isdigit(value[0]))
               {
-                const char *sensor = FormatMetricName(key2.getStr());                                                                                        //cleanup sensor name
-                const char *type = FormatMetricName(key3.getStr());                                                                                          //cleanup sensor type
+                char *sensor = FormatMetricName(key2.getStr());                                                                                        //cleanup sensor name
+                char *type = FormatMetricName(key3.getStr());                                                                                          //cleanup sensor type
                 const char *unit = UnitfromType(type);                                                                                                       //grab base unit corresponding to type
                 WSContentSend_P(PSTR("# TYPE tasmota_sensors_%s%s gauge\ntasmota_sensors_%s%s{sensor=\"%s\"} %s\n"), type, unit, type, unit, sensor, value); //build metric as "# TYPE tasmota_sensors_%type%_%unit% gauge\ntasmotasensors_%type%_%unit%{sensor=%sensor%"} %value%""
+                free(sensor);
+                free(type);
               }
             }
           }
@@ -173,10 +175,16 @@ void HandleMetrics(void)
             const char *value = value2.getStr(nullptr);
             if (value != nullptr && isdigit(value[0]))
             {
-              const char *sensor = FormatMetricName(key1.getStr());
-              const char *type = FormatMetricName(key2.getStr());
+              char *sensor = FormatMetricName(key1.getStr());
+              char *type = FormatMetricName(key2.getStr());
               const char *unit = UnitfromType(type);
-              WSContentSend_P(PSTR("# TYPE tasmota_sensors_%s%s gauge\ntasmota_sensors_%s%s{sensor=\"%s\"} %s\n"), type, unit, type, unit, sensor, value);
+              if (strcmp(type, "totalstarttime") == 0) {
+                  // this metric causes prometheus of fail
+              } else {
+                  WSContentSend_P(PSTR("# TYPE tasmota_sensors_%s%s gauge\ntasmota_sensors_%s%s{sensor=\"%s\"} %s\n"), type, unit, type, unit, sensor, value);
+              }
+              free(sensor);
+              free(type);
             }
           }
         }
@@ -186,8 +194,9 @@ void HandleMetrics(void)
         const char *value = value1.getStr(nullptr);
         if (value != nullptr && isdigit(value[0] && strcmp(key1.getStr(), "Time") != 0))  //remove false 'time' metric
         { 
-          const char *sensor = FormatMetricName(key1.getStr());
+          char *sensor = FormatMetricName(key1.getStr());       
           WSContentSend_P(PSTR("# TYPE tasmota_sensors_%s gauge\ntasmota_sensors{sensor=\"%s\"} %s\n"), sensor, sensor, value);
+          free(sensor);
         }
       }
     }
```

closed 2020-12-24 by ascillato2