arendst/Tasmota #10223: ESP32 on WiFi Kit 32 , I2C assignment creates additional relay

billygr, 2020-12-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): WiFi Kit 32
- [ ] Tasmota binary firmware version number used: 9.2.0.1(display)
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
6:39:47.778 CMD: Backlog Template; Module; GPIO 255
16:39:47.835 RSL: stat/tasmota_8C3E75/RESULT = {"NAME":"WiFi Kit 32","GPIO":[1,1,1,1,640,1,1,1,1,1,1,608,1,1,1,1,0,1,1,1,0,224,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
16:39:48.045 RSL: stat/tasmota_8C3E75/RESULT = {"Module":{"0":"WiFi Kit 32"}}
16:39:48.297 RSL: stat/tasmota_8C3E75/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"608":"I2C SCL"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"224":"Relay1"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"}}}
16:39:48.342 RSL: stat/tasmota_8C3E75/RESULT = {"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:40:27.520 CMD: Status 0
16:40:27.526 RSL: stat/tasmota_8C3E75/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_8C3E75","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:40:27.551 RSL: stat/tasmota_8C3E75/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32-display.bin","RestartReason":"Software reset CPU","Uptime":"0T00:05:34","StartupUTC":"2020-12-22T15:34:53","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2020-12-22T15:28:44","SaveCount":48}}
16:40:27.583 RSL: stat/tasmota_8C3E75/STATUS2 = {"StatusFWR":{"Version":"9.2.0.1(display)","BuildDateTime":"2020-12-22T14:31:29","Core":"1_0_5_RC4","SDK":"v3.3.4-286-gb4c075169","CpuFrequency":80,"Hardware":"ESP32","CR":"356/699"}}
16:40:27.605 RSL: stat/tasmota_8C3E75/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["billy",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
16:40:27.634 RSL: stat/tasmota_8C3E75/STATUS4 = {"StatusMEM":{"ProgramSize":1078,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAA858E","00041FA1","000000C3","00000000","00000080","00000020","00000000"],"Drivers":"1,2,4,5,8,9,10,13","Sensors":"1,2,5,6"}}
16:40:27.662 RSL: stat/tasmota_8C3E75/STATUS5 = {"StatusNET":{"Hostname":"tasmota_8C3E75-7797","IPAddress":"172.31.0.234","Gateway":"172.31.0.254","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"B4:E6:2D:8C:3E:75","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:40:27.686 RSL: stat/tasmota_8C3E75/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8C3E75","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:40:27.706 RSL: stat/tasmota_8C3E75/STATUS7 = {"StatusTIM":{"UTC":"2020-12-22T15:40:27","Local":"2020-12-22T16:40:27","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:41","Sunset":"16:55"}}
16:40:27.726 RSL: stat/tasmota_8C3E75/STATUS10 = {"StatusSNS":{"Time":"2020-12-22T16:40:27"}}
16:40:27.737 RSL: stat/tasmota_8C3E75/STATUS11 = {"StatusSTS":{"Time":"2020-12-22T16:40:27","Uptime":"0T00:05:34","UptimeSec":334,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"X","BSSId":"X","Channel":7,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Assign an I2C SCL/SDA to GPIO15/GPIO4. Reason the OLED display is connected to these pin on the WiFi Kit 32

Not working (doesn't matter if you assign it to the correct pin or not) it just creates another power/relay
{"NAME":"WiFi Kit 32","GPIO":[1,1,1,1,640,1,1,1,1,1,1,608,1,1,1,1,0,1,1,1,0,224,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
{"NAME":"WiFi Kit 32","GPIO":[1,1,1,1,608,1,1,1,1,1,1,640,1,1,1,1,0,1,1,1,0,224,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}

Working (if you remove the I2C assignment) = just the on board led assigned
{"NAME":"WiFi Kit 32","GPIO":[1,1,1,1,1,1,1,1,1,1,1,608,1,1,1,1,0,1,1,1,0,224,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}

### EXPECTED BEHAVIOUR
Use the display as I2C device without the pins been assigned as power 

### SCREENSHOTS
When you enable the I2C
[image]

### ADDITIONAL CONTEXT
i do understand that ESP32 is experimental

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-12-22:
The extra power is supposed to control oled backlight....

billygr, 2020-12-22:
ok understood, i though that it was an issue. still the devices is not been detected on the i2cscan
16:56:33.010 CMD: i2cscan
16:56:33.060 RSL: stat/tasmota_8C3E75/RESULT = {"I2CScan":"No devices found"}

Followed pin assignment here: https://resource.heltec.cn/download/WiFi_Kit_32/WIFI_Kit_32_pinoutDiagram_V2.pdf
Including the RST

Jason2866, 2020-12-22:
Which OLED driver ic is used? It has to be one of the Tasmota supported ones?
Anyway the setup can not be correct if no i2c device is found.

arendst, 2020-12-22:
Did some tests and oled works fine on ESP32. Both SSD1306 and SH1106 work out-of-the-box.

Make sure the I2C address is recognised correctly. Watch the restart messages where it says like:
```
00:00:00.057 I2C: SH1106 found at 0x3c
```
But first the command ``i2cscan`` should report the correct I2C address. Either 0x3C or 0x3D.

My SSH1106 display setup looks like:
```
18:06:07.891 MQT: stat/esp32d/RESULT = {"Display":{"Model":7,"Width":128,"Height":64,"Mode":5,"Dimmer":1,"Size":2,"Font":0,"Rotate":0,"Refresh":1,"Cols":[21,8],"Rows":8}}
```
Make sure 
```
18:11:54.136 CMD: displayaddress
18:11:54.146 MQT: stat/esp32d/RESULT = {"DisplayAddress1":60}
```
is setup correctly too. (60 = 0x3c)

Also a SSD1306 may need command ``DisplayModel 2``. A SH1106 may need command ``DisplayModel 7``

billygr, 2020-12-22:
Driver in use is SSD1306

The below code in Arduino 1.8.13 works as expected (i can see the Hello World printed on the OLED, which also verifies pinout/connectivity etc)

```
#include <U8x8lib.h>

U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ 15, /* data=*/ 4, /* reset=*/ 16);

void setup()
{
  u8x8.begin();
  u8x8.setFont(u8x8_font_chroma48medium8_r);
  u8x8.drawString(0, 0, "Hello World");
}

void loop()
{

}
```
Does tasmota uses/assumes only the default (first I2C interface on the ESP32 ?) 
SDA (default is GPIO 21)
SCL (default is GPIO 22)

billygr, 2020-12-22:
And another one not sure how can help it.

If i don't pull the OLED RST high, the device is not available on the I2C (meaning it will not report the address 0x3C)

```
#include <Wire.h>

void setup()
{
  pinMode(16, OUTPUT); // OLED RST PIN
  digitalWrite(16, HIGH);
  Serial.begin (115200);  
  Wire.begin (4, 15);   // sda= GPIO_4 /scl= GPIO_15
}

void Scanner ()
{
  Serial.println ();
  Serial.println ("I2C scanner. Scanning ...");
  byte count = 0;

  Wire.begin();
  for (byte i = 8; i < 120; i++)
  {
    Wire.beginTransmission (i);          // Begin I2C transmission Address (i)
    if (Wire.endTransmission () == 0)  // Receive 0 = success (ACK response) 
    {
      Serial.print ("Found address: ");
      Serial.print (i, DEC);
      Serial.print (" (0x");
      Serial.print (i, HEX);
      Serial.println (")");
      count++;
    }
  }
  Serial.print ("Found ");      
  Serial.print (count, DEC);
  Serial.println (" device(s).");
}

void loop()
{
  Scanner ();
  delay (1000);
}
```

billygr, 2020-12-23:
Applying the same strategy to Tasmota eventually it works
GPIO16 = > Output Hi
SSD1306 found at 0x3c

I don't know if it should be considered a bug or a documentation update

ascillato2, 2020-12-23:
Please, help us and edit the docs. Thanks.

closed 2020-12-23 by ascillato2