arendst/Tasmota #10231: Crash after some minutes/hours

nodiaque, 2020-12-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hello,

I have a nodemcu that I flashed tasmota release with tasmotizer with erase option. Flashing work great. In the config, I select generic then assign 3 relay and 3 buttons. This all work fine, until I actually connect everything.

When everything is connected, sooner or later, the firmware will crash beyond repair. This take some minutes, regardless of if I'm using it or not. Even if I power down or hit reset, it never restart and I need to flash back tasmota.

I tried other firmware and no problem. I bought another nodemcu to see if the problem is with the nodemcu but I doubt. I tried leaving it running without anything connected and it never crashed, but one I have my button (ttp223) and 5v relay (iduino 5v relay) connected, it will crash,.even with only 1.

Current connection is

D0-2 ttp223, d3-5 relay.

Im using a USB 12v to 5v converter and power the node thru the USB. Then, I used the vin pin to power the relay and the 3.3v pins to power the buttons. Everything connected to gnd pins. This work well with my other nodemcu using wled.

Thank you
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): nodemcu
- [X ] Tasmota binary firmware version number used: 9.2.0
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [X ] Flashing tools used: Tasmotiser
- [X ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
Cannot since its crashed

```
- [ X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No rules

```
- [X ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
Cannot since its crashed

```
- [ X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Cannot since its crashes

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash, set WiFi and mqtt. Then, put but to  from D0 to D2 and relay from D3 to D5. Press button, wait a bit, sooner or later, web interface cannot be reached, nearly bricked nodemcu. 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Just continue to work! 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-12-23:
Cannot reproduce your issue. Does the relay have a driver circuit and a flashback diode?
Your electrical setup seems not to be okay.

sfromis, 2020-12-23:
One risk factor with a button in standard setup is that if Tasmota sees it as "pushed" for long time (default 40 seconds), it will trigger a device recovery (full reset).

closed 2020-12-23 by ascillato2