arendst/Tasmota #10233: Power consumption api - no units

silentbil, 2020-12-23:
### PROBLEM DESCRIPTION
Hi,
im doing a web request to get the power consumption:
192.168.1.16/cm?cmnd=status%208

the response is:
{
"StatusSNS": {
"Time": "2020-12-23T08:28:42",
"ENERGY": {
"TotalStartTime": "2020-11-23T08:22:08",
"Total": 4.957,
"Yesterday": 2.497,
"Today": 1.201,
"Power": 1,
"ApparentPower": 3,
"ReactivePower": 3,
"Factor": 0.21,
"Voltage": 253,
"Current": 0.013
}
}
}

as you can see, there are no units for the values.
so "Power": 1 , I know its supposed to be watt but for example: "Today": 1.201 can be watt or kilo watt

is there another api to get the status with units?
thanks

Core/SDK Version | 2_7_4_1/2.2.2-dev(38a443e

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
{
    "Status": {
        "Module": 0,
        "DeviceName": "Tasmota",
        "FriendlyName": [
            "Tasmota"
        ],
        "Topic": "tasmota_D3F517",
        "ButtonTopic": "0",
        "Power": 0,
        "PowerOnState": 0,
        "LedState": 1,
        "LedMask": "FFFF",
        "SaveData": 1,
        "SaveState": 1,
        "SwitchTopic": "0",
        "SwitchMode": [
            0,
            0,
            0,
            0,
            0,
            0,
            0,
            0
        ],
        "ButtonRetain": 0,
        "SwitchRetain": 0,
        "SensorRetain": 0,
        "PowerRetain": 0
    },
    "StatusPRM": {
        "Baudrate": 115200,
        "SerialConfig": "8N1",
        "GroupTopic": "tasmotas",
        "OtaUrl": "http://ota.tasmota.com/tasmota/release/tasmota.bin",
        "RestartReason": "Power On",
        "Uptime": "1T12:18:13",
        "StartupUTC": "2020-12-21T19:22:17",
        "Sleep": 50,
        "CfgHolder": 4617,
        "BootCount": 13,
        "BCResetTime": "2020-11-23T08:22:08",
        "SaveCount": 210,
        "SaveAddress": "FA000"
    },
    "StatusFWR": {
        "Version": "8.5.0(tasmota)",
        "BuildDateTime": "2020-10-10T09:40:03",
        "Boot": 31,
        "Core": "2_7_4_1",
        "SDK": "2.2.2-dev(38a443e)",
        "CpuFrequency": 80,
        "Hardware": "ESP8266EX",
        "CR": "350/699"
    },
    "StatusLOG": {
        "SerialLog": 2,
        "WebLog": 2,
        "MqttLog": 0,
        "SysLog": 0,
        "LogHost": "",
        "LogPort": 514,
        "SSId": [
            "virus 2.4",
            ""
        ],
        "TelePeriod": 300,
        "Resolution": "558180C0",
        "SetOption": [
            "00208009",
            "0405C8000100060000005A00000000000000",
            "08000000",
            "00006000",
            "00000000"
        ]
    },
    "StatusMEM": {
        "ProgramSize": 595,
        "Free": 408,
        "Heap": 24,
        "ProgramFlashSize": 2048,
        "FlashSize": 2048,
        "FlashChipId": "1540A1",
        "FlashFrequency": 40,
        "FlashMode": 3,
        "Features": [
            "00000809",
            "8FDAE797",
            "04368001",
            "000000CD",
            "010013C0",
            "C000F981",
            "00004004",
            "00000000"
        ],
        "Drivers": "1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37",
        "Sensors": "1,2,3,4,5,6"
    },
    "StatusNET": {
        "Hostname": "tasmota_D3F517-5399",
        "IPAddress": "192.168.1.16",
        "Gateway": "192.168.1.1",
        "Subnetmask": "255.255.255.0",
        "DNSServer": "192.168.1.1",
        "Mac": "C4:4F:33:D3:F5:17",
        "Webserver": 2,
        "WifiConfig": 4,
        "WifiPower": 17.0
    },
    "StatusMQT": {
        "MqttHost": "",
        "MqttPort": 1883,
        "MqttClientMask": "DVES_%06X",
        "MqttClient": "DVES_D3F517",
        "MqttUser": "DVES_USER",
        "MqttCount": 0,
        "MAX_PACKET_SIZE": 1200,
        "KEEPALIVE": 30
    },
    "StatusTIM": {
        "UTC": "2020-12-23T07:40:30",
        "Local": "2020-12-23T08:40:30",
        "StartDST": "2020-03-29T02:00:00",
        "EndDST": "2020-10-25T03:00:00",
        "Timezone": "+01:00",
        "Sunrise": "08:42",
        "Sunset": "16:56"
    },
    "StatusPTH": {
        "PowerDelta": [
            0,
            0,
            0
        ],
        "PowerLow": 0,
        "PowerHigh": 0,
        "VoltageLow": 0,
        "VoltageHigh": 0,
        "CurrentLow": 0,
        "CurrentHigh": 0
    },
    "StatusSNS": {
        "Time": "2020-12-23T08:40:30",
        "ENERGY": {
            "TotalStartTime": "2020-11-23T08:22:08",
            "Total": 4.957,
            "Yesterday": 2.497,
            "Today": 1.201,
            "Power": 1,
            "ApparentPower": 3,
            "ReactivePower": 3,
            "Factor": 0.21,
            "Voltage": 253,
            "Current": 0.013
        }
    },
    "StatusSTS": {
        "Time": "2020-12-23T08:40:30",
        "Uptime": "1T12:18:13",
        "UptimeSec": 130693,
        "Heap": 26,
        "SleepMode": "Dynamic",
        "Sleep": 50,
        "LoadAvg": 19,
        "MqttCount": 0,
        "POWER": "OFF",
        "Wifi": {
            "AP": 1,
            "SSId": "virus 2.4",
            "BSSId": "94:E4:BA:4B:DF:F4",
            "Channel": 6,
            "RSSI": 86,
            "Signal": -57,
            "LinkCount": 2,
            "Downtime": "0T00:00:12"
        }
    }
}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-12-23:
Yes, as designed. It is always "V", "A", "W" and "kWh"  See WebUi.

silentbil, 2020-12-23:
oh its const
thanks!

closed 2020-12-23 by silentbil