arendst/Tasmota #10237: Shelly 2.5 restarts itself

yinonco, 2020-12-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I get occasional restarts on the device with `RestartReason` = `Hardware Watchdog`.
I use the device as a shutter controller so sometimes when I try to open the shutter to 50% for example and a restart occurs in the middle of the motion, it stops for a fraction of a second and then keeps opening to 100% because of the restart.
I really have no idea what causes the restarts and how to prevent them.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly 2.5
- [X] Tasmota binary firmware version number used: 9.1.0.2
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:13:59.372 CMD: Backlog Template; Module; GPIO 255
15:13:59.375 SRC: WebConsole from 192.168.1.23
15:13:59.378 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
15:13:59.448 SRC: Backlog
15:13:59.451 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
15:13:59.458 MQT: room/stat/blind_one/RESULT = {"NAME":"Shelly 2.5 (buttons)","GPIO":[320,0,34,0,224,129,0,0,0,128,0,225,0,4736],"FLAG":0,"BASE":18}
15:13:59.691 SRC: Backlog
15:13:59.694 CMD: Group 0, Index 1, Command "MODULE", Data ""
15:13:59.700 MQT: room/stat/blind_one/RESULT = {"Module":{"0":"Shelly 2.5 (buttons)"}}
15:13:59.949 SRC: Backlog
15:13:59.952 CMD: Group 0, Index 1, Command "GPIO", Data "255"
15:13:59.961 MQT: room/stat/blind_one/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"34":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"129":"Button_in2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"128":"Button_in1"},"GPIO14":{"0":"None"},"GPIO15":{"225":"Relay2"},"GPIO16":{"0":"None"},"GPIO17":{"4736":"ADC Temp"}}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
15:15:22.643 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
15:15:22.710 SRC: Backlog
15:15:22.713 CMD: Group 0, Index 1, Command "RULE", Data ""
15:15:22.720 MQT: room/stat/blind_one/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":124,"Free":387,"Rules":"ON button1#state=11 DO publish room/covers openall ENDON ON button2#state=11 DO publish room/covers closeall ENDON"}
15:15:22.941 SRC: Backlog
15:15:22.944 CMD: Group 0, Index 2, Command "RULE", Data ""
15:15:22.951 MQT: room/stat/blind_one/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
15:15:23.149 SRC: Backlog
15:15:23.152 CMD: Group 0, Index 3, Command "RULE", Data ""
15:15:23.158 MQT: room/stat/blind_one/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:16:25.454 CMD: Status 0
15:16:25.456 SRC: WebConsole from 192.168.1.23
15:16:25.460 CMD: Group 0, Index 1, Command "STATUS", Data "0"
15:16:25.466 MQT: room/stat/blind_one/STATUS = {"Status":{"Module":0,"DeviceName":"Blind One","FriendlyName":["Blind One","Tasmota2"],"Topic":"blind_one","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
15:16:25.474 MQT: room/stat/blind_one/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Hardware Watchdog","Uptime":"0T01:45:44","StartupUTC":"2020-12-23T12:30:41","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2020-12-02T16:54:17","SaveCount":189,"SaveAddress":"F4000"}}
15:16:25.482 MQT: room/stat/blind_one/STATUS2 = {"StatusFWR":{"Version":"9.2.0.1(tasmota)","BuildDateTime":"2020-12-19T16:30:43","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"441/699"}}
15:16:25.490 MQT: room/stat/blind_one/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Cohen_HB",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0002C029","2805C8000100060000005A00000000000000","40000200","00006000","00000000"]}}
15:16:25.499 MQT: room/stat/blind_one/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
15:16:25.508 MQT: room/stat/blind_one/STATUS5 = {"StatusNET":{"Hostname":"blind_one-1609","IPAddress":"192.168.1.51","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"D8:BF:C0:1A:26:49","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:16:25.513 MQT: room/stat/blind_one/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.112","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1A2649","MqttUser":"smarthome","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:16:25.521 MQT: room/stat/blind_one/STATUS7 = {"StatusTIM":{"UTC":"2020-12-23T14:16:25","Local":"2020-12-23T15:16:25","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:42","Sunset":"16:56"}}
15:16:25.528 MQT: room/stat/blind_one/STATUS10 = {"StatusSNS":{"Time":"2020-12-23T15:16:25","ANALOG":{"Temperature":61.5},"Shutter1":{"Position":100,"Direction":0,"Target":100},"TempUnit":"C"}}
15:16:25.533 MQT: room/stat/blind_one/STATUS11 = {"StatusSTS":{"Time":"2020-12-23T15:16:25","Uptime":"0T01:45:44","UptimeSec":6344,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Cohen_HB","BSSId":"98:1E:19:7F:3C:1E","Channel":1,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:05"}}}
15:16:25.539 MQT: room/stat/blind_one/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":226,"Close":213,"50perc":50,"Delay":0,"Opt":"0100","Calib":[172,369,608,877,1000],"Mode":"0"}}}

```
[... 66 lines cut ...]
13:30:56.466 SHT: time: 262, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 261000, next_stop 0, target: 452000,  max_vel_change 0, dir: 1
13:30:56.512 SHT: time: 263, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 262000, next_stop 0, target: 452000,  max_vel_change 0, dir: 1
13:30:56.570 SHT: time: 264, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 263000, next_stop 0, target: 452000,  max_vel_change 0, dir: 1
13:30:56.621 SHT: time: 265, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 264000, next_stop 0, target: 452000,  max_vel_change 0, dir: 1
15:00:48.183 WIF: Checking connection...
15:00:52.384 MQT: room/tele/blind_one/STATE = {"Time":"2020-12-23T15:00:52","Uptime":"0T01:30:11","UptimeSec":5411,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Cohen_HB","BSSId":"98:1E:19:7F:3C:1E","Channel":1,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:00:52.394 MQT: room/tele/blind_one/SENSOR = {"Time":"2020-12-23T15:00:52","ANALOG":{"Temperature":61.5},"Shutter1":{"Position":100,"Direction":0,"Target":100},"TempUnit":"C"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
It happens by itself sometimes during an opening or a closing (it is possible that it happens on standby as well but I am not aware of this because the shutter just stay in place)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
no restarts

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-12-23:
Please use latest release or development version.

yinonco, 2020-12-23:
I had this with the latest release so I moved to the latest development and the problem remains.
I will try to update again

yinonco, 2020-12-23:
Ok, so I have just updated to latest development version from http://ota.tasmota.com/tasmota/tasmota.bin.gz
and the problem still remains.

```
21:13:49.280 SHT: time: 89, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 358632, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:49.331 SHT: time: 90, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 357571, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:49.383 SHT: time: 91, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 356510, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:49.434 SHT: time: 92, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 355449, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.023 SHT: time: 93, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 354388, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.075 SHT: time: 94, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 353327, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.128 SHT: time: 95, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 352266, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.131 MQT: Attempting connection...
21:13:50.149 MQT: Connected
21:13:50.154 MQT: room/tele/blind_one/LWT = Online (retained)
21:13:50.158 MQT: room/cmnd/blind_one/POWER = 
21:13:50.162 MQT: Subscribe to room/cmnd/blind_one/#
21:13:50.165 MQT: Subscribe to room/cmnd/tasmotas/#
21:13:50.168 MQT: Subscribe to cmnd/DVES_1A2649_fb/#
21:13:50.171 MQT: Unsubscribe from homeassistant/status
21:13:50.176 MQT: room/tele/blind_one/INFO1 = {"Module":"Shelly 2.5 (buttons)","Version":"9.2.0.1(tasmota)","FallbackTopic":"cmnd/DVES_1A2649_fb/","GroupTopic":"room/cmnd/tasmotas/"}
21:13:50.187 MQT: room/tele/blind_one/INFO2 = {"WebServerMode":"Admin","Hostname":"blind_one-1609","IPAddress":"192.168.1.51"}
21:13:50.199 MQT: room/tele/blind_one/INFO3 = {"RestartReason":"Hardware Watchdog"}
21:13:50.207 MQT: room/stat/blind_one/RESULT = {"POWER1":"OFF"}
21:13:50.212 MQT: room/stat/blind_one/POWER1 = OFF (retained)
21:13:50.217 MQT: room/stat/blind_one/RESULT = {"POWER2":"ON"}
21:13:50.224 MQT: room/stat/blind_one/POWER2 = ON (retained)
21:13:50.237 SHT: time: 97, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 350144, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.287 SHT: time: 98, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 349083, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.335 SHT: time: 99, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 348022, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.386 SHT: time: 100, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 346961, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.390 SHT: Shutter 1 Real 346961, Start 452000, Stop 0, Dir -1, Delay 0, Rtc 5.00 [s], Freq 0, PWM 0
21:13:50.395 MQT: room/stat/blind_one/RESULT = {"Shutter1":{"Position":70,"Direction":-1,"Target":0}}
21:13:50.445 SHT: time: 101, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 345900, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.491 SHT: time: 102, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 344839, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.544 SHT: time: 103, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 343778, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.725 SHT: time: 107, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 339534, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.811 SHT: time: 109, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 337412, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.869 SHT: time: 110, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 336351, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.919 SHT: time: 111, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 335290, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:50.972 SHT: time: 112, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 334229, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:51.023 SHT: time: 113, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 333168, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:51.075 SHT: time: 114, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 332107, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:51.126 SHT: time: 115, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 331046, next_stop 0, target: 0,  max_vel_change 0, dir: -1
[... cut ...]
21:13:53.733 SHT: time: 167, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 275874, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:53.786 SHT: time: 168, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 274813, next_stop 0, target: 0,  max_vel_change 0, dir: -1
21:13:53.838 SHT: time: 169, toBeAcc 0, current_stop_way 0,vel_cur 0, vel_max 0, act_vel_change 1000, min_runtime_ms 0, act.pos 273752, next_stop 0, target: 0,  max_vel_change 0, dir: -1
```

yinonco, 2020-12-23:
Ok, I got it, my mistake, I wrote 9.1.0.2 while the true version I have is 9.2.0.1 :)
[image]

I will fix it in the original post

arendst, 2020-12-24:
You might want to try the default Shelly 2.5 template as from what I see above your template contains non-standard GPIOs.
```
{"NAME":"Shelly 2.5","GPIO":[320,0,32,0,224,193,0,0,640,192,608,225,3456,4736],"FLAG":0,"BASE":18}
```

yinonco, 2020-12-24:
I am almost sure I had the problem with that one as well but I will give it a try and update

yinonco, 2020-12-24:
does that mean I cannot use the inputs of the shelly as a "button" instead of a "switch"?

arendst, 2020-12-24:
No. It means the ADE IRQ will be controlled as it is supposed to be.

yinonco, 2020-12-24:
I am not sure I understand you.
Anyway, after changing it to the template you provided it still happens

```lua
15:18:44.354 MQT: Attempting connection...
15:18:44.997 MQT: Connected
15:18:45.000 MQT: room/tele/blind_one/LWT = Online (retained)
15:18:45.003 MQT: room/cmnd/blind_one/POWER = 
15:18:45.009 MQT: room/tele/blind_one/INFO1 = {"Module":"Shelly 2.5","Version":"9.2.0.1(tasmota)","FallbackTopic":"cmnd/DVES_1A2649_fb/","GroupTopic":"room/cmnd/tasmotas/"}
15:18:45.019 MQT: room/tele/blind_one/INFO2 = {"WebServerMode":"Admin","Hostname":"blind_one-1609","IPAddress":"192.168.1.51"}
15:18:45.032 MQT: room/tele/blind_one/INFO3 = {"RestartReason":"Hardware Watchdog"}
```

yinonco, 2020-12-24:
Oh, I got it, you talked about the current sensor.
I removed it because I thought I read somewhere that the "hardware watchdog" restarts occur because of the sensors so Ive tried to disable it to see if it would help. (obviously, it did not :) )

yinonco, 2020-12-24:
By the way, I have the same issue with both of my shellys 2.5 so I don't believe it is a hardware defect

Roxxor91, 2020-12-24:
I just flashed my first Shelly 2.5 via MGOS to tasmota. Firmware 9.2, using the template from the template database. Uptime 3 hours, so no reboot, I switched it on 3 hours ago.
You use the switch as a button by using [SwitchMode](https://tasmota.github.io/docs/Buttons-and-Switches/#switchmode). SwitchMode 3 works well.

yinonco, 2020-12-25:
I am using it as a shutter and it happens when I use it (not when on standby)
Plus, I have the issue also when I set the shelly inputs to switch instead of button - so I believe it is not the cause

schrej-zz, 2020-12-26:
I also have random reboots when l press a switch to OFF.
I ignored it, as I changed poweronstate so that it not noticeable for me anymore. No shuttermode, just switch.

arendst, 2020-12-26:
@yinonco so it happens if you configure it as a button AND as a switch. Is your hardware switch a momentary one or an On/Off one

@schrej-zz pls report your used template. Is your hardware switch a momentary one or an On/Off one?

yinonco, 2020-12-26:
@arendst That's correct and I use a momentary switch

yinonco, 2020-12-26:
@arendst Thanks for your help. I've updated to the newest development (which should contain your commit to my understanding) and I still get restart sometimes.
Can you please give a short explanation about the watchdog restarts? (I couldn't find documentation about the watchdog itself)
I assume that this is a software watchdog in tasmota that can cause a restart for various reasons - so maybe it is possible to add to the log the reason for why the watchdog restarted the system

yinonco, 2020-12-26:
As for the `ButtonDebounce` and `ButtonDebounce` both are configured to 50

bolislav2, 2020-12-27:
See the DHCP settings. When the lease is updated, the socket breaks. The problem of the kernel. Try using a static IP, or change the lease time in the router settings

yinonco, 2020-12-27:
@bolislav2 I got one device with dynamic IP and one with static and I get restarts on both

Jason2866, 2020-12-27:
@bolislav2  No, renew of the DHCP lease do not break connection to the device.

bolislav2, 2020-12-27:
> Нет НИКАКИХ проблем с используемым ядром Tasmota 2.7.4.9 (на основе ядра 2.7.4 Arduino ESP8266).
There is! I had the same problem as the link.  https://github.com/arendst/Tasmota/issues/6816   I can show you. With a Kinetic router and 60 seconds of DHCP lease 
If you don't believe me, I can reproduce the problem.

Jason2866, 2020-12-27:
You have a problem with your router. Why does thousands of installations have zero issues

yinonco, 2020-12-27:
Anyway, this is not true for my case for a few reasons:

1. I have the issue both on dynamic and static IP configuration
2. My DHCP lease is for 2 hours and I don't see a restart every 2 hours
3. **This one is important - I see the problem only when the shutter is moving**. I didn't send commands for 2 days to one of my shutters and I didn't get any restarts

Frogmore42, 2020-12-28:
> This one is important - I see the problem only when the shutter is moving. I didn't send commands for 2 days to one of my shutters and I didn't get any restarts

That points to a questionable power supply and/or interference/noise from the rolling shutter motor causing issues.  But, could be something else.

yinonco, 2020-12-28:
As for the power supply, it is connected to the 220v directly.
About the motor - I have two different motors

closed 2021-01-10 by github-actions[bot]

uchristo, 2021-02-08:
Has this issue been closed due to being fixed? I'm encountering the same issue (hardware watchdog Restart on some Shelly2.5  in the middle of a movement)....

yinonco, 2021-02-08:
No, I still have this issue

IlmLV, 2021-03-02:
> By the way, I have the same issue with both of my shellys 2.5 so I don't believe it is a hardware defect

@yinonco actually just till today I was having the same issue with my shelly 2.5, only I was using it as single relay switch and for some reason it kept restarting when I switched it off with around 15% chance
There is around 10 shellys currently at my apartment and this was only one with this issue, so I replaced shelly, updated configuration to match the old one and for now looks like I am not able to reproduce this issue
btw, I am not using Tasmota, just original latest firmware

revk, 2021-03-18:
I am having this issue, with an ordinary light switch (momentary)!

PPP01, 2021-06-15:
I have exaclty the same problem with several shellies 2.5 and tasmota 9.4.0 (used as shutters). It restarts on nearly every drive and is unusesable

schrej-zz, 2021-06-15:
@PPP01 ,
does this happen, if it#s build in somewhere?
I'm asking as I had the same issues, when it was build in a flush-mount box under pressure.
I had no problems, when it was outside or in a box with enough space.
My assumption is, that it's not related to Tasmota, but to the circumstances, being very small.
This might result in shortcuts or other unwanted physical connections, when pressing e.g. a light switch.

uchristo, 2021-06-15:
Mine (I got 12 Shelly2.5, all used for shutters) are all inside the installation socket in the wall behind the two-way mechanical toggle switch. Some of them restart the action more often than others do.  I doubt that there are shortcuts. I would suppose it's something related to the environmental temperature.

PPP01, 2021-06-15:
Mine were installed in half-wall boxes in sheetrock walls. But I also had the restarts when they were hanging outside the wall (on the fly). With the original firmware, there are no problems.

In all the places where there are no multiple boxes, I have Maxico roller shutter controls with Tasmota 9.4. They don't have these problems at all.

stefanbode, 2021-07-01:
The shutter software uses during movement a 50ms interrupt. This could be an issue for the restart, but never seen before. When there is NO operation and the system justs restarts the only explanation I have is the current measure sensor. In this case the the reboot should also happen with setoption80 OFF. I have one shelly 2.5 flashed with 9.4 and using the two relay to control the coffee machine. No reboots there as far as I can see. current uptime >5d.

stefanbode, 2021-07-01:
There is also a temperature sensor. As far as I remember there is a heat protection included in the standard. Might this be your restart reason? Can you monitor the temperature before the reboot?

uchristo, 2021-07-01:
Is there a definite value for a temperature-caused software defined reboot? Some of my devices are running at ~57degrees in idle. If the threshold is maybe 60 degrees, it‘s possible to reach this on movement.

stefanbode, 2021-07-02:
As far as I know it is 80 or something.should be in the Shelly documentation of tasmota.therefore reboot only when there is load on the system

foobar26, 2021-07-18:
I'm having exactly the same issue with two out of 5 shelly2.5 which I'm also using for shutters. Had this for one of them since the beginning. I doubt that temperature is the reason. The two operate at around 60-65°C, but I have two others at the same temperature without any issue.
The second one was operating with the original Shelly FW until yesterday, when I used OTA (https://github.com/yaourdt/mgos-to-tasmota) to migrate it to tasmota 8.5.1. With the original FW it worked like a charm. Now with tasmota on it, it sometimes restart during operation.
By the way, all of them are at 8.5.1.
The first one showed this problem since I started with tasmota at 7.1.2.

foobar26, 2021-07-18:
By the way I'm also using some Teepao shutter switches with Tasmota on it and they never showed this issue. Seems Shelly related.

uchristo, 2021-08-16:
Anybody got a clue on how to track down this issue? Shall I setup a local syslog and debug everything herein?

revk, 2021-08-16:
> Anybody got a clue on how to track down this issue? Shall I setup a local syslog and debug everything herein?

I think https://uk.rs-online.com/web/p/rc-network-capacitors/6167682/

uchristo, 2021-08-16:
> > Anybody got a clue on how to track down this issue? Shall I setup a local syslog and debug everything herein?
> 
> I think https://uk.rs-online.com/web/p/rc-network-capacitors/6167682/

Pardon? Why should a rc-network cure an issue which is not present on the original firmware? Less power-consuming on original shelly-fw?

revk, 2021-08-16:
> Pardon? Why should a rc-network cure an issue which is not present on the original firmware? Less power-consuming on original shelly-fw?

OK no idea then.

Expello, 2022-02-01:
I have the Shelly 2.5 for my roller shutters in operation and these constant reboots have driven me crazy.
I have tried all possible settings....nothing has helped.

Then I did the following and was actually the solution.

I have removed the button from the template that goes to GPIO2.
After that was peace...no more crashes on all Shelly´s! 

I think the background is that in the narrow switch box the button is pressed and held unintentionally and it led to the unusual behavior. Since a month no more reboots! I am happy!

[image]

uchristo, 2022-02-01:
Great news. Did the same on my 12 shutters right now and am curious what happens now. I even set up a syslog-Server inside my LAN to collect Debugs from the connected shellies..... Without any trace to the exact error.
Will update you in a week.

foobar26, 2022-02-01:
Just tried this on my end. Unfortunately I still have unwanted reboots during operation for the shutter that showed this effect very steadily before already :-\

PPP01, 2022-02-01:
I tried too, but i still have those reboots, too. In the meantime, I have replaced all but one shelly. This one operates at a small shutter only and do not reboot too often. But every 3-12 days, I have an unwanted reboot.

Expello, 2022-02-01:
> Habe das gerade bei mir ausprobiert. Leider habe ich immer noch ungewollte Neustarts während des Betriebs für den Shutter, der diesen Effekt schon vorher sehr stetig zeigte :-\

How many Shelly´s 2.5 are you using? And the reboots are every hour...that's heavy, would rather assume a hardware defect...

foobar26, 2022-02-01:
> How many Shelly´s 2.5 are you using? And the reboots are every hour...that's heavy, would rather assume a hardware defect...

I have 4 shellys and two of them show this behavior. The one I tested with has a very deterministic behavior.
Since the other one is also showing these restarts I really doubt that it is a hardware issue. Especially because for the second one I flashed the old shelly HW on it and there I don't experience any issues with restarts now.

The reboots are not every hour, they most likely appear if the shutter is operating. But there were also times when the device was rebooting without any shutter going up or down.
The one I tested with is rebooting very reliably after moving the shutter down and up again, sometimes even if I just close it, but that's more rare.

stefanbode, 2022-02-02:
As the owner of the shutter code, can you check if you compile it with 160MHz operation speed, if this makes a difference? The shutter uses a RTC timer but this is very well optimized to avoid hardware watchdog. Also the behavior is not seen on wemos d1 and similar devices. I also would try to disable the onboard current sensor. This might be also a possible root cause. Please do not change both! Change on one device the speed and on the other disable in configuration the current sensor. Looking forward to the results.

stefanbode, 2022-02-02:
The behavior after restart is totally normal. Nothing to investigate here.

foobar26, 2022-02-05:
Hi Stefan,
thanks for the hint. I just compiled with 160mhz cpu but no change. The device still reboots after I close and open it again.
Here the log when it restarts.
How do I disable the current sensor?
```
00:00:00 CFG: Loaded from flash at F7, Count 7159
00:00:00 QPC: Count 1
00:00:00 I2C: ADE7953 found at 0x38
00:00:00 Project tasmota Shelly-bad Version 8.5.1(tasmota)-2_7_4_9
00:00:00 RSL: stat/shelly-bad/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0}}
00:00:00 RSL: stat/shelly-bad/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0}}
00:00:00 WIF: Connecting to AP1 downset-new Channel 6 BSSId FC:EC:DA:87:56:6D in mode 11N as Shelly-bad...
00:00:01 RSL: stat/shelly-bad/RESULT = {"Shutter1":{"Position":97,"Direction":-1,"Target":0}}
00:00:01 RSL: stat/shelly-bad/RESULT = {"Shutter1":{"Position":96,"Direction":-1,"Target":0}}
00:00:02 RSL: stat/shelly-bad/RESULT = {"Shutter1":{"Position":91,"Direction":-1,"Target":0}}
00:00:03 RSL: stat/shelly-bad/RESULT = {"Shutter1":{"Position":86,"Direction":-1,"Target":0}}
00:00:03 WIF: Connected
18:58:00 HTP: Web server active on Shelly-bad with IP address 192.168.200.33
18:58:00 RSL: stat/shelly-bad/RESULT = {"Shutter1":{"Position":82,"Direction":-1,"Target":0}}
18:58:01 MQT: Attempting connection...
18:58:01 MQT: Connected
18:58:01 MQT: tele/shelly-bad/LWT = Online (retained)
18:58:01 MQT: cmnd/shelly-bad/POWER = 
18:58:01 MQT: tele/shelly-bad/INFO1 = {"Module":"Shelly 2.5","Version":"8.5.1(tasmota)","FallbackTopic":"cmnd/Shelly_bad_fb/","GroupTopic":"cmnd/tasmotas/"}
18:58:01 MQT: tele/shelly-bad/INFO2 = {"WebServerMode":"Admin","Hostname":"Shelly-bad","IPAddress":"192.168.200.33"}
18:58:01 MQT: tele/shelly-bad/INFO3 = {"RestartReason":"Hardware Watchdog"}
18:58:01 MQT: stat/shelly-bad/RESULT = {"POWER1":"OFF"}
18:58:01 MQT: stat/shelly-bad/POWER1 = OFF
18:58:01 MQT: stat/shelly-bad/RESULT = {"POWER2":"ON"}
18:58:01 MQT: stat/shelly-bad/POWER2 = ON
18:58:01 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":77,"Direction":-1,"Target":0}}
18:58:02 QPC: Reset
18:58:02 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":72,"Direction":-1,"Target":0}}
18:58:03 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":67,"Direction":-1,"Target":0}}
18:58:04 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":63,"Direction":-1,"Target":0}}
18:58:05 MQT: tele/shelly-bad/STATE = {"Time":"2022-02-05T18:58:05","Uptime":"0T00:00:11","UptimeSec":11,"Heap":27,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":100,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"downset-new","BSSId":"FC:EC:DA:87:56:6D","Channel":6,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:05"}}
18:58:05 MQT: tele/shelly-bad/SENSOR = {"Time":"2022-02-05T18:58:05","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":57.9},"ENERGY":{"TotalStartTime":"2020-01-31T20:08:38","Total":1.513,"Yesterday":0.001,"Today":0.003,"Period":0,"Power":[0,149],"ApparentPower":[0,155],"ReactivePower":[0,40],"Factor":[0.00,0.97],"Frequency":50,"Voltage":226,"Current":[0.000,0.680]},"Shutter1":{"Position":100,"Direction":-1,"Target":0},"TempUnit":"C"}
18:58:05 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":57,"Direction":-1,"Target":0}}
18:58:06 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":53,"Direction":-1,"Target":0}}
18:58:07 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":48,"Direction":-1,"Target":0}}
18:58:08 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":44,"Direction":-1,"Target":0}}
18:58:09 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":39,"Direction":-1,"Target":0}}
18:58:10 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":34,"Direction":-1,"Target":0}}
18:58:11 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":29,"Direction":-1,"Target":0}}
18:58:12 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":25,"Direction":-1,"Target":0}}
18:58:13 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":20,"Direction":-1,"Target":0}}
18:58:14 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":15,"Direction":-1,"Target":0}}
18:58:15 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":10,"Direction":-1,"Target":0}}
18:58:16 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":6,"Direction":-1,"Target":0}}
18:58:17 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":1,"Direction":-1,"Target":0}}
18:58:18 MQT: stat/shelly-bad/RESULT = {"POWER2":"OFF"}
18:58:18 MQT: stat/shelly-bad/POWER2 = OFF
18:58:18 MQT: stat/shelly-bad/SHUTTER1 = 0
18:58:18 MQT: stat/shelly-bad/RESULT = {"Shutter1":{"Position":0,"Direction":0,"Target":0}}

```

foobar26, 2022-02-05:
Ha, sorry, I just set the three GPIOs to 0 and disabled energy monitoring.
But still no change. After opening + closing the shutter it restarts during operation :-\

uchristo, 2022-02-05:
Same here. For my devices it seems like disabling button on GPIO2 cured the issue on the devices which are not operated with switches but timer/rule only. But it‘s been just 3 days since I edited the template …..

uchristo, 2022-02-28:
The issue persists on Version 9.5 / 10.1 for all my devices.

stefanbode, 2022-02-28:
I saw some bad sideeffects when optimizing the zero-cross dimmer driver for tasmota. `savedata 0` might be a workaround. If power state or shutter state is written to the flash and other stuff is happening this can cause crashes. I git this on the COUNTER. As long as you do not use STEPPER motors this should work. Anyhow. Can you give this option a try and let us know if it makes a difference? If yes we can work around this later and fix it.

stefanbode, 2022-02-28:
I do see in a picture above that GPIO16 is connected to ADE7953 IRQ (this sounds like an interrrupt). If yes the mentioned savedata 0 is a good hint to avoid the reboot.

foobar26, 2022-03-01:
Thanks for your proposal, Stefan. I configured it yesterday on my most problematic Shelly, but the problem unfortunately still exists. :-\

sander1988, 2022-03-02:
I also have a hardware watchdog restart issue on the same hardware (Shelly 2.5) and see some similarity between this my issue.

My logs are being send to a syslog server so I can see what's happening just before the restart. Mine always crashes directly after a rule publishes something to MQTT. I noticed your rule also publishes to MQTT. 

So I'm wondering, can this and mine issue be related to MQTT publishing? I might be wrong. But I wanted to share this information. I will continue doing some more tests. But will take some time as it only crashes once or twice a week.

---

```
...
Feb 28 22:49:46 XXXXXX-3730 ESP-RUL: SWITCH1#STATE=1 performs "Power1 ON"
Feb 28 22:49:46 XXXXXX-3730 ESP-MQT: hc-tasmota/XXXXXX/POWER1 = {"POWER1":"ON"}
Feb 28 22:49:46 XXXXXX-3730 ESP-MQT: hc-tasmota/XXXXXX/POWER1 = ON
Feb 28 22:49:46 XXXXXX-3730 ESP-RUL: SWITCH1#STATE=1 performs "Backlog RuleTimer1 0; Publish hc-tasmota/XXXXXX/POWER3 ON"
Feb 28 22:49:46 XXXXXX-3730 ESP-MQT: hc-tasmota/XXXXXX/RULETIMER = {"T1":0,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
Feb 28 22:49:46 XXXXXX-3730 ESP-MQT: hc-tasmota/XXXXXX/POWER3 = ON
---------------> CRASH <---------------
Feb 28 22:49:58 XXXXXX-3730 ESP-I2C: ADE7953 found at 0x38
Feb 28 22:49:58 XXXXXX-3730 ESP-Project hc-tasmota XXXXXX Version 9.3.0(tasmota)-2_7_4_9(2021-11-30T16:03:31)
Feb 28 22:49:58 XXXXXX-3730 ESP-WIF: Connecting to AP1 XXXXXX Channel 11 BSSId XX:XX:XX:XX:XX:XX in mode 11n as XXXXXX-3730...
Feb 28 22:49:59 XXXXXX-3730 ESP-WIF: Connected
...
````

[56 more replies]