arendst/Tasmota #10243: Waveshare epaper 2.9 does not show any output

and-zak, 2020-12-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

My Waveshare epaper 2.9 Rev 2.1 module does not show any output with Tasmota.
https://www.waveshare.com/wiki/2.9inch_e-Paper_Module

1. The HW module has no defects: I have successfully made it work with an Ardunio UNO R3 with GxEPD2 with a 4-line SPI connection.
2. After this check I have soldered the jumper to support 3-line SPI and connected it with a Wemos D1 mini.
3. The Tasmota log reports: EPD: SoftSPI CS 4, CLK 14, MOSI 13
4.  I have added a log command in the splash screen section of the code, which proofs that it is passing this section (without showing any output on the display).
5. With a DSO nano oscilloscope I can see periodic activity on all three SPI lines when set to DisplayMode 1.
6. Besides using USB power, I have ruled out undervoltage by powering the Wemos and the epaper with an external 3.3 V power supply.

What else could I do to make it work?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

16:44:56 CMD: Group 0, Index 1, Command "BACKLOG", Data "template"
16:44:56 SRC: Backlog
16:44:56 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
16:44:56 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

No rules used.

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

16:45:38 CMD: Status 0
16:45:38 SRC: WebConsole from 192.168.1.113
16:45:38 CMD: Group 0, Index 1, Command "STATUS", Data "0"
16:45:38 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"wemos-1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:45:38 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:29:22","StartupUTC":"2020-12-23T15:16:16","Sleep":50,"CfgHolder":4617,"BootCount":149,"BCResetTime":"2020-11-30T10:21:38","SaveCount":586,"SaveAddress":"FA000"}}
16:45:38 RSL: STATUS2 = {"StatusFWR":{"Version":"9.1.0.2(tasmota)","BuildDateTime":"2020-12-22T13:39:34","Boot":31,"Core":"2_7_4_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"448/699"}}
16:45:38 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"MqttLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["office",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
16:45:38 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":654,"Free":348,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FF0E79F","04368921","00000CCF","050013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,6,7,8,9,10,12,13,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,9,45"}}
16:45:38 RSL: STATUS5 = {"StatusNET":{"Hostname":"wemos-1","IPAddress":"192.168.1.141","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"40:F5:20:28:12:C1","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:45:38 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-12-23T15:45:38","Local":"2020-12-23T16:45:38","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:42","Sunset":"16:56"}}
16:45:38 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-12-23T16:45:38"}}
16:45:38 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-12-23T16:45:38","Uptime":"0T00:29:22","UptimeSec":1762,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"office","BSSId":"9C:C7:A6:34:04:7E","Channel":6,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

00:00:00 CFG: Loaded from flash at FA, Count 586
00:00:00 QPC: Count 1
00:00:00 EPD: SoftSPI CS 4, CLK 14, MOSI 13
00:00:00 EPD: Splash Screen complete
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 9.1.0.2(tasmota)-2_7_4_7(2020-12-22T13:39:34)
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 office Channel 6 BSSId 9C:C7:A6:34:04:7E in mode 11N as wemos-1...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Connected
00:00:01 RUL: WIFI#CONNECTED performs "Backlog Dimmer 30; Color2 2"
00:00:01 SRC: Rule
00:00:01 CMD: Group 0, Index 1, Command "BACKLOG", Data "Dimmer 30; Color2 2"
00:00:01 SRC: Backlog
00:00:01 CMD: Group 0, Index 1, Command "DIMMER", Data "30"
00:00:01 RSL: RESULT = {"Command":"Unknown"}
00:00:02 HTP: Web server active on wemos-1 with IP address 192.168.1.141
00:00:02 RSL: INFO1 = {"Module":"Generic","Version":"9.1.0.2(tasmota)","FallbackTopic":"cmnd/DVES_2812C1_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:02 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"wemos-1","IPAddress":"192.168.1.141"}
00:00:02 RSL: INFO3 = {"RestartReason":"Software/System restart"}
00:00:02 RSL: RESULT = {"POWER":"ON"}
00:00:02 RSL: POWER = ON
00:00:02 SRC: Backlog
00:00:02 CMD: Group 0, Index 2, Command "COLOR", Data "2"
00:00:02 RSL: RESULT = {"Command":"Unknown"}
00:00:03 NTP: No reply
16:46:28 RTC: UTC 2020-12-23T15:46:28, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
16:46:31 QPC: Reset
16:46:31 RSL: STATE = {"Time":"2020-12-23T16:46:31","Uptime":"0T00:00:08","UptimeSec":8,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"office","BSSId":"9C:C7:A6:34:04:7E","Channel":6,"RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:03"}}
16:46:32 HTP: Main Menu
16:46:33 APP: Boot Count 150
16:46:33 CFG: Saved to flash at F9, Count 587, Bytes 4096
16:46:34 HTP: Console

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Connect Waveshare epaper 2.9 (configured for 3-line SPI) with Wemos D1 mini
2. Connect with USB power

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expect to see a splash screen when booting up.
I expect to see output when setting DisplayMode 1

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

None.

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I could not find an epaper 2.9 datasheet that specifically addresses HW Rev2.1
The only information I could find, might not be related:

From: https://www.waveshare.com/wiki/2.9inch_e-Paper_Module
"If the board you have is printed with Rev2.1, module enter low-ultra mode after DEV_Module_Exit(). (as we test, the current is about 0 in this mode);"

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-12-24:
@and-zak 
i only have a more than 2 years old version of this display and it now works about 2 years without issues.
i recompiled with current dev version and it also works without any problems.
it may have to do with the new features of the controller (zero power ??? sleep mode)
since we do not use reset, may be the board is in a kind of deep sleep mode?
try to toggle the reset line by hand and if that works i will implement support for the reset pin.
the driver however never enters ultra low power mode.

i did let my unused lines open without any issues but
according to the data sheet the not used dc line should be tied low ?

besides that i have no other ideas why yours may fail.

and-zak, 2020-12-24:
@gemu2015 
I have tried to toggle the reset line and to pull DC to low; one after another and also both at the same time. Unfortunately without any success.

closed 2020-12-24 by ascillato2