arendst/Tasmota #10244: Module 48 or SetOption92 1 with inverted CT values

stleusc, 2020-12-23:
### PROBLEM DESCRIPTION
LampUx light that supports Brightness + CT using Module 48 mode.
Everythings works BUT the CT is 'inverted'. Meaning, when setting it to cold, it is warm and when set to warm it is cold. And all in between.
GPIO5 -> PWM1
GPIO12 -> PWM2

Somehow I would need to tell the SW to do the CT the opposite way. Or simply change the meaning of 500..137 or whatever the exact range was.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): LampUx light
- [X ] Tasmota binary firmware version number used: 9.2.0
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [X ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
16:57:07 MQT: tasmota/stat/fireplace_1/RESULT = {"NAME":"LampUx","GPIO":[0,0,0,0,0,416,0,0,417,0,0,0,0,0],"FLAG":0,"BASE":48}
16:57:07 MQT: tasmota/stat/fireplace_1/RESULT = {"Module":{"0":"LampUx"}}
16:57:07 MQT: tasmota/stat/fireplace_1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"416":"PWM1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"417":"PWM2"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stleusc, 2020-12-23:
literally in file xdrv_04_light.ino
// Xiaomi Philips bulbs follow a different scheme:
      cur_col_10[cw1] = light_state.getCT10bits();
cur_col_10[cw1] is in range of 0..1023 and I think my use case simply would need to do:
cur_col_10[cw1] = 1023 - cur_col_10[cw1]; // this will flip the range around and I guess will flip warm and cold

should that not be a generic function? Surprise that can not be done already. I must have missed that?

stleusc, 2020-12-23:
OMG! I feel so stupid... Simply changing the port from PWM to PWM_i and the problem was fixed.
I knew it had to be there. Sorry for wasting your time on this...

s-hadinger, 2020-12-23:
Closing as it is solved.

closed 2020-12-23 by s-hadinger