arendst/Tasmota #10248: RF commands sent fast are ignored

clau-bucur, 2020-12-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Most of the RF commands issues fast (about two times per second) are ignored.
I use a SONOFF 4key remote, and if I press the same key fast, most of the commands are not relayed to MQTT.
The bridge receives the command, the red led turns on for each of them. I also enabled the DEBUG mode for syslog and I see tasmota receives the ESP-DMP data, however it does not pass it over to MQTT.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): SONOFF RF Bridge
- [x] Tasmota binary firmware version number used: 9.2.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:41:46.001 MQT: tasmota/stat/rfbridge/RESULT = {"NAME":"Bridge No Led","GPIO":[32,3200,1,3232,1,1,0,0,1,0,1,0,0,0],"FLAG":0,"BASE":25}
00:41:46.221 MQT: tasmota/stat/rfbridge/RESULT = {"Module":{"25":"Sonoff Bridge"}}
00:41:46.433 MQT: tasmota/stat/rfbridge/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:42:41.618 MQT: tasmota/stat/rfbridge/STATUS = {"Status":{"Module":25,"DeviceName":"RF Bridge","FriendlyName":["RF Bridge"],"Topic":"rfbridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:42:41.626 MQT: tasmota/stat/rfbridge/STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:01:50","StartupUTC":"2020-12-23T22:40:51","Sleep":50,"CfgHolder":4617,"BootCount":72,"BCResetTime":"2020-04-08T21:13:18","SaveCount":325,"SaveAddress":"FB000"}}
00:42:41.634 MQT: tasmota/stat/rfbridge/STATUS2 = {"StatusFWR":{"Version":"9.2.0.1(tasmota)","BuildDateTime":"2020-12-23T17:56:14","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"442/699"}}
00:42:41.641 MQT: tasmota/stat/rfbridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":4,"LogHost":"xx","LogPort":514,"SSId":["xx","xx"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A009","2805C8000100060000005A00000000000000","00800000","00006000","00000000"]}}
00:42:41.652 MQT: tasmota/stat/rfbridge/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
00:42:41.659 MQT: tasmota/stat/rfbridge/STATUS5 = {"StatusNET":{"Hostname":"rfbridge","IPAddress":"192.168.xx","Gateway":"192.168.xx","Subnetmask":"255.255.255.0","DNSServer":"192.168.xx","Mac":"xx","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:42:41.665 MQT: tasmota/stat/rfbridge/STATUS6 = {"StatusMQT":{"MqttHost":"xx","MqttPort":1883,"MqttClientMask":"rfbridge","MqttClient":"rfbridge","MqttUser":"xx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:42:41.674 MQT: tasmota/stat/rfbridge/STATUS7 = {"StatusTIM":{"UTC":"2020-12-23T22:42:41","Local":"2020-12-24T00:42:41","StartDST":"2020-03-29T03:00:00","EndDST":"2020-10-25T04:00:00","Timezone":99,"Sunrise":"09:42","Sunset":"17:57"}}
00:42:41.682 MQT: tasmota/stat/rfbridge/STATUS10 = {"StatusSNS":{"Time":"2020-12-24T00:42:41"}}
00:42:41.690 MQT: tasmota/stat/rfbridge/STATUS11 = {"StatusSTS":{"Time":"2020-12-24T00:42:41","Uptime":"0T00:01:50","UptimeSec":110,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xx","BSSId":"xx","Channel":10,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:46:20.338 SRC: WebConsole from 192.168.1.153
00:46:20.341 CMD: Group 0, Index 1, Command "WEBLOG", Data "4"
00:46:20.347 MQT: tasmota/stat/rfbridge/RESULT = {"WebLog":4}
00:46:20.691 CFG: Saved to flash at FA, Count 326, Bytes 4096
00:46:25.361 DMP: A4 29 54 01 68 04 1A 17 EB 74 55
00:46:25.369 MQT: tasmota/tele/rfbridge/RESULT = {"Time":"2020-12-24T00:46:25","RfReceived":{"Sync":10580,"Low":360,"High":1050,"Data":"17EB74","RfKey":"None"}}
00:46:25.579 DMP: A4 29 4A 01 68 04 24 17 EB 74 55
00:46:26.145 DMP: A4 29 54 01 68 04 24 17 EB 74 55
00:46:26.653 DMP: A4 29 5E 01 68 04 24 17 EB 74 55
00:46:27.127 DMP: A4 29 68 01 68 04 24 17 EB 74 55
00:46:31.493 DMP: A4 29 5E 01 68 04 24 17 EB 74 55
00:46:31.499 MQT: tasmota/tele/rfbridge/RESULT = {"Time":"2020-12-24T00:46:31","RfReceived":{"Sync":10590,"Low":360,"High":1060,"Data":"17EB74","RfKey":"None"}}
00:46:32.469 DMP: A4 29 68 01 68 04 24 17 EB 74 55
00:46:33.736 DMP: A4 29 72 01 68 04 24 17 EB 74 55
00:46:33.742 MQT: tasmota/tele/rfbridge/RESULT = {"Time":"2020-12-24T00:46:33","RfReceived":{"Sync":10610,"Low":360,"High":1060,"Data":"17EB74","RfKey":"None"}}
00:46:34.215 DMP: A4 29 5E 01 72 04 24 17 EB 74 55
00:46:34.624 DMP: A4 29 5E 01 68 04 24 17 EB 74 55
00:46:35.093 WIF: Checking connection...
00:46:35.108 DMP: A4 29 68 01 68 04 24 17 EB 74 55
00:46:35.527 DMP: A4 29 68 01 72 04 24 17 EB 74 55

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Use a RF remote and press the same button a few times per second.
Only the first press is detected as RfReceived event.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
All of the RF commands should be translated to a proper MQTT RfReceived event.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-12-24:
As designed.

Most remotes send their commands many times hoping one will be detected. If tasmota detects one it will delay further detection for about 2 seconds.

If you want you may change line 27 in file xdrv_06_snfbridge.ino from the default 2000 mSecs:
```
const uint32_t SFB_TIME_AVOID_DUPLICATE = 2000;  // Milliseconds
```
Happy toying.

clau-bucur, 2020-12-24:
Thanks! I changed that down to 200ms, built the firmware and now it works as I had hoped.
Would it be possible to make this value configurable? Otherwise I'd need a custom build would be required on every new release.

closed 2020-12-24 by ascillato2

swissjon, 2022-03-20:
Did this setting make its way to the ui/command settings?

clau-bucur, 2022-05-30:
Nope.