arendst/Tasmota #10252: Rule on counter trigger executed all the time and not only on counter change

to-scho, 2020-12-25:
### PROBLEM DESCRIPTION
Rule on counter trigger executed all the time and not only on counter change. According to documentation a rule could be triggered on sensor change. In case of a counter that does not work as described. A rule with a trigger "on counter#c1" is triggered continuously even when there is no counter change. One could limit trigger frequency by "on tele-counter#c1" but that would not fit every use case. Maybe that was already discussed before but I haven't found that. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Basic
- [x] Tasmota binary firmware version number used: 9.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
08:04:07 MQT: stat/gas_wasser/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
08:04:07 MQT: stat/gas_wasser/RESULT = {"Module":{"1":"Sonoff Basic"}}
08:04:08 MQT: stat/gas_wasser/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"352":"Counter1"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
  Configuration output here:

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
on counter#c1 do publish2 cmnd/%topic%/event c1restore=%value% endon on event#c1restore do backlog var2 %value%; event c1crestore=%var1% endon on event#c1crestore= do backlog var1 1; counter1 %var2% endon"}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:06:16 MQT: stat/gas_wasser/STATUS = {"Status":{"Module":1,"DeviceName":"gas_wasser","FriendlyName":["Tasmota"],"Topic":"gas_wasser","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:06:16 MQT: stat/gas_wasser/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T16:08:17","StartupUTC":"2020-12-24T14:57:59","Sleep":50,"CfgHolder":4617,"BootCount":39,"BCResetTime":"2020-11-17T06:56:46","SaveCount":115,"SaveAddress":"F8000"}}
08:06:16 MQT: stat/gas_wasser/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-17T10:05:31","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"416/699"}}
08:06:16 MQT: stat/gas_wasser/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SPIDER_NET_WLAN_FB",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
08:06:16 MQT: stat/gas_wasser/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
08:06:16 MQT: stat/gas_wasser/STATUS5 = {"StatusNET":{"Hostname":"gas_wasser-6966","IPAddress":"192.168.42.82","Gateway":"192.168.42.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.42.1","Mac":"C4:4F:33:B3:3B:36","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:06:16 MQT: stat/gas_wasser/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.42.149","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B33B36","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:06:16 MQT: stat/gas_wasser/STATUS7 = {"StatusTIM":{"UTC":"2020-12-25T07:06:16","Local":"2020-12-25T08:06:16","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:42","Sunset":"16:57"}}
08:06:16 MQT: stat/gas_wasser/STATUS10 = {"StatusSNS":{"Time":"2020-12-25T08:06:16","COUNTER":{"C1":5343682}}}
08:06:16 MQT: stat/gas_wasser/STATUS11 = {"StatusSTS":{"Time":"2020-12-25T08:06:16","Uptime":"0T16:08:17","UptimeSec":58097,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SPIDER_NET_WLAN_FB","BSSId":"DC:39:6F:03:D0:D4","Channel":1,"RSSI":76,"Signal":-62,"LinkCount":2,"Downtime":"0T00:01:42"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Hav a counter as sensor and setup simple rule "rule1 on counter#c1 do delay endon"

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I would expect that rule is only triggered when there is a counter change. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-12-25:
The docs does not suggest that your JSON trigger is only when the value is changed, it is instead when the value is potentially changed. You could do something like:
`on Counter#C1>%var1% do Backlog var1 %value%; publish2 ... endon`

to-scho, 2020-12-25:
thanks for the tip, I will use it. 
Anyhow, I do not see a reason why a user rule should be triggered when there is no change. Either this is a coding artifact or the idea is to have something executed whenever e.g. a sensor has being processed. This differs from other rule handling like buttons where there is only a trigger when state is changed.

sfromis, 2020-12-25:
Because 1) many pulses a second would produce an unmanageable number of events 2) due to very limited memory, it does not save previous value to compare. If your pulses are button-rare, you could also config it as such.

closed 2020-12-26 by to-scho