arendst/Tasmota #10254: Roller shutter opens by itself after (each) restart

DL6ER, 2020-12-25:
### PROBLEM DESCRIPTION

Roller shutter opened shortly after restarting. This also happens when the restart is triggered because of switching to the strongest WiFi network.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Dual R2
- [X] Tasmota binary firmware version number used: 9.2.0.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: FTDI + Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
{"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
{"Module":{"39":"Sonoff Dual R2"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"225":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"32":"Button1"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [X] Provide the output of this command: `Status 0`:
```lua
{"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":260,"Close":250,"50perc":57,"Delay":0,"Opt":"0100","Calib":[730,447,1907,900,1000],"Mode":"0"}}}
```

### TO REPRODUCE

1. Restart device using `mosquitto_pub -h 127.0.0.1 -t cmnd/rolladen-garten/Restart -m '1'`
2. After a restart, the first relay is enabled = The roller shutter opens

### EXPECTED BEHAVIOUR

The roller shutter should stay closed after a restart. This first appeared to me after setting `SetOption57 1` which - ironically - did not only change the WiFi it is connected to, but also opened the shutter. 

### SCREENSHOTS

``` plain
cmnd/rolladen-garten/Restart 1
stat/rolladen-garten/RESULT {"Restart":"Restarting"} 
tele/rolladen-garten/LWT Offline
tele/rolladen-garten/LWT Online
cmnd/rolladen-garten/POWER (null)
tele/rolladen-garten/INFO1 {"Module":"Sonoff Dual R2","Version":"9.2.0.1(tasmota)","FallbackTopic":"cmnd/DVES_E5C289_fb/","GroupTopic":"cmnd/rolladen/"}
tele/rolladen-garten/INFO2 {"WebServerMode":"Admin","Hostname":"rolladen-garten","IPAddress":"192.168.2.2"}
tele/rolladen-garten/INFO3 {"RestartReason":"Software/System restart"}
stat/rolladen-garten/RESULT {"POWER1":"OFF"}
stat/rolladen-garten/POWER1 OFF
stat/rolladen-garten/RESULT {"POWER2":"OFF"}
stat/rolladen-garten/POWER2 OFF
stat/rolladen-garten/RESULT {"POWER1":"ON"}
stat/rolladen-garten/POWER1 ON
stat/rolladen-garten/RESULT {"ShutterPosition1":100} 
stat/rolladen-garten/RESULT {"Shutter1":{"Position":0,"Direction":1,"Target":104}}
stat/rolladen-garten/RESULT {"Command":"Unknown"}
stat/rolladen-garten/RESULT {"Shutter1":{"Position":0,"Direction":1,"Target":104}}
stat/rolladen-garten/RESULT {"Shutter1":{"Position":1,"Direction":1,"Target":104}}
stat/rolladen-garten/RESULT {"Shutter1":{"Position":2,"Direction":1,"Target":104}}
stat/rolladen-garten/RESULT {"Shutter1":{"Position":3,"Direction":1,"Target":104}}
tele/rolladen-garten/STATE {"Time":"2020-12-25T18:11:18","Uptime":"0T00:00:13","UptimeSec":13,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Mr. Spock","BSSId":"3C:A6:CC:DD:EE:FF","Channel":6,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:07"}}
tele/rolladen-garten/SENSOR {"Time":"2020-12-25T18:11:18","Shutter1":{"Position":0,"Direction":1,"Target":104}}
stat/rolladen-garten/RESULT {"Shutter1":{"Position":4,"Direction":1,"Target":104}}
stat/rolladen-garten/RESULT {"Shutter1":{"Position":6,"Direction":1,"Target":104}}
stat/rolladen-garten/RESULT {"Shutter1":{"Position":11,"Direction":1,"Target":104}}
stat/rolladen-garten/RESULT {"Shutter1":{"Position":15,"Direction":1,"Target":104}}
stat/rolladen-garten/RESULT {"Shutter1":{"Position":20,"Direction":1,"Target":104}}
tasmota/discovery/600194E5C289/config {"ip":"192.168.2.2","dn":"Rolladen Garten","fn":["Tasmota","Tasmota2",null,null,null,null,null,null],"hn":"rolladen-garten","mac":"AABBCCDDEEFF","md":"Sonoff Dual R2","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.2.0.1","t":"rolladen-garten","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[3,3,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0},"lk":1,"lt_st":0,"sho":[4,0,0,0],"ver":1}
tasmota/discovery/600194E5C289/sensors {"sn":{"Time":"2020-12-25T18:11:23","Shutter1":{"Position":0,"Direction":1,"Target":104}},"ver":1}
stat/rolladen-garten/RESULT {"Shutter1":{"Position":24,"Direction":1,"Target":104}}
stat/rolladen-garten/RESULT {"Shutter1":{"Position":29,"Direction":1,"Target":104}}

[ and further going up ... ]
```

### ADDITIONAL CONTEXT

`PowerOnState` was set to `3`, however, even changing it to `0` makes no difference. Especially the automatic opening after enabling option `SetOption57 1` (Wi-Fi network re-scan, alternate AP) was quite surprising because it happened out of nothing.

arendst, 2020-12-26:
Just enable more logging to find out where ``stat/rolladen-garten/RESULT {"POWER1":"ON"}`` is coming from. (i suspect a retained message).

DL6ER, 2020-12-26:
Thanks, it indeed came from a retained message for a second GroupTopic I forgot about. This raises the question if this is expected. What I found is the following:

1. `%topic%` is `shutter-garden`
2. `GroupTopic1` is `shutters`
3. `GroupTopic2 is `shutters-ground-floor`

The following was sent:
```
cmnd/shutters-ground-floor/ShutterSetPosition1 100
cmnd/shutter-garden/ShutterClose1
cmnd/shutters/ShutterClose1
```

The first was sent a long time ago and retained (I didn't see this, hence the ghost switching, sorry).

From the detailed log:
```
tele/shutter-garden/INFO3 {"RestartReason":"Software/System restart"}
stat/shutter-garden/RESULT {"POWER1":"OFF"}
stat/shutter-garden/POWER1 OFF
stat/shutter-garden/RESULT {"POWER2":"OFF"}
stat/shutter-garden/POWER2 OFF
stat/shutter-garden/LOGGING 00:00:08.493 RTC: UTC 2020-12-26T18:06:29, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
stat/shutter-garden/LOGGING 19:06:29.264 CFG: Saved to flash at FA, Count 283, Bytes 4096
stat/shutter-garden/LOGGING 19:06:30.177 MQT: Attempting connection...
stat/shutter-garden/LOGGING 19:06:30.192 MQT: Connected
stat/shutter-garden/LOGGING 19:06:30.196 MQT: tele/shutter-garden/LWT = Online (retained)
stat/shutter-garden/LOGGING 19:06:30.200 MQT: cmnd/shutter-garden/POWER = 

[emphasis here:]
stat/shutter-garden/LOGGING 19:06:30.203 MQT: Subscribe to cmnd/shutter-garden/#
stat/shutter-garden/LOGGING 19:06:30.207 MQT: Subscribe to cmnd/shutters/#
stat/shutter-garden/LOGGING 19:06:30.211 MQT: Subscribe to cmnd/shutters-ground-floor/#
stat/shutter-garden/LOGGING 19:06:30.214 MQT: Subscribe to cmnd/DVES_E5C289_fb/#
stat/shutter-garden/LOGGING 19:06:30.217 MQT: Unsubscribe from homeassistant/status
```

Is it expected the the order of the topics Tasmote subscribed to matters here? As in the command to `shutters-ground-floor` overwrote the others because it was subscribed to last?

ascillato2, 2020-12-27:
> Thanks, it indeed came from a retained message...

Great that you found the source of your problem.

closed 2020-12-27 by ascillato2