arendst/Tasmota #10258: Publish to MQTT Topic with the space character

JZ-SmartThings, 2020-12-26:
### PROBLEM DESCRIPTION
I created a simple rule to publish to MQTT when the button is pressed multiple times. However, I cannot get the space character to work correctly. For folks using SmartThings MQTT Bridge, it literally takes the device name with the spaces and turns that into a topic. It would be great if something like %space% was valid like %topic% is inside of rules, etc.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Multiple but mostly Sonoff S31 and Gosund WP6
- [ ] Tasmota binary firmware version number used: tasmota.bin
  - [ ] Pre-compiled
  - [X] Self-compiled --- only changed MQTT Event subscriptions to be on.
- [X] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
05:58:52 MQT: stat/Extra_Gosund/STATUS7 = {"StatusTIM":{"UTC":"2020-12-26T04:58:52","Local":"2020-12-26T05:58:52","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"16:58"}}
05:58:52 MQT: stat/Extra_Gosund/STATUS10 = {"StatusSNS":{"Time":"2020-12-26T05:58:52"}}
05:58:52 MQT: stat/Extra_Gosund/STATUS11 = {"StatusSTS":{"Time":"2020-12-26T05:58:52","Uptime":"0T04:43:41","UptimeSec":17021,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ZIOT","BSSId":"76:83:C2:AD:F6:80","Channel":11,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:16"}}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua

06:00:23 MQT: stat/Extra_Gosund/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
06:00:24 MQT: stat/Extra_Gosund/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":144,"Free":367,"Rules":"ON Button1#Action=SINGLE do POWER1 TOGGLE endon  ON Button1#Action=PENTA do Publish smartthings/Office%space%Door%space%Lock/lock unlocked endon"}
06:00:24 MQT: stat/Extra_Gosund/RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":288,"Free":223,"Rules":"ON mqtt#connected DO Subscribe rcvMQTT, smartthings/%topic%/switch ENDON   ON Event#rcvMQTT=on DO Power1 1 ENDON   ON Event#rcvMQTT=off DO Power1 0 ENDON   ON Power1#state=1 DO Publish smartthings/%topic%/switch on ENDON   ON Power1#state=0 DO Publish smartthings/%topic%/switch off ENDON"}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
06:01:01 MQT: stat/Extra_Gosund/STATUS = {"Status":{"Module":0,"DeviceName":"Extra_Gosund","FriendlyName":["Extra_Gosund"],"Topic":"Extra_Gosund","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:01:01 MQT: stat/Extra_Gosund/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.0.1/tasmota/tasmota-minimal.bin","RestartReason":"Power On","Uptime":"0T04:45:50","StartupUTC":"2020-12-26T00:15:11","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2020-12-16T05:43:21","SaveCount":58,"SaveAddress":"F5000"}}
06:01:01 MQT: stat/Extra_Gosund/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-22T09:41:15","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"371/699"}}
06:01:01 MQT: stat/Extra_Gosund/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ZIOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008809","1405C8000100060000005A00000000000000","00800008","00006000","00000000"]}}
06:01:01 MQT: stat/Extra_Gosund/STATUS4 = {"StatusMEM":{"ProgramSize":590,"Free":412,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
06:01:01 MQT: stat/Extra_Gosund/STATUS5 = {"StatusNET":{"Hostname":"Extra_Gosund","IPAddress":"192.168.30.190","Gateway":"192.168.30.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.30.1","Mac":"F4:CF:A2:20:87:63","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
06:01:01 MQT: stat/Extra_Gosund/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.251","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_208763","MqttUser":"mqtt","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
06:01:01 MQT: stat/Extra_Gosund/STATUS7 = {"StatusTIM":{"UTC":"2020-12-26T05:01:01","Local":"2020-12-26T06:01:01","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"16:58"}}
06:01:01 MQT: stat/Extra_Gosund/STATUS10 = {"StatusSNS":{"Time":"2020-12-26T06:01:01"}}
06:01:01 MQT: stat/Extra_Gosund/STATUS11 = {"StatusSTS":{"Time":"2020-12-26T06:01:01","Uptime":"0T04:45:50","UptimeSec":17150,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ZIOT","BSSId":"76:83:C2:AD:F6:80","Channel":11,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:16"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Try to publish from the console to a topic with a space character.

### EXPECTED BEHAVIOUR
To have some ability for special character recognition.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-12-26:
Nice issue.

The MQTT standard does indeed support spaces in topics.

In the case of the publish command I use the space to seperate the topic from the payload.

The easiest approach I think would be if I introduce a special character as space replacement (not %space% as it wastes too many characters.) Let me think about a useful character to use and see if this can work...

JZ-SmartThings, 2020-12-26:
@arendst you're friggin awesome, thank you!

Don't get me wrong, the fix was super simple, just NodeRED with MQTT in then out using same payload. However, internal Tasmota space character ability would eliminate one of the thousand things already being maintained :)

arendst, 2020-12-26:
Pls give it a try.

Yash1903, 2020-12-26:
Use Separator(like..@,#,$,&,*) instead of %Space%

JZ-SmartThings, 2020-12-26:
@Yash1903 I definitely saw that necessity and Theo's warning above about topics. However, with an already established environment with MANY things plugged in and some that automatically pass the space (SmartThings MQTT Bridge). It's just not realistic for me to start renaming stuff with underscores --- and it looks ugly :)

@arendst I'll check it out ASAP and get back to you.

JZ-SmartThings, 2020-12-26:
@arendst I'm very impressed, mostly with your speed :) it worked like a charm, which I kinda expected with your level of work. Thanks for sharing your amazing firmware. The reliability and flexibility of local control is off the charts.

Quick question: have you ever considered releasing tasmota.bin with SUPPORT_MQTT_EVENT enabled... or maybe tasmota-mqttsub.bin... just wondering if there has been enough interest for that or if I'm alone with that request.

All the best!

ascillato2, 2020-12-27:
Closing this issue as it has been fixed.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-12-27 by ascillato2