arendst/Tasmota #10264: `ON switch1#state DO` is not executed, when I push the button

rosch100, 2020-12-26:
### PROBLEM DESCRIPTION
In my setup, 
`ON switch1#state DO`
is not executed, when I push the button.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff TH16 with temp sensor DS18B20
- [x] Tasmota binary firmware version number used: 9.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
18:04:20 CMD: Backlog Template; Module; GPIO 255
18:04:20 MQT: kuechenheizung/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:04:20 MQT: kuechenheizung/stat/RESULT = {"Module":{"4":"Sonoff TH"}}
18:04:20 MQT: kuechenheizung/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"1312":"DS18x20"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
17:57:23 MQT: kuechenheizung/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":32,"Free":479,"Rules":"ON switch1#state DO mem5 1 ENDON"}
17:57:23 MQT: kuechenheizung/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
17:57:23 MQT: kuechenheizung/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [x] Provide the output of this command: `Status 0`:
```lua
17:58:13 MQT: kuechenheizung/stat/STATUS = {"Status":{"Module":4,"DeviceName":"Küchenheizung","FriendlyName":["Küchenheizung"],"Topic":"kuechenheizung","ButtonTopic":"kuechenheizung","Power":1,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"","SwitchMode":[3,3,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:58:13 MQT: kuechenheizung/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin","RestartReason":"Software/System restart","Uptime":"0T03:38:29","StartupUTC":"2020-12-26T13:19:44","Sleep":50,"CfgHolder":4617,"BootCount":24,"BCResetTime":"2020-04-09T07:47:50","SaveCount":178,"SaveAddress":"F4000"}}
17:58:13 MQT: kuechenheizung/stat/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020.12.21 15:09:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"481/699"}}
17:58:13 MQT: kuechenheizung/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Taube 4","Borkheide"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["040A800A","2805C8000100060000005A00000000000000","00800200","00006000","00000001"]}}
17:58:13 MQT: kuechenheizung/stat/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14605E","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
17:58:14 MQT: kuechenheizung/stat/STATUS5 = {"StatusNET":{"Hostname":"Kuechenheizung","IPAddress":"192.168.101.41","Gateway":"192.168.101.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.101.1","Mac":"B4:E6:2D:7F:E4:2D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:58:14 MQT: kuechenheizung/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.101.32","MqttPort":1883,"MqttClientMask":"kuechenheizung","MqttClient":"kuechenheizung","MqttUser":"mosquitto","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:58:14 MQT: kuechenheizung/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-12-26T16:58:14","Local":"2020-12-26T17:58:14","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"16:58"}}
17:58:14 MQT: kuechenheizung/stat/STATUS10 = {"StatusSNS":{"Time":"2020-12-26T17:58:14","DS18B20":{"Id":"01145D84CFE4","Temperature":9.1},"TempUnit":"C"}}
17:58:14 MQT: kuechenheizung/stat/STATUS11 = {"StatusSTS":{"Time":"2020-12-26T17:58:14","Uptime":"0T03:38:30","UptimeSec":13110,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","Wifi":{"AP":2,"SSId":"Borkheide","BSSId":"38:10:D5:DD:DF:51","Channel":6,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
 17:59:19 MQT: kuechenheizung/stat/RESULT = {"WebLog":4}
17:59:19 CFG: in Flash gespeichert am FB, zählen 179, Bytes 4096
17:59:30 WIF: Prüfe Verbindung...
17:59:33 APP: Knopf1 Mehrfachdruck 1
17:59:34 MQT: kuechenheizung/cmnd/POWER1 = ON
17:59:34 MQT: kuechenheizung/stat/BUTTON1T = {"TRIG":"SINGLE"}
17:59:34 MQT: Datengröße 2, "kuechenheizung/cmnd/POWER1 ON"
17:59:34 SRC: MQTT
17:59:34 CMD: Gruppe 0, Index 1, Befehl "POWER", Daten "ON"
17:59:34 MQT: kuechenheizung/tele/STATE = {"Time":"2020-12-26T17:59:34","Uptime":"0T03:39:50","UptimeSec":13190,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","Wifi":{"AP":2,"SSId":"Borkheide","BSSId":"38:10:D5:DD:DF:51","Channel":6,"RSSI":38,"Signal":-81,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:59:34 MQT: kuechenheizung/stat/RESULT = {"POWER1":"ON"}
17:59:34 MQT: kuechenheizung/stat/POWER1 = ON
17:59:50 WIF: Prüfe Verbindung...
17:59:51 CMD: mem5
17:59:51 SRC: WebConsole from 192.168.101.24
17:59:51 CMD: Gruppe 0, Index 5, Befehl "MEM", Daten ""
17:59:51 MQT: kuechenheizung/stat/RESULT = {"Mem5":"0"}
```

### TO REPRODUCE
Rule set:
```
Backlog SwitchMode1 3; Rule 1; mem5 0
Rule1 
  ON switch1#state DO mem5 1 ENDON
```
Result:
```
17:49:27 CMD: Backlog SwitchMode1 3; Rule 1; mem5 0
17:49:27 MQT: kuechenheizung/stat/RESULT = {"SwitchMode1":3}
17:49:27 MQT: kuechenheizung/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":2,"Free":509,"Rules":"''"}
17:49:28 MQT: kuechenheizung/stat/RESULT = {"Mem5":"0"}
17:49:38 CMD: Rule1    ON switch1#state DO mem5 1 ENDON
17:49:38 RUL: Stored uncompressed, would compress from 32 to 28 (-12%)
17:49:38 MQT: kuechenheizung/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":32,"Free":479,"Rules":"ON switch1#state DO mem5 1 ENDON"}
17:49:53 MQT: kuechenheizung/cmnd/POWER1 = ON
17:49:53 MQT: kuechenheizung/stat/BUTTON1T = {"TRIG":"SINGLE"}
17:49:53 MQT: kuechenheizung/tele/STATE = {"Time":"2020-12-26T17:49:53","Uptime":"0T03:30:09","UptimeSec":12609,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER1":"ON","Wifi":{"AP":2,"SSId":"Borkheide","BSSId":"38:10:D5:DD:DF:51","Channel":6,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:49:53 MQT: kuechenheizung/stat/RESULT = {"POWER1":"ON"}
17:49:53 MQT: kuechenheizung/stat/POWER1 = ON
17:49:54 MQT: kuechenheizung/tele/HASS_STATE = {"Version":"9.2.0(tasmota)","BuildDateTime":"2020.12.21 15:09:06","Module or Template":"Sonoff TH","RestartReason":"Software/System restart","Uptime":"0T03:30:10","Hostname":"Kuechenheizung","IPAddress":"192.168.101.41","RSSI":"46","Signal (dBm)":"-77","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:03","MqttCount":1,"LoadAvg":19}
17:49:55 MQT: kuechenheizung/tele/STATE = {"Time":"2020-12-26T17:49:55","Uptime":"0T03:30:11","UptimeSec":12611,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","Wifi":{"AP":2,"SSId":"Borkheide","BSSId":"38:10:D5:DD:DF:51","Channel":6,"RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:49:55 MQT: kuechenheizung/tele/SENSOR = {"Time":"2020-12-26T17:49:55","DS18B20":{"Id":"01145D84CFE4","Temperature":9.0},"TempUnit":"C"}
17:50:06 CMD: mem5
17:50:06 MQT: kuechenheizung/stat/RESULT = {"Mem5":"0"}
```

### EXPECTED BEHAVIOUR
I expect that the rule `ON switch1#state DO mem5 1  ENDON` will be extecuted after pushing the button.

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-12-26:
Assuming that your button is configured as a `BUTTON` and not a `SWITCH`, you'll need to trigger on `BUTTON1#STATE`.

rosch100, 2020-12-26:
Thanks for the hint - I did extend to rule set to:
```
Rule1
  ON switch1#state DO mem5 1 ENDON
  ON button1#state DO mem5 1 ENDON
```
so that this case should be covered.
Still no reaction on pushing the button.

meingraham, 2020-12-26:
When you push the button, is the device toggling?

I'm not familiar with this message - `MQT: kuechenheizung/stat/BUTTON1T = {"TRIG":"SINGLE"}`. Have you decoupled the relay from the button (`SetOption73`)? Other than that, I wonder if any of your HASS configuration settings is affecting the "coupling" of the button. I don't use HASS so cannot provide any insight but I do know that the state of buttons/switches is transmitted differently for HASS configured Tasmota.

@effelle ?

closed 2020-12-27 by ascillato2