arendst/Tasmota #10277: Incomplete effect list (schemes) in Home Assistant with WS2812 and automatic discovery

MBaNL, 2020-12-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When using WS2812 addressable LED-strips in combination with the Home Assistant automatic discovery (SetOption19) the effect list (schemes) in Home Assistant is the one for non-addressable LED strips (only scheme 0 to 4, instead of scheme 0 to 12).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): D1 Mini
- [x] Tasmota binary firmware version number used: 9.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:59:08 MQT: stat/tasmota_9F74BF/RESULT = {"NAME":"Klok","GPIO":[0,0,0,0,0,1376,0,0,0,0,1216,0,0,0],"FLAG":0,"BASE":18}
15:59:08 MQT: stat/tasmota_9F74BF/RESULT = {"Module":{"0":"Klok"}}
15:59:09 MQT: stat/tasmota_9F74BF/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"1376":"WS2812"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"1216":"AM2301"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
16:00:25 MQT: stat/tasmota_9F74BF/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
16:00:25 MQT: stat/tasmota_9F74BF/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
16:00:26 MQT: stat/tasmota_9F74BF/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:01:41 MQT: stat/tasmota_9F74BF/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_9F74BF","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:01:41 MQT: stat/tasmota_9F74BF/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:03:23","StartupUTC":"2020-12-27T14:58:18","Sleep":50,"CfgHolder":4617,"BootCount":1207,"BCResetTime":"2020-12-13T20:54:41","SaveCount":7476,"SaveAddress":"F8000"}}
16:01:41 MQT: stat/tasmota_9F74BF/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-17T10:05:31","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"388/699"}}
16:01:41 MQT: stat/tasmota_9F74BF/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iot_netwerk_1","iot_netwerk_2"],"TelePeriod":15,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005AFF000000000000","00000200","00006000","00000000"]}}
16:01:41 MQT: stat/tasmota_9F74BF/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
16:01:41 MQT: stat/tasmota_9F74BF/STATUS5 = {"StatusNET":{"Hostname":"tasmota_9F74BF-5311","IPAddress":"192.168.1.103","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.10","Mac":"68:C6:3A:9F:74:BF","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:01:41 MQT: stat/tasmota_9F74BF/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9F74BF","MqttUser":"admin","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:01:41 MQT: stat/tasmota_9F74BF/STATUS7 = {"StatusTIM":{"UTC":"2020-12-27T15:01:41","Local":"2020-12-27T16:01:41","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"08:43","Sunset":"16:59"}}
16:01:41 MQT: stat/tasmota_9F74BF/STATUS10 = {"StatusSNS":{"Time":"2020-12-27T16:01:41","AM2301":{"Temperature":21.7,"Humidity":47.4,"DewPoint":10.0},"TempUnit":"C"}}
16:01:41 MQT: stat/tasmota_9F74BF/STATUS11 = {"StatusSTS":{"Time":"2020-12-27T16:01:41","Uptime":"0T00:03:23","UptimeSec":203,"Heap":19,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":104,"MqttCount":3,"POWER":"ON","Dimmer":5,"Color":"13,6,8","HSBColor":"338,57,5","Channel":[5,2,3],"Scheme":5,"Width":1,"Fade":"ON","Speed":4,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"iot_netwerk_1","BSSId":"04:B3:F5:FB:C9:40","Channel":5,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:10"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
16:19:21 MQT: homeassistant/device_automation/9F74BF_SW_7_HOLD/config =  (retained)
16:19:21 MQT: homeassistant/binary_sensor/9F74BF_SW_7/config =  (retained)
16:19:21 MQT: homeassistant/device_automation/9F74BF_SW_8_TOGGLE/config =  (retained)
16:19:21 MQT: homeassistant/device_automation/9F74BF_SW_8_HOLD/config =  (retained)
16:19:21 MQT: homeassistant/binary_sensor/9F74BF_SW_8/config =  (retained)
16:19:21 MQT: homeassistant/sensor/9F74BF_AM2301_Temperature/config = {"name":"Tasmota AM2301 Temperature","stat_t":"tele/tasmota_9F74BF/SENSOR","avty_t":"tele/tasmota_9F74BF/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"9F74BF_AM2301_Temperature","dev":{"ids":["9F74BF"]},"unit_of_meas":"°C","dev_cla":"temperature","frc_upd":true,"val_tpl":"{{value_json['AM2301']['Temperature']}}"} (retained)
16:19:21 MQT: homeassistant/sensor/9F74BF_AM2301_Humidity/config = {"name":"Tasmota AM2301 Humidity","stat_t":"tele/tasmota_9F74BF/SENSOR","avty_t":"tele/tasmota_9F74BF/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"9F74BF_AM2301_Humidity","dev":{"ids":["9F74BF"]},"unit_of_meas":"%","dev_cla":"humidity","frc_upd":true,"val_tpl":"{{value_json['AM2301']['Humidity']}}"} (retained)
16:19:21 MQT: homeassistant/sensor/9F74BF_AM2301_DewPoint/config = {"name":"Tasmota AM2301 DewPoint","stat_t":"tele/tasmota_9F74BF/SENSOR","avty_t":"tele/tasmota_9F74BF/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"9F74BF_AM2301_DewPoint","dev":{"ids":["9F74BF"]},"unit_of_meas":"°C","ic":"mdi:weather-rainy","frc_upd":true,"val_tpl":"{{value_json['AM2301']['DewPoint']}}"} (retained)
16:19:21 MQT: homeassistant/cover/9F74BF_SHT_1/config =  (retained)
16:19:21 RSL: homeassistant/cover/9F74BF_SHT_2/config = 
16:19:22 RSL: homeassistant/cover/9F74BF_SHT_3/config = 
16:19:22 MQT: homeassistant/cover/9F74BF_SHT_4/config =  (retained)
16:19:22 MQT: homeassistant/switch/9F74BF_RL_1/config =  (retained)
16:19:22 MQT: homeassistant/light/9F74BF_LI_1/config = {"name":"Tasmota","stat_t":"tele/tasmota_9F74BF/STATE","avty_t":"tele/tasmota_9F74BF/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"cmnd/tasmota_9F74BF/POWER","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","uniq_id":"9F74BF_LI_1","dev":{"ids":["9F74BF"]},"bri_cmd_t":"cmnd/tasmota_9F74BF/Dimmer","bri_stat_t":"tele/tasmota_9F74BF/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer}}","rgb_cmd_t":"cmnd/tasmota_9F74BF/Color2","rgb_stat_t":"tele/tasmota_9F74BF/STATE","rgb_val_tpl":"{{value_json.Color.split(',')[0:3]|join(',')}}","fx_cmd_t":"cmnd/tasmota_9F74BF/Scheme","fx_stat_t" ... (retained)
16:19:22 MQT: homeassistant/switch/9F74BF_RL_2/config =  (retained)
16:19:22 RSL: homeassistant/light/9F74BF_LI_2/config = 
16:19:22 RSL: homeassistant/switch/9F74BF_RL_3/config = 
[... cut ...] "cmnd/tasmota_9F74BF/Color2","rgb_stat_t":"tele/tasmota_9F74BF/STATE","rgb_val_tpl":"{{value_json.Color.split(',')[0:3]|join(',')}}","fx_cmd_t":"cmnd/tasmota_9F74BF/Scheme","fx_stat_t":"tele/tasmota_9F74BF/STATE","fx_val_tpl":"{{value_json.Scheme}}","fx_list":["0","1","2","3","4"]}
```
[image]
[image]

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
See EXPECTED BEHAVIOUR and CURRENT BEHAVIOUR.

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

ascillato, 2020-12-27:
Please, try setoption19 0 and use Tasmota Native Integration for home assistant. When ysing setoption19 0, you will have your device added automatically under home assistant integrations menu.

MBaNL, 2020-12-27:
Haven't touched the Tasmota integration before since automatic discovery under MQTT worked flawlessly for me but gave it a try for this specific issue. Unfortunately this doesn't solve the issue.
Although it creates a nice list with user friendly names it does not show the schemes/effects for addressable LEDs.

[image]

closed 2021-04-14 by ascillato2