arendst/Tasmota #10282: Arduino_ST7789: Hard-coded _xstart and _ystart prevent 135*240 TFT from working correct

jeroenvermeulen, 2020-12-28:
### PROBLEM DESCRIPTION
A 135*240 ST7789 TFT is not working correct, caused by hard-coded [_xstart](https://github.com/arendst/Tasmota/blob/83b3aa576da923748e33df154922e1fa2caa980b/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.cpp#L338-L339) and [_ystart](https://github.com/arendst/Tasmota/blob/83b3aa576da923748e33df154922e1fa2caa980b/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.cpp#L346-L347) in  [Arduino_ST7789.cpp](https://github.com/arendst/Tasmota/blob/83b3aa576da923748e33df154922e1fa2caa980b/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.cpp#L338-L347)

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): [LILYGO® TTGO T-Display ESP32 WiFi And Bluetooth Module Development Board 1.14 Inch LCD](https://www.aliexpress.com/i/33050639690.html)
- [x] Tasmota binary firmware version number used: 9.2.90
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: `esptool.py --chip esp32 --port /dev/tty.usbserial-01F93A32 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 build_output/firmware/bootloader_dout_40m.bin 0x8000 build_output/firmware/partitions.bin 0xe000 build_output/firmware/boot_app0.bin 0x10000 build_output/firmware/tasmota32.bin`
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:31:52 MQT: stat/ttgo-garage/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
00:31:53 MQT: stat/ttgo-garage/RESULT = {"Module":{"1":"ESP32-DevKit"}}
00:31:53 MQT: stat/ttgo-garage/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"928":"SSPI CS"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"960":"SSPI DC"},"GPIO17":{"0":"None"},"GPIO18":{"896":"SSPI SCLK"},"GPIO19":{"864":"SSPI MOSI"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"1024":"OLED Reset"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"}}}
00:31:53 MQT: stat/ttgo-garage/RESULT = {"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

00:32:36 MQT: stat/ttgo-garage/STATUS = {"Status":{"Module":1,"DeviceName":"ttgo-garage","FriendlyName":["TTGO Garage","Tasmota2"],"Topic":"ttgo-garage","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:32:36 MQT: stat/ttgo-garage/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:09:31","StartupUTC":"2020-12-27T23:23:05","Sleep":50,"CfgHolder":4617,"BootCount":72,"BCResetTime":"2020-12-27T20:47:59","SaveCount":191}}
00:32:36 MQT: stat/ttgo-garage/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-27T23:51:58","Core":"1_0_4_2","SDK":"v3.3.4-232-g44ec7972b","CpuFrequency":80,"Hardware":"ESP32","CR":"449/699"}}
00:32:36 MQT: stat/ttgo-garage/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["GRS-IOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
00:32:37 MQT: stat/ttgo-garage/STATUS4 = {"StatusMEM":{"ProgramSize":1408,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FFAC78F","001403A1","000000CF","010013C0","40000981","00004020","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,13,20,21,24,26,27,29","Sensors":"1,2,3,5,6,62"}}
00:32:37 MQT: stat/ttgo-garage/STATUS5 = {"StatusNET":{"Hostname":"ttgo-garage","IPAddress":"192.168.7.62","Gateway":"192.168.7.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.7.1","Mac":"10:52:1C:66:D8:B0","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:32:37 MQT: stat/ttgo-garage/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.7.2","MqttPort":1883,"MqttClientMask":"ttgo-garage_%06X","MqttClient":"ttgo-garage_66D8B0","MqttUser":"user","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:32:37 MQT: stat/ttgo-garage/STATUS7 = {"StatusTIM":{"UTC":"2020-12-27T23:32:37","Local":"2020-12-28T00:32:37","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:00"}}
00:32:37 MQT: stat/ttgo-garage/STATUS10 = {"StatusSNS":{"Time":"2020-12-28T00:32:37"}}
00:32:37 MQT: stat/ttgo-garage/STATUS11 = {"StatusSTS":{"Time":"2020-12-28T00:32:37","Uptime":"0T00:09:32","UptimeSec":572,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"GRS-IOT","BSSId":"02:9F:C2:2D:4C:FD","Channel":9,"RSSI":38,"Signal":-81,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Configure in [Arduino_ST7789.h](https://github.com/arendst/Tasmota/blob/development/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.h#L42-L46):
```
#define ST7789_TFTWIDTH 	135
#define ST7789_TFTHEIGHT 	240

#define ST7789_240x240_XSTART 53
![IMG_4501](https://user-images.githubusercontent.com/658024/103182007-ca8e6500-48a7-11eb-811f-07204df30c29.jpg)

#define ST7789_240x240_YSTART 40
```
Compile `tasmota32`, flash.
Enter in console:
```
BackLog  DisplayRotate 0; DisplayFont 1; DisplayCols 19; DisplayRows 20; DisplayMode 0
DisplayText [zl1c1]Hello, World!
```
### EXPECTED BEHAVIOUR
A black display with only the text `Hello World!`

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
Next step would be to make the width, height, xstart and ystart configurable.
When I undo the two hard-coded overrides in  [Arduino_ST7789.cpp](https://github.com/arendst/Tasmota/blob/83b3aa576da923748e33df154922e1fa2caa980b/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.cpp#L338-L347), and with my custom settings in [Arduino_ST7789.h](https://github.com/arendst/Tasmota/blob/development/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.h#L42-L46) it works fine:
[image]

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-12-28:
Thanks for sharing this. Please, can you make a pull request with your change? Thanks.

gemu2015, 2020-12-28:
@jeroenvermeulen 

i made display width and height configurable. since i only have a 256*256 display i can't test.

you may try this version

[st7789.zip](https://github.com/arendst/Tasmota/files/5746609/st7789.zip)

jeroenvermeulen, 2020-12-28:
@gemu2015 Your change is good, it indeed solves the part of configuring the size of the display, I no longer need to set
```
#define ST7789_TFTWIDTH 	135
#define ST7789_TFTHEIGHT 	240
```
inside [Arduino_ST7789.h](https://github.com/arendst/Tasmota/blob/development/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.h#L42-L43).

Thanks for making the size configurable. Do you also have an idea how to make `ST7789_240x240_XSTART` and `ST7789_240x240_YSTART` (confusing names) configurable?

To make my 135*240 display work, I need to set in [Arduino_ST7789.h](https://github.com/arendst/Tasmota/blob/development/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.h#L42-L43):
```
#define ST7789_240x240_XSTART 53
#define ST7789_240x240_YSTART 40
```
+ undo the hard-coded `80` overrides of `_ystart` and `_xstart` in [Arduino_ST7789.cpp](https://github.com/arendst/Tasmota/blob/83b3aa576da923748e33df154922e1fa2caa980b/lib/lib_display/Arduino_ST7789-gemu-1.0/Arduino_ST7789.cpp#L338-L347)

With these + your changes from st7789.zip my 135*240 display works both in portrait and landscape.
I use these settings:
```
BackLog  DisplayRotate 0; DisplayWidth 135; DisplayHeight 240; DisplayFont 0; DisplayCols 22; DisplayRows 29;
BackLog  DisplayRotate 0; DisplayWidth 135; DisplayHeight 240; DisplayFont 1; DisplayCols 19; DisplayRows 20;
BackLog  DisplayRotate 1; DisplayWidth 240; DisplayHeight 135; DisplayFont 0; DisplayCols 40; DisplayRows 17;
BackLog  DisplayRotate 1; DisplayWidth 240; DisplayHeight 135; DisplayFont 1; DisplayCols 34; DisplayRows 11;
```

@ascillato2 I could make a PR but since @gemu2015 is the expert in this area and he has already made the size configurable, I think it is best he makes  PR for solving this issue (#10282) as well. I think the goal should be:
[ ] Make XSTART and YSTART configurable
[ ] Remove the `80` overrides
I hope the 240*240 display works when you set XSTART and YSTART to `80`.

I am happy to help testing a PR or other code with my 135*240 display. 
I could order a 240*240 but it will take multiple weeks to get it from Mr Ali.

gemu2015, 2020-12-28:
@jeroenvermeulen 

ok, checked other st7789 drivers for the issue and they solved this by a #defines
as far as i found there are mostly 2 versions with this controller available 
240x240 and 240x135
while the 1. one has zero offsets the 2. one has offset 40 and 52 or 53 depending on rotation

so i will try to set the offsets depending on the resolution.

i will post the driver in my GitHub fork, and you should test the new versions and report back.

https://github.com/gemu2015/Sonoff-Tasmota

when both versions are tested i will make a pr on tasmota

please open an issue on my fork and we can communicate there until it works

gemu2015, 2020-12-28:
@jeroenvermeulen 

first try is online

jeroenvermeulen, 2020-12-28:
We are working on it here: https://github.com/gemu2015/Sonoff-Tasmota/issues/28

blakadder, 2020-12-29:
I have version using
```
backlight_pin: GPIO5
cs_pin: GPIO22
dc_pin: GPIO21
reset_pin: GPIO18
clk_pin: GPIO19
mosi_pin: GPIO23
miso_pin: GPIO25
TFT_WIDTH=240
TFT_HEIGHT=320
TFT_INVERSION_OFF     ; for normal colors
TFT_RGB_ORDER=TFT_BGR ; Colour order Blue-Green-Red
```
with an FT6336 Utouch chip

PR #10310 (2020-12-30): fix: ST7789 display driver for 135*240px
Improved the driver to support both 135\*240 and 240\*240 formats. These are the most common for the ST7789 display.
Thanks to @gemu2015 for your help and improvements.

## Description:

**Related issue (if applicable):** fixes #10282

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested a [... cut]
commit 8fb6ba2b (2020-12-30): fix: ST7789 display driver for 135*240px

Improved the driver to support both 135*240 and 240*240 formats. These are the most common for the ST7789 display.
Fixes #10282
closed 2020-12-30 by arendst